1technophile/OpenMQTTGateway #1224: Cannot find where to specify Mi Bind Keys

mihsu81, 2022-06-10:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Cannot find where to specify Mi Bind Keys in order to decrypt service data sent by a Xiaomi XMWSDJ04MMC.
Since support for this device was added in https://github.com/1technophile/OpenMQTTGateway/pull/1162, I assume there's a way to specify the Bind Key, but I cannot find it.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.12)

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2022-06-11:
The support added enables to connect to the device to retrieve data.
It doesn't use broadcasted advertisements, for the latest we would need a bind key, but OpenMQTTGateway doesn't have the function yet.

mihsu81, 2022-06-12:
Understood. Closing the issue, since my expectations exceed what OMG currently offers.

closed 2022-06-12 by mihsu81