1technophile/OpenMQTTGateway #1225: M5Stack/M5Stick w/ RF?

colinxs, 2022-06-13:
Hope its alright to ask this here. I'm interested in using an M5StickC Plus with OMG but need RF Tx/Rx. It looks like GPIO12 is available but not 27. Are there alternate ports that would work without issue?

Mirdinus, 2023-06-06:
Hi @colinxs, it depends on your RF board. If you have the simple 433MHz board you can use build flags `  '-DRF_RECEIVER_GPIO=13'` and `'-DRF_EMITTER_GPIO=15'` with pretty much any available pin. If you are using CC1101, then I believe there is no option to change pins as of right now

[Using build flags with RF](https://github.com/1technophile/OpenMQTTGateway/blob/15416e99b1a0f8ab433d8c8a1101db6f2d62ee3c/docs/upload/builds.md?plain=1#LL96C1-L97C25)

DigiH, 2023-06-06:
CC1101 pin build flags

https://github.com/1technophile/OpenMQTTGateway/blob/development/environments.ini#L703

Mirdinus, 2023-06-06:
Thank you for correcting my comment. I wasn't clear about the type of pins I was mentioning. Please let me correct the comment. 
You cannot update pins that are used for SPI, but as @DigiH correctly pointed out, you can easily change the GDO pins.

DigiH, 2023-06-06:
@Mirdinus I already saw your other issue now :) sorry, was only commenting for @colinxs GDO pin 27 above, in case of a CC1101.

closed 2023-10-16 by github-actions[bot]