1technophile/OpenMQTTGateway #1228: Publishing rules are ambiguous

bvduijn, 2022-06-26:
Hi,

I was testing my new openmqttgateway and noticed that, even after I've disabled simplePublishing I was still seeing it being set  up in the serial monitor 

My personal platformio_env.ini contains  -DsimplePublishing=false as the examples in pio contain the value 'true'
https://github.com/1technophile/OpenMQTTGateway/blob/6efc90b19f2c28b98f6549df8fdb9014c3924c11/platformio.ini#L287

This would lead me to think setting it to false would disable it. However in 

https://github.com/1technophile/OpenMQTTGateway/blob/6efc90b19f2c28b98f6549df8fdb9014c3924c11/main/ZgatewayRF.ino#L176

it looks just if it is defined and doesn't take the value into account. (true or false)

Perhaps the platformio.ini could just read 
```
-DsimplePublishing
```

Alternatively another option would be:

```
#  ifdef simpleReceiving
if (simpleReceiving) { 
    //do setup
}
```
But this will add more code and I think the first option is cleaner

Any thoughts?

DigiH, 2022-06-26:
Hi @bvduijn, yes there are inconsistencies between the definitions and implementations of these publishing options.

Until this is being addressed and a fix submitted you best just comment out/delete 

`'-DsimplePublishing=false' `

in your personal environment for it to work the way you want it.

DigiH, 2022-07-24:
Hi @bvduijn, this has been addressed and merged with

https://github.com/1technophile/OpenMQTTGateway/pull/1229

closed 2023-04-10 by 1technophile