1technophile/OpenMQTTGateway #1232: About the use bug of esp8266-12s module

mildrabbit, 2022-07-10:
I use the esp8266-12s module to write nodemcuv2 IR firmware for the first network distribution, and configure the mqtt server user name and password. Everything is normal, and it is used normally. But the binary of homeassistant cannot be used after the module is restarted_ sensor. sys_ Connectivity is always monitored as disconnected. How can I deal with it。I use a module, not a MCU

1technophile, 2022-07-17:
Could you share an extract of your serial monitor ?

mildrabbit, 2022-07-18:
The problem has been found. For the same firmware, there will be problems when using tasmoter program to burn, and there will be no problems when using esphone flasher to burn.

closed 2023-10-16 by github-actions[bot]