1technophile/OpenMQTTGateway #1239: [Feature request] Store some BT parameters persistently in SPIFFS

BadWolf42, 2022-07-15:
I also would like to know if you are interested in a PR, regarding writing persistently some BT parameters (set today on-the-fly by `ZgatewayBT.ino: MQTTtoBT()`) into SPIFFS (example in "/bt_config.json"), like it's done in `main.ino: saveMqttConfig()` for global config ?

1technophile, 2022-07-17:
Not sure if you are aware, there is a solution to retain the parameters by publishing those with the retain flag to the broker.
This way, they are reloaded at the start of the gateway. Regarding retaining when the broker restart, I think there is a need for a special parameter in the broker configuration.
Not sure of how this is easy with the controller to do, so your solution can be interesting also and maybe more intuitive.

1technophile, 2023-04-10:
Done

closed 2023-04-10 by 1technophile