1technophile/OpenMQTTGateway #1241: Cannot connect to wifi

fariouche, 2022-07-16:
Hello,

I'm using 0.9.3 version, and tested with development version, compiled myself.
It is very strange, I used two different boards (M5stack core and wroom dev kit esp32)
Now about the problem: I'm not sure about the conditions, but I was able to configure the wroom32 dev kit to connect to the wofo, but everytime the wifi router reboot, the devkit is unable to reconnect.... After numerous atempts, erasing the flash and set up again all the wifi using the portal it connected again... until the next disconnect.

With the M5stack I cannot even connect once. I can see in the traces that the wifi got connected, and then restarts and then goes back to portal mode! (same behavior as the wroom dev kit... but cannot check yet the traces as I need to open the device again to access the usb)

Here are the traces from the m5stack core:

M5Stack initializing...OK
*WM: [1] Free heap:        150556
*WM: [1] ESP-IDF version:  v3.3.5-1-g85c43024c
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: secure
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: cert
*WM: [2] Added Parameter: name
*WM: [2] Added Parameter: topic
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] your_password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OpenMQTTGateway_ESP32_M5STACK_BLE
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 5113 ms
*WM: [2] WiFi Scan ASYNC found: 7
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] Scan is cached 9746 ms ago
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> 192.168.4.1 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 15580 ms ago
*WM: [1] 7 networks found
*WM: [2] AP:  -47 OpenMQTTGateway_ESP32_BLE
*WM: [2] AP:  -59 XXXXXXX
*WM: [2] AP:  -61 XXXXXXXXXXx
*WM: [2] AP:  -73 XXXXXXXXX
*WM: [2] AP:  -81 XXXXXXXXXXX
*WM: [2] AP:  -87 XXXXXXXXXXX
*WM: [2] AP:  -93 XXXXXXXXXXX
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [3] Sent config page 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] Portal Timeout In 117 seconds
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [2] Parameters 
*WM: [2] -------------------- 
*WM: [2] server: 192.168.30.17
*WM: [2] port: 8883
*WM: [2] user: 
*WM: [2] pass: 
*WM: [2] secure: 1
*WM: [2] cert: -----BEGIN CERTIFICATE----- MIIFXXXXXXXXXXXXXXXXXXXXXXXXX
*WM: [2] name: OpenMQTTGateway_ESP32_M5STACK_BLE
*WM: [2] topic: home/
*WM: [2] -------------------- 
*WM: [3] Sent wifi save page 
*WM: [2] process connect 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to new AP: XXXXXXXX
*WM: [3] Using Password: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXx
*WM: [3] WiFi station enable 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] 192.168.30.186 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] disconnect configportal 
*WM: [0] [ERROR] disconnect configportal - softAPdisconnect FAILED 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
{
  "mqtt_server": "192.168.30.17",
  "mqtt_port": "8883",
  "mqtt_user": "",
  "mqtt_pass": "",
  "mqtt_topic": "home/",
  "gateway_name": "OpenMQTTGateway_ESP32_M5STACK_BLE",
  "mqtt_broker_secure": true,
  "mqtt_broker_cert": "-----BEGIN CERTIFICATE-----\nMIIFfXXXXXXXXXXXXXXXXXXXXXXXXXB,
  "mqtt_ss_index": 0,
  "ota_server_cert": ""
}ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
N: 
   ************* WELCOME TO Ope�M5Stack initializing...OK
*WM: [1] Free heap:        150676
*WM: [1] ESP-IDF version:  v3.3.5-1-g85c43024c
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: secure
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: cert
*WM: [2] Added Parameter: name
*WM: [2] Added Parameter: topic
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] your_password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OpenMQTTGateway_ESP32_M5STACK_BLE
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 5115 ms
*WM: [2] WiFi Scan ASYNC found: 7

1technophile, 2022-07-17:
Hi,

Do you reproduce the issue if you don't use a certificate ?

fariouche, 2022-07-17:
Yes, same :(
No issues with tasmota or esphome (I think they are both also using wifi manager)

1technophile, 2022-07-22:
Did you try to erase the flash before uploading ?

fariouche, 2022-07-22:
Yes, and in that case it will work.... until the next router reboot. After that it will never connect again, even the erase option in the captive portal.

1technophile, 2023-02-04:
Could you try with the last version please?

fariouche, 2023-02-11:
Hello. I switch to esphome , works stable and no issues since several weeks now.

closed 2023-02-11 by fariouche

1technophile, 2023-02-11:
For information, I have changed the connection loop to allow for more connection time. If you want to give it a try, feel free.