1technophile/OpenMQTTGateway #1261: Sonoff Bridge does not boot

morgansimonsen, 2022-08-30:
I have a Sonoff Bridge R2 V2.2 with the direct hack. Have used it with Tasmota and other firmwares without problems. Wanted to try OMG on it.

Flashing from the web is successful, but the device never boots. If I use esptool.py to first erase the flash and then flash the sonoff-rfbridge-direct-firmware.bin the device boots and allows me to connect to the OMG Wifi and configure it with my SSID etc. The device shuts down when I hit Save, but never boots and is inaccessible (no LED).

I have tried version 0.9.11, 0.9.15 and 0.9.16. Any idea what is going on?

DrEVILish, 2022-11-01:
Had the same issue, go to https://web.esphome.io/ flash to a "blank" it makes the software erase the device.
Then reflash with the OMG image, then everything works as expected.

Unknown why this happened.

DigiH, 2023-05-03:
@baunan You could also try

https://espressif.github.io/esptool-js/

for a clean erase.

Let us know if it works after.

baunan, 2023-05-03:
Thanks for the hint, but for some reason I am getting a timeout error when trying to do the erase. This unit works OK on esphome.

esptool.js
Serial port WebSerial VendorID 0x10c4 ProductID 0xea60
Connecting.......
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi,Embedded Flash
Crystal is 26MHz
MAC: (removed)
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 115200
Changed
Erasing flash (this may take a while)...
Error: Timeout

Using this command for esptool, then I got no error. But still no luck on the openmqttgateway with the web version uploader
esptool.exe --port COM5 erase_flash

budda85, 2023-12-03:
Same here.

> Had the same issue, go to https://web.esphome.io/ flash to a "blank" it makes the software erase the device. Then reflash with the OMG image, then everything works as expected.
> 
> Unknown why this happened.

and

> @baunan You could also try
> 
> https://espressif.github.io/esptool-js/
> 
> for a clean erase.
> 
> Let us know if it works after.

does not work

thomasshawn, 2024-02-01:
same issue

closed 2024-05-16 by github-actions[bot]