1technophile/OpenMQTTGateway #1263: Support for Amphiro / Oras digital shower head

savikko, 2022-09-05:
First of all, thanks for the great project!

There is nowadays quite nice BLE showerhead (takes power from water flow) which tells how much water was consumed (realtime and total), also water temperature etc.

Protocol has been reverse-engineered already, ref https://gitlab.com/baze/amphiro_oras_bluetooth_shower_hub/-/tree/main/Protocol_description

What would be the process to get that supported?

Thanks!

DigiH, 2022-09-05:
Hi @savikko,

from your link it looks as if the data can only be accessed by connecting to the shower head, which you can also already do with OpenMQTTGateway with it's [READ/WRITE](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt-esp32-only) commands. You would then need to decode the data yourself according to the reverse-engineered specs though.

If you find that the shower head also freely broadcasts BLE data when it's being used, a decoder for it could be added to [Theengs Decoder](https://github.com/theengs/decoder), which OpenMQTTGateway uses for BLE broadcast decoding. Not sure if you're seeing any of such freely broadcast undecoded messages in OpenMQTTGateway though.

Hope this helps.

savikko, 2022-09-05:
Thanks @DigiH , will check these links. 

I am actually getting something like this from there currently:

```json
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000210008000008000800140001194f000000","rssi":-93}
```

Have not yet tried to decode it.

DigiH, 2022-09-05:
Was the water 33ºC warm for this test by any chance or the test shower time was 33 seconds by then ? ;)

There seems to be some information freely broadcast, but we'd have to figure it out together to find out what is what.

How frequently do these manufacturerdata broadcasts update when the shower is in use? 
Could you give a few more of such examples as above, to see what and how things are changing over time?

savikko, 2022-09-05:
Well, tried again with another shower-test-session:
```json
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023001f01001f00200028000f194f000000","rssi":-85}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023001e01001e001f3f28000e194f000000","rssi":-81,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023001d01001d001e3f28000e194f000000","rssi":-81,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023001c01001c001d3e28000d194f000000","rssi":-82}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000230019010019001a3f27000c194f000000","rssi":-89,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023001801001800193f27000b194f000000","rssi":-88}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023001701001700183f27000b194f000000","rssi":-87,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023000d01000d000e3e250006194f000000","rssi":-87,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023000b01000b000b3e240005194f000000","rssi":-91}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023000901000900093f240004194f000000","rssi":-92}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023000801000800083f240003194f000000","rssi":-87}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023000701000700073f230003194f000000","rssi":-93}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000230005010005000548230002194f000000","rssi":-88,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000023000401000300034a220001194f000000","rssi":-93}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000230002010002000141210001194f000000","rssi":-89,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","rssi":-88,"txpower":0}
```

Messages arrive about 1/sec, but actually a lots of variation:
```
05.09.2022 19.31.50(-1.84 seconds) 
05.09.2022 19.31.49(-1.04 seconds) 
05.09.2022 19.31.49(-0.08 seconds) 
05.09.2022 19.31.45(-3.91 seconds) 
05.09.2022 19.31.45(-0.23 seconds) 
05.09.2022 19.31.43(-1.27 seconds) 
05.09.2022 19.31.33(-10.36 seconds) 
05.09.2022 19.31.32(-1.45 seconds) 
05.09.2022 19.31.29(-2.13 seconds) 
05.09.2022 19.31.28(-1.01 seconds) 
05.09.2022 19.31.27(-1.01 seconds) 
05.09.2022 19.31.25(-2.48 seconds) 
05.09.2022 19.31.24(-1.44 seconds) 
05.09.2022 19.31.21(-2.06 seconds) 
05.09.2022 19.31.21(-0.82 seconds)
```

On this session temperature reached 40ºC and IIRC it was 1.7 litres at the end.

Will do some little more detailed session soon :)

savikko, 2022-09-05:
And another example. This time I took video so I would remember those values :)

So it started with 36'C temp, then stated to rise until 41'C and total was 2,5 litres

Times:
```
05.09.2022 19.42.54 
05.09.2022 19.42.54(-0.2 seconds) 
05.09.2022 19.42.52(-1.81 seconds) 
05.09.2022 19.42.51(-0.85 seconds) 
05.09.2022 19.42.50(-1.3 seconds) 
05.09.2022 19.42.48(-1.89 seconds) 
05.09.2022 19.42.47(-1.02 seconds) 
05.09.2022 19.42.46(-1.52 seconds) 
05.09.2022 19.42.45(-0.62 seconds) 
05.09.2022 19.42.44(-1.43 seconds) 
05.09.2022 19.42.43(-1.01 seconds) 
05.09.2022 19.42.42(-0.54 seconds) 
05.09.2022 19.42.41(-1.3 seconds) 
05.09.2022 19.42.40(-0.62 seconds) 
05.09.2022 19.42.39(-1.23 seconds) 
05.09.2022 19.42.38(-0.92 seconds) 
05.09.2022 19.42.36(-1.86 seconds) 
05.09.2022 19.42.35(-1.26 seconds) 
05.09.2022 19.42.34(-1.26 seconds) 
```

```json
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240015000015001a0029000c194f000000","rssi":-84,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240015000015001a0029000c194f000000","rssi":-83,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240013000013001a3129000c194f000000","rssi":-80,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000024001300001300193729000b194f000000","rssi":-93,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000024001100001100175b29000a194f000000","rssi":-88}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000024000f00000f00145d280009194f000000","rssi":-84,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000024000e00000e00135c260008194f000000","rssi":-81,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000024000d00000d001159240007194f000000","rssi":-80,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000024000b00000b000e59240006194f000000","rssi":-86}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000024000a00000a000d58240005194f000000","rssi":-93,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240009000009000b58240005194f000000","rssi":-88,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240009000009000a56240004194f000000","rssi":-86}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240008000008000953240004194f000000","rssi":-88,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240007000007000851240003194f000000","rssi":-86}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","rssi":-89,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240004000004000449230002194f000000","rssi":-85,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240003000003000346220001194f000000","rssi":-87,"txpower":0}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa000024000100000100013b1e0000194f000000","rssi":-87}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","manufacturerdata":"eefa0000240001000001000035250000194f000000","rssi":-87,"txpower":0}
```

DigiH, 2022-09-05:
eefa000023001f01001f002000**28**000f194f000000

so this seems to be the temperature, here 28 hex = 40 decimal, and the starting temperature was 21 hex = 33ºC?

May I ask how you install OpenMQTTGateway? Pre-built binaries or do you make your own builds in Paltformio?

savikko, 2022-09-05:
Wow, good find! Starting temperature might actually be 33, should have had one more hand to check everything :)

> May I ask how you install OpenMQTTGateway? Pre-built binaries or do you make your own builds in Paltformio?

Prebuilt binaries, `esp32dev-ble-cont`. I have platformio here installed so can compile as well, but started with the easy way :)

DigiH, 2022-09-05:
eefa 000024 **0015** 00 **0015** **001a00** **29** **000c**    194f000000

so seems to be the same broadcast as when READING the status by connection. With the above being

24 showers taken since this unit has been installed
15 seconds shower time with the last test
2.6 litres water consumed
41ºC temperature
0.12 kW used

savikko, 2022-09-05:
Wow. Thanks. Will try compiling here to get everything ready.

DigiH, 2022-09-05:
Ok, whenever you are ready, change the decoder URL in platformio.ini from

`decoder = https://github.com/theengs/decoder.git#v0.6.2`

to

`decoder = https://github.com/DigiH/decoder#amphiro`

Then you need to rebuild and upload. Let's see what you get then and how it compares to the display on the shower head and/or the app.

savikko, 2022-09-05:
```json
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","rssi":-86,"txpower":0,"brand":"Amphiro","model":"Digital Hand Shower","model_id":"ADHS","sessions":39,"time":6,"litres":0.63125,"tempc":35,"tempf":95,"power":0.02}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","rssi":-81,"txpower":0,"brand":"Amphiro","model":"Digital Hand Shower","model_id":"ADHS","sessions":39,"time":4,"litres":0.427344,"tempc":36,"tempf":96.8,"power":0.01}
{"id":"D8:71:4D:6A:A3:96","mac_type":0,"name":"DHS","rssi":-86,"txpower":0,"brand":"Amphiro","model":"Digital Hand Shower","model_id":"ADHS","sessions":39,"time":2,"litres":0.223828,"tempc":27,"tempf":80.6,"power":0}
```

Me happy. Have to check little more but now looks very good!

savikko, 2022-09-05:
No, thanks to you!

I think `time` is just good this way. I will do few tests and check if there is something to be added there. Water flow (litres per minute) is not there (device provides that on that non-present flow-message) but it is possible to calculate within HA from these values.

Is there something which would prevent this to be discovered automatically by Home Assistant? Is there some specific moment when it would send discovery message? Currently did not see this yet on HA.

Again: thank you.

savikko, 2022-09-05:
One thing to fix:
`power` should be `energy` (as it is kWh).

DigiH, 2022-09-05:
> Water flow (litres per minute) is not there

Yes, that seems to be in the **flow** connection information, but doesn't seem to be freely broadcast in the manufacturerdata, but as you said, this is easily calculated in HA or any other controller.

> `power` should be `energy` (as it is kWh).

Ah ok, so it really is kWh, as the linked info only mentioned kWatts/kW, but yes, it seems to accumulate in your examples above. Wil also need to change the unit then ...

savikko, 2022-09-05:
Got this integrated to HA:

[image]

But anyways, I think after power -> energy change I declare this completed! 🚀

DigiH, 2022-09-05:
Ok, could you rebuild - you might need to Clear/Clear All first so that the decoder URL picks up the new changes I just submitted.

savikko, 2022-09-05:
On it.

savikko, 2022-09-05:
Autodiscovery now brought entities to HA ✔️ 
Power is now energy ✔️ 

For some reason litres and energy do not have units:
[image]

DigiH, 2022-09-05:
Could you just show the OMG messages as well again, to see if it is all with my latest changes?

savikko, 2022-09-05:
Edited last message to previous post.

Here is discovery message for energy:
```json
{
  "stat_t": "+/+/BTtoMQTT/D8714D6AA396",
  "dev_cla": "energy",
  "name": "ADHS-energy",
  "uniq_id": "D8714D6AA396-energy",
  "val_tpl": "{{ value_json.energy | is_defined }}",
  "state_class": "measurement",
  "device": {
    "identifiers": [
      "D8714D6AA396"
    ],
    "connections": [
      [
        "mac",
        "D8714D6AA396"
      ]
    ],
    "manufacturer": "Amphiro",
    "model": "ADHS",
    "name": "Digital Hand Shower",
    "via_device": "OpenMQTTGateway_ESP32_BLE_C"
  }
}
```

And here for time:
```json
{
  "stat_t": "+/+/BTtoMQTT/D8714D6AA396",
  "unit_of_meas": "s",
  "name": "ADHS-time",
  "uniq_id": "D8714D6AA396-time",
  "val_tpl": "{{ value_json.time | is_defined }}",
  "state_class": "measurement",
  "device": {
    "identifiers": [
      "D8714D6AA396"
    ],
    "connections": [
      [
        "mac",
        "D8714D6AA396"
      ]
    ],
    "manufacturer": "Amphiro",
    "model": "ADHS",
    "name": "Digital Hand Shower",
    "via_device": "OpenMQTTGateway_ESP32_BLE_C"
  }
}
```

DigiH, 2022-09-05:
There should definitely be the units "L" and "kWh" as well now. 

Possibly another Clean/Clean All and rebuild and upload ;)

savikko, 2022-09-05:
Will do!

savikko, 2022-09-05:
While this is compiling here, little explanation of this product:
This product is originally developed by Amphiro AG (Switzerland) and they were acquired by Oras (Finnish company).

Original product name was Amphiro Digital Hand Shower but now it is rebranded as Oras Hydractiva Digital.

So, both names probably should be visible, but Oras is current.

DigiH, 2022-09-05:
I was just wondering about that ands put both names into the documentation page, but probably best to then have Oras as the main info in the MQTT messages, and also change the device model to Hydractiva Digital. Thanks

Also looking through OMG MA discovery definitions, energy should definitely be there, but litres might have an issue. Not being a HA user myself I'm not too sure, but am sure @1technophile will have a look into it, if L is still not showing up for you.

savikko, 2022-09-05:
New version is now running. One extra thing: litres probably should be rounded to 2 decimals (have no idea where that rounding should happen).

Units from litres and energy are still missing. 
This has to be somehow related to those device classes. Probably.

Double checked that this compiled with right config:
[image]

Anyways, heavy lifting is now done, thanks to you.

DigiH, 2022-09-05:
> One extra thing: litres probably should be rounded to 2 decimals (have no idea where that rounding should happen).

That would be in the controller, HA in your case.

> Units from litres and energy are still missing.
> This has to be somehow related to those device classes. Probably.

Best to wait and see what the HA discovery pro has to say to it :)

> Anyways, heavy lifting is now done, thanks to you.

And to you, bringing it to our attention and giving the very helpful reverse-engineering link right at the start.

I'll merge this into the Decoder development branch soon, then my personal branch you used so far will be deleted. After the merge you can use the URL

`decoder = https://github.com/theengs/decoder`

in case you need/want to rebuild, until this has propagates into an official release and into OpenMQTTGateway.

1technophile, 2022-09-05:
Nice device!

> Units from litres and energy are still missing.
This has to be somehow related to those device classes. Probably.

Just added those into OMG, see https://github.com/1technophile/OpenMQTTGateway/pull/1265

DigiH, 2022-09-06:
Also just found that in Germany, France, Spain and several other European countries it is available under a different brand and product name.

[HANSA**ACTIVEJET** Digital ](https://www.hansa.com/de/produkte/designlinien/hansaactivejet/handbrause/84310180)

Country selection popup at the top right. Might be good to also include that brand/device name.

savikko, 2022-09-06:
And thats also cheaper in Germany :) 

But yes, those alternative brand names should be added.

@1technophile : I tested your PR and now there are litres, but no kWh. Weird.

DigiH, 2022-09-06:
> And thats also cheaper in Germany :)

Yes, the street price seems to be from around €120. Very tempting, if I didn't like my switchable shower head so much :)

1technophile, 2022-09-07:
> @1technophile : I tested your PR and now there are litres, but no kWh. Weird.

Strange, could you share with me the autodiscovery message published to the broker for kWh ?

savikko, 2022-09-07:
Sure.

Here is for litres:
```json
{
  "stat_t": "+/+/BTtoMQTT/D8714D6AA396",
  "unit_of_meas": "L",
  "name": "ADHS-litres",
  "uniq_id": "D8714D6AA396-litres",
  "val_tpl": "{{ value_json.litres | is_defined }}",
  "state_class": "measurement",
  "device": {
    "identifiers": [
      "D8714D6AA396"
    ],
    "connections": [
      [
        "mac",
        "D8714D6AA396"
      ]
    ],
    "manufacturer": "Oras",
    "model": "ADHS",
    "name": "Hydractiva Digital",
    "via_device": "OpenMQTTGateway_ESP32_BLE_C"
  }
}
```

And here is for energy:
```json
{
  "stat_t": "+/+/BTtoMQTT/D8714D6AA396",
  "dev_cla": "energy",
  "name": "ADHS-energy",
  "uniq_id": "D8714D6AA396-energy",
  "val_tpl": "{{ value_json.energy | is_defined }}",
  "state_class": "measurement",
  "device": {
    "identifiers": [
      "D8714D6AA396"
    ],
    "connections": [
      [
        "mac",
        "D8714D6AA396"
      ]
    ],
    "manufacturer": "Oras",
    "model": "ADHS",
    "name": "Hydractiva Digital",
    "via_device": "OpenMQTTGateway_ESP32_BLE_C"
  }
}
```

1technophile, 2022-09-08:
Thanks, a comma is missing after the unit let me fix this

1technophile, 2022-09-10:
@savikko fixed, should be good now if you can confirm

savikko, 2022-09-10:
What component is actually changed? Will confirm as soon as I know what I should update :)

DigiH, 2022-09-10:
It's been squashed into the same branch

https://github.com/1technophile/OpenMQTTGateway/tree/add-units-for-oras

savikko, 2022-09-10:
Just tested. Works as expected, looks good to me!

```json
{
  "stat_t": "+/+/BTtoMQTT/D8714D6AA396",
  "dev_cla": "energy",
  "unit_of_meas": "kWh", <-- 
  "name": "ADHS-energy",
  "uniq_id": "D8714D6AA396-energy",
  "val_tpl": "{{ value_json.energy | is_defined }}",
  "state_class": "measurement",
  "device": {
    "identifiers": [
      "D8714D6AA396"
    ],
    "connections": [
      [
        "mac",
        "D8714D6AA396"
      ]
    ],
    "manufacturer": "Oras",
    "model": "ADHS",
    "name": "Hydractiva Digital",
    "via_device": "OpenMQTTGateway_ESP32_BLE_C"
  }
}

DigiH, 2022-09-10:
All has been merged. Good to close @savikko

savikko, 2022-09-10:
Thank you @DigiH and @1technophile! Closing as completed.

closed 2022-09-10 by savikko