1technophile/OpenMQTTGateway #1272: Uploading firmware to ESP8266

FerronN, 2022-09-16:
**Describe the bug**
I'm trying to flash the nodemcuv2-ir onto the ESP8266, but it stop after initializing. The LED flashes (erase flash), but after that nothing happens.

**To Reproduce**
[image]

The COM port driver is installed. I uploaded some code with the Arduino IDE to the ESP8266 and that works without any problem.
So I think it's not the device or some driver.

**Expected behavior**
Upload firmware

1technophile, 2022-09-17:
Do you reproduce on several ESP8266 ?

FerronN, 2022-09-17:
No, because I only have one. I now managed to do it with `NodeMCU-PyFlasher`. But I had to change the flash mode to `DOUT`. On the website you can't choose the flash mode, so maybe the site uses `DIO`? I couldn't find the scirpt that's running.

1technophile, 2022-10-10:
Here is the script https://github.com/1technophile/OpenMQTTGateway/blob/development/scripts/gen_wu.py
Not sure we detail the flash mode

1technophile, 2023-02-04:
Closing, feel free if there are other questions

closed 2023-02-04 by 1technophile