1technophile/OpenMQTTGateway #1273: Feature request: publish discovery topics to MQTT regularly or on every reconnect

ssalonen, 2022-09-18:
**Is your feature request related to a problem? Please describe.**

I use openmqttgateway with raspberry pi running mosquitto in volatile/read-only mode. This means that mosquitto and/or restart results in topic contents being lost.

Effectively, after mosquitto/raspberry pi restart the discovery topics are gone.

Since openmqttgateway publishes the discovery topics only once (after restart), the discovery topics will not re-appear in the broker.

**Describe the solution you'd like**

I would like openmqttgateway either 1) to re-publish discovery topics on re-connect, or 2) re-publish discovery topics regularly (configurable interval similar to SYStoMQTT).

This makes openmqttgateway very robust on any topic data loss on broker end.

**Describe alternatives you've considered**

As a workaround, I can trigger remote restart of openmqttgateway using MQTT commands topic.

**Additional context**

The change itself is probably fairly simple. I wanted to open the discussion here whether there is interest towards this change, I can provide PR.

1technophile, 2022-09-18:
Hello,

Thanks for proposing this.
I think 1) could be interesting in your case.

I'm worried that 2) will lead to missing BLE packets due to the wifi use it will generate.

ssalonen, 2022-09-19:
Thanks! 

Would it then make sense if I introduce new configurable parameter
- `MQTT_PUBLISH_DISCOVERY_ON_CONNECT`. Either `true` or `false`. Default `false`.

When `MQTT_PUBLISH_DISCOVERY_ON_CONNECT` is enabled we would publish discovery information right after successful mqtt reconnect. If it is disabled, we would keep the default behavior of publishing on first connect only.