1technophile/OpenMQTTGateway #1278: OMG on ESP32 stops updating BTtoMQTT

ssalonen, 2022-10-08:
**Describe the bug**

Data stops being published to BTtoMQTT thread, after some week(s) of running smoothly.

Other topics such as SYStoMQTT and LWT are still getting updated.

After issuing restart command via MQTT, data starts to flow in BTtoMQTT

**To Reproduce**

TBD 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

ssalonen, 2022-10-08:
I am planning to enable trace logging via mqtt when reflashing the gateways

1technophile, 2022-10-08:
It could be interesting to check the values of `freemem, freestack,uptime` do you have the history of those before the issue?

ssalonen, 2022-10-08:
Unfortunately no… next time it happens I can record the values

ssalonen, 2023-02-02:
I think I have this issue now. Have not restarted yet

Anything off here?

```json
{"uptime":229578,"version":"version_tag","freemem":144592,"mqttport":"1883","mqttsecure":false,"freestack":3228,"rssi":-67,"SSID":"REDACTED","BSSID":"REDACTED","ip":"REDACTED","mac":"REDACTED","lowpowermode":0,"btqblck":0,"btqsum":5894,"btqsnd":5893,"btqavg":1.00017,"interval":0,"scanbcnct":10,"scnct":7409,"modules":["BT"]}
```

1technophile, 2023-02-02:
Seems good in term of values, could you update your version to 1.3.0 indeed

ssalonen, 2023-02-02:
Thanks for the fast response. Will do and report back if this reoccurs

ssalonen, 2023-02-02:
OK updated now, now version is on top of https://github.com/1technophile/OpenMQTTGateway/commit/bd9dcf1ec858c8ad504cd7a154f689373f2cf4a9. Let's see if the issue reoccurs.

1technophile, 2023-02-02:
@h2zero do you have other ideas on this one?

h2zero, 2023-02-04:
Not much to add, it's possible the BLE stack has gotten stuck and stops reporting the scanned devices, need logs to confirm that though.

1technophile, 2023-04-10:
@ssalonen please try the latest development branch. We added a watchdog to BLE.
Feel free to reopen if needed

closed 2023-04-10 by 1technophile