1technophile/OpenMQTTGateway #1285: Upload Page (Documentation) does not work initially

Lerosen, 2022-10-14:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Select a Firmware (ttgo-lora32-v1-868)
2. Click on Install
3. Select a Serial Port in the Dialog
4. Nothing happens

**Expected behavior**
I expect to see some progress screen or confirmation after selecting the serial port. I tried it a couple of times and only after waiting a couple of minutes and trying it again it worked. At least there should be a note on this behaviour.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
The page says `Last Updated: 8/13/2022, 2:32:22 PM` in the bottom.

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2022-10-22:
When selecting the serial port and clicking Connect, could you try to press the boot button or reset of your esp32 ?

Lerosen, 2022-10-22:
Yes, I did. Were you able to reproduce it when not pressing the button?

As described the flashing did work after waiting some time without manually entering flash mode using the button.

1technophile, 2022-10-22:
Not sure if there is a particular behavior with this board, usually, you upload either by pushing or without, but this one seems to work sometime without pressing the PRG button.

closed 2023-10-03 by github-actions[bot]