1technophile/OpenMQTTGateway #1286: ClearGlass CGG1 flashed with ATC sensors recognition

hallard, 2022-10-14:
Since ATC device are working fine, just surprised that my ClearGlass CGG1 flashed wit [ATC](https://github.com/pvvx/ATC_MiThermometer) is not decoded (ATC are), may be it's just a matter of name?

Here what is received for information

```json
{
"id":"58:2D:34:12:33:DC",
"mac_type":0,
"name":"CGG_1233DC",
"rssi":-64,
"servicedata":"582d341233dc00e03e4d0b4e0a",
"servicedatauuid":"0x181a"
}
```

DigiH, 2022-10-14:
Hi @hallard,

thanks for poinying this out. The ClearGlass CGG1 flashed with ATC is aparently not being decoded with the sample you have provided above, even though we already account for two different firmware broadcasting versions for the CGG1.

As the data you provided is very different, both in servicedata length as well as the  service uuid, we can create a third decoding option with your help.

Are you able to create your own OpenMQTTGateway builds in Platformio, so as to be able to try out some test decoder in the next days?

hallard, 2022-10-14:
@DigiH 

Thanks for your quick answer, sure I'm able to do that

Interesting, I switched frame format in [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) from `ATC441` to `Mija MiHome`  frame is different but stil not decoded, strange 

```json

{
"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-49,
"servicedata":"5058480b5cdc3312342d580a10014d",
"servicedatauuid":"0xfe95"
}

{
"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-47,"
servicedata":"5058480b5ddc3312342d580d1004ed005b02",
"servicedatauuid":"0xfe95"
}
```

DigiH, 2022-10-15:
Yes, there seem to be four different advertising formats for the ATC firmware, three applicable to the BLE Decoder library with OpenMQTTGateway as the fourth is encrypted.

To be honest, I'd have to look further into all the different formats, as the currently two supported formats are not specified ;) also not having a CGG1 myself I'll work together with a colleague to hopefully get all the different formats correctly decoded.

hallard, 2022-10-15:
I can help on this (decoding stuff) if you point me out where I need to go, if you have a chance of creating skeleton empty code for these advertisement, I can fill them with decoded values with a PR
just synced my old fork of this repo

hallard, 2022-10-15:
And here the 3 applicable

```shell

charles@mini-m1:explorer$ mosquitto_sub -h 192.168.1.8 -t 'OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233DC' 
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-55,"servicedata":"582d341233dc00e03e490b2c2e","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-61,"servicedata":"582d341233dc00e03e4b0b3c32","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-56,"servicedata":"582d341233dc00e03e4b0b3c35","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-64,"servicedata":"582d341233dc00e03e4b0b3c38","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-63,"servicedata":"582d341233dc00e13e4a0b353b","servicedatauuid":"0x181a"}

charles@mini-m1:explorermosquitto_sub -h 192.168.1.8 -t 'OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233DC' 
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-49,"servicedata":"5058480b5cdc3312342d580a10014d","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-47,"servicedata":"5058480b5ddc3312342d580d1004ed005b02","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-44,"servicedata":"5058480b69dc3312342d580d1004ec005d02","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-44,"servicedata":"5058480b70dc3312342d580a100148","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-43,"servicedata":"5058480b77dc3312342d580d1004eb005e02","servicedatauuid":"0xfe95"}

charles@mini-m1:explorer$ mosquitto_sub -h 192.168.1.8 -t 'OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233DC' 
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-45,"servicedata":"dc3312342d582f09aa173d0b4b7c05","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-47,"servicedata":"dc3312342d582909a1172a0b498305","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-50,"servicedata":"dc3312342d582809ac173c0b4b8605","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-45,"servicedata":"dc3312342d582909c017350b4a8c05","servicedatauuid":"0x181a"}
```

DigiH, 2022-10-15:
And these are the three formats from the pvvx firmware you linked to above.

Thanks, that will give us a good sample along with the format definitions on the pvvx page.

You can have a look at the current two decoder versions included so far at
https://github.com/theengs/decoder/blob/development/src/devices/CGG1_json.h

with a description of the decoder definitions at
https://decoder.theengs.io/participate/adding-decoders.html

hallard, 2022-10-15:
OMG thought it was in raw C code, need to check theengs, looks very cool

DigiH, 2022-10-15:
The decoding has been moved from being directly in OMG in C to the Theengs Decoder library.

Bit of background reading on it at
https://1technophile.blogspot.com/2021/10/externalize-ble-devices-decoding-in.html

> Any on line decoder tester

Not quite yet ;) but any changes to a fork of Decoder can easily be specified in the decoder library URL in platformio.ini for a quick test build.

I won't be able to look at this much further tonight as the bed is desperately calling me, but will have a closer look again tomorrow.

hallard, 2022-10-15:
Thank you very much
Something not clear to me since ATC441 works for LYWSD03MMC_ATC that adverstise also on `UUID 0x181A` for LYWSD03MMC_ATC still don't understand why in `LYWSD03MMC_ATC_json.h` we use service data (in our new case may be this one is different)

changing
```json
 "condition":["servicedata", "index", 0, "a4c138"],
```
by 
```json
 "condition":["uuid", "index", 0, "181A"],
```
should to the trick correct (adding check on size)?

h2zero, 2022-10-15:
The uuid `181A` is a Bluetooth standardized uuid for environmental sensing. The proposed change would detect any device that advertises this uuid as a LYWSD03MMC_ATC device, which will potentially be problematic. We might need a generic handler for this.

hallard, 2022-10-15:
Agree, but since ATC decoding for LYWSD03MMC_ATC is the same than CGG1_ATC (and may be future) it should be noo problem and by the way I think we don't have ultimate solution because today check is done based on 3 bytes of the mac address that would expose same issues than checking uuid.
May be we could combine both and also checking frame size would limit errors. No best idea for now may be tomorrow :-)

h2zero, 2022-10-15:
Yes, the additional check to differentiate would be necessary for this case, and we need to consider the options for it. I just wanted to point that out in case it was not known.

hallard, 2022-10-15:
Take your breath, I'm on the PR :-)

hallard, 2022-10-15:
Ok looks like succeded to PR and do my own build based on the PR of decoder, but looks like I never see anymore my CGG1 device I may have missed something, I copied `ATC` and `PVXX` from `LYWSD03MMC_ATC` and `LYWSD03MMC_PVXX` and changed MAC address to detect, but I may have missed something, I need to take a rest going to sleep.
If you had a chance to look at this [PR](https://github.com/theengs/decoder/pull/216) if you see something wrong just in case

hallard, 2022-10-15:
OMG you won't believe it, sensor was just freezed (too much association during test may be) it does works fine after removing replacing battery (BTW never saw a so stupid battery cover, headache each time)

```shell
charles@mini-m1:decoder$ mosquitto_sub -h 192.168.1.8 -t 'OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233DC' 
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-49,"brand":"Qingping","model":"CGG1","model_id":"CGG1_ATC","tempc":26.3,"tempf":79.34,"hum":55,"batt":72,"volt":2.856}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-50,"brand":"Qingping","model":"CGG1","model_id":"CGG1_ATC","tempc":25.9,"tempf":78.62,"hum":54,"batt":73,"volt":2.858}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-57,"brand":"Qingping","model":"CGG1","model_id":"CGG1_ATC","tempc":25.5,"tempf":77.9,"hum":55,"batt":71,"volt":2.847}
```

DigiH, 2022-10-15:
Hi @hallard,

nicely done with adding the decoders! Just a few thoughts:
I think we should keep all the different format decoders in one decoder file for a clearer overview
As you stated, having the model conditions include part of your device's MAC address obviously was great for a test decoder. For a general decoder we'd need to adjust this to be compatible with everybody's CGG1.

With this thought I published the branch
https://github.com/DigiH/decoder/tree/CGG1

This also required to create a new test case possibility, which shows how important these test cases are for avoiding regressions and testing and clearing new decoders.

I've currently left out the voltage to be able to use the existing common properties, but this can easily be added once we decide this is the way forward.

Please have a look at it and run some real world tests with your CGG1 and the different formats.

Thanks

hallard, 2022-10-15:
Hi @DigiH 

> I think we should keep all the different format decoders in one decoder file for a clearer overview

Yes that was my thought afterward, better like that. Sorry about my 1st time learning when coding onto new project.
By the way instead of naming `_CGG1_json_v2` (it took me some time to understand why) why not using something more friendly such as `_CGG1_json_atc441`, easier to follow when there are more and more, just my 2 cents 

> This also required to create a new test case possibility, which shows how important these test cases are for avoiding regressions and testing and clearing new decoders.

I saw that with auto build error but too lazy to check where was the error, given the time of the night on my side.

> I've currently left out the voltage to be able to use the existing common properties, but this can easily be added once we decide this is the way forward.

Oh yes, would love to see that one, in fact I better trust voltage than percent at least I see real values and depending on devices and battery type it can be a little bit different.

> Please have a look at it and run some real world tests with your CGG1 and the different formats. And there are five different formats now ;) so we probably need to talk through them to clarify each one.

Sure, I will unbox my 2nd one and set it up to PVVX

Thank you for your help

DigiH, 2022-10-15:
> Sorry about my 1st time learning when coding onto new project.

Absolutely no need to apologise, it's great when someone dives into it all anew!

> By the way instead of naming `_CGG1_json_v2` (it took me some time to understand why) why not using something more friendly such as `_CGG1_json_atc441`, easier to follow when there are more and more, just my 2 cents

That is also an option, and equally for the reported model_id naming. For the time being though, until we sort out the decoders and if we really need all of them the v1-5 makes it easier to talk about and look up for any changes.

> Sure, I will unbox my 2nd CGG1 and set it up to PVVX

Great, thanks. If you could also do some testing with the stock firmware that the CGG1s came with, that would be good to verify the default firmware decoder.

Thanks

hallard, 2022-10-15:
> Great, thanks. If you could also do some testing with the stock firmware that the CGG1s came with, that would be good to verify the default firmware decoder.

Thanks for pointing that one out, I will start by this one doing nothing except unbox :-)

DigiH, 2022-10-15:
BTW, would it also be possible for you to install the `esp32dev-ble-datatest` environment/binary for this testing?

We included this to allow to still show all relevant data along with decoded messages.

hallard, 2022-10-15:
> BTW, would it also be possible for you to install the esp32dev-ble-datatest environment/binary for this testing?

Yeah was my idea to dive into that one to have more information to see

hallard, 2022-10-15:
OMG Thought I got 2 I have 3, damned compulsive Ali buy 

[image]

hallard, 2022-10-15:
Ok just to let you know that with the PR you made, stock firmware is not decoded but ATC1441 is fine

```json
{
"id":"58:2D:34:12:33:F7",
"mac_type":0,
"name":"Qingping Temp & RH ",
"rssi":-41,
"servicedata":"3058480b01f73312342d58280100",
"servicedatauuid":"0xfe95"
}
```

hallard, 2022-10-15:
BTW more for @1technophile may be an option to prefix MQTT topic by `model_id` because on my side I got so much devices that's it's a total mess on MQTT Explorer

So something like that 
`OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233F7`

could becomes
`OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/CGG1_ATC1441/582D341233F7`

But may be it already exists need to check

DigiH, 2022-10-15:
> OMG Thought I got 2 I have 3, damned compulsive Ali buy

LOL, yeah, I know that as well, but keep trying to avoid it these days.

> stock firmware is not decoded

> ```json
> "name":"Qingping Temp & RH ",
> …
> servicedata":"3058480b01f73312342d58280100",
> ```

I think we also will have to try and differentiate between the different hardware revisions of the CGG1, between 2020-2021 and 2022+ and their stock firmware, which doesn't make things easier. But correct, so far we do not have any decoder for an 28 long servicedata as described here. Do you know from when this version is? From the Bluetooth 5.0 in the picture I would assume it's one of the newer models. I think this might fall under the v5 decoder, which would need reworking, but I was never sure for exactly which model/firmware it was applicable.
     
>   
> ATC1441 (strange I see volt you already made some changes?) answer is yes just saw your commit
> ```

Yes, and the v2 and v3 `model_id` renaming as well I did before dinner here :)

DigiH, 2022-10-15:
BTW, which firmware did you have installed/refer to with the 

> #MI
> charles@mini-m1:explorermosquitto_sub -h 192.168.1.8 -t 'OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233DC' 
> {"id":"58:2D:34:12:33:DC","mac_type":0…

above? As this should also already be implemented as v3.

1technophile, 2022-10-15:
Hello @hallard 

Yeah, with random beacons plus changing mac addresses, I understand your point.
Note that you can already white list your mac address or ask the gateway to publish recognized beacons only (=detected by the decoder)
https://docs.openmqttgateway.com/use/ble.html#setting-if-the-gateway-publishes-all-the-ble-devices-scanned-or-only-the-detected-sensors

1technophile, 2022-10-15:
@hallard your puck.js could be interesting to integrate

hallard, 2022-10-15:
> your puck.js could be interesting to integrate

For sure but depends on FW on it I know there are classic iBeacon or Edison version what do you need from me to test them?

DigiH, 2022-10-15:
> > your puck.js could be interesting to integrate
> 
> For sure but depends on FW on it I know there are classic iBeacon or Edison version what do you need from me to test them?

Guys, let's stick with the multiple CGG1 versions for now ;) Enough time to get into different firmware versions of the puck.js after that LOL

DigiH, 2022-10-15:
So @1technophile confirmed, his CGG1 2020-2021 with the stick firmware is nicely recognised my our v1 decoder. So your CGG1, @hallard, with BT5 and the stock firmware must be a 2022+ model. Can you view the firmware version of it in the default app just to confirm?

We would then need to add, or amend the v5 decoder to inslude this stock firmware for the newer models.

hallard, 2022-10-15:
> BTW, which firmware did you have installed/refer to with the
> 
> > #MI
> > charles@mini-m1:explorermosquitto_sub -h 192.168.1.8 -t 'OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233DC'
> > {"id":"58:2D:34:12:33:DC","mac_type":0…
> 
> above? As this should also already be implemented as v3.

Your latest built after Clean Clean at 18H17 French time, here are the 3 devices results 

oups does not work anymore did you changed something?

FYI my decoder in platformio.ini is 
`decoder = https://github.com/DigiH/decoder#CGG1`

```json
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-45,"brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":23.5,"tempf":74.3,"hum":58.5,"batt":100}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-68,"servicedata":"582d341233dc00df3f4c0b4ca0","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-51,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-57,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-61,"servicedata":"582d341233dc00e03f4b0b3da6","servicedatauuid":"0x181a"}
```

hallard, 2022-10-15:
Oh my bad, should have waited looks like got good frames now

```json
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-45,"brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":23.5,"tempf":74.3,"hum":58.5,"batt":100}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-68,"servicedata":"582d341233dc00df3f4c0b4ca0","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-51,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-57,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-61,"servicedata":"582d341233dc00e03f4b0b3da6","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-59,"brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.4,"tempf":72.32,"hum":63,"batt":74,"volt":2.87}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-53,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-52,"brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":23.5,"tempf":74.3,"hum":58.5,"batt":100}

```

DigiH, 2022-10-15:
> ```json
> {"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-51,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
> {"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-57,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
> ```

Apart from these, which are not caught yet at all, and @1technophile seeing the same with his CGG!. I think it is the different name braodcast, compared to your above #MI example, as otherwie they seem to be the same.

DigiH, 2022-10-15:
Just updated the decoder.

Could you Clean and reBuild please and Upload, to verify if the last undecoded occurrences are now decoded?

DigiH, 2022-10-15:
Ok, the fact that @hallard's CGG1 with stock firmware behaves differently to @1technophile's CGG1 with stock firmware, possibly due to different version due to the hardware revisions, plus the fact of:

https://github.com/pvvx/ATC_MiThermometer/issues/41#issuecomment-812803456

being either the true or fake **FCC-ID 2AQ3F-CGG1**, makes it even harder ;)

I think we need to, step by step, confirm/eliminate the different decoder versions, ideally starting with the v2 (ATC1441) and v3 (PVVX) version, which are most likely to be the most consistent formats.

So white-listed message logs of both these versions for a couple of minutes should show all the possible broadcast data, to hopefully allow these two decoders to be confirmed and GM'd.

Then several minutes of the stock firmware for @1technophile's device, plus all three of @hallard's CGG1s with `esp32dev-ble-datatest` message logging should show up any deficiencies in the remaining decoders, and/or if it is only due to the ClearGrass/Qingping protocol/serviceuuid differences, and enough decoded messages come through with only either.

The above mentioned
`{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-61,"servicedata":"582d341233dc00e03f4b0b3da6","servicedatauuid":"0x181a"}`
will likely remain undecoded, as there doesn't seem to be a name broadcast with it, and with the universal 181a uuid and only the data length it would be a bit to ambiguous as a model condition.

All a bit cumbersome, but I think this is the only way we will get nicely formatted properly decoded messages - for all existing test devices at least.

Comments or suggestions on this? :)

hallard, 2022-10-16:
@DigiH 

> BTW @hallard you didn't seem to like the esp32dev-ble-datatest environment, as the decoded messages do not have the raw servicedata in them ;)

Of course not, I like it, in fact I was in a hurry saturday and didn't find it "as I would" because on VScode+PIO, targets are not sorted by name (my bad) so now just rebuild from scratch.

here are results, I need to check something the PVVX version looks like to be more verbose they are 3 at the same place with same config that is strange.

As a reminder
```
58:2D:34:12:3C:E8 is stock FW
58:2D:34:12:33:DC is ATC
58:2D:34:12:33:F7 is PVVX

```

logs

```json
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-52,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-56,"brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.8,"tempf":73.04,"hum":66,"batt":74,"volt":2.873}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-59,"servicedata":"582d341233dc00e4424d0b4ed1","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.8,"tempf":73.04,"hum":66,"batt":77,"volt":2.894}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-53,"servicedata":"8816e83c12342d580104d800b102020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":21.6,"tempf":70.88,"hum":68.9,"batt":100}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-55,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-54,"servicedata":"8816e83c12342d580104e200a102020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22.6,"tempf":72.68,"hum":67.3,"batt":100}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-60,"servicedata":"582d341233dc00e4424a0b36d8","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-56,"servicedata":"582d341233dc00e4424a0b37db","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.8,"tempf":73.04,"hum":66,"batt":74,"volt":2.871}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-55,"servicedata":"8816e83c12342d580104df00a502020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22.3,"tempf":72.14,"hum":67.7,"batt":100}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-51,"servicedata":"f73312342d58d709c917290b490205","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":25.19,"tempf":77.342,"hum":60.89,"batt":73,"volt":2.857}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-54,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-55,"servicedata":"582d341233dc00e442490b2ee5","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.8,"tempf":73.04,"hum":66,"batt":73,"volt":2.862}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-56,"servicedata":"8816e83c12342d580104dc00ad02020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22,"tempf":71.6,"hum":68.5,"batt":100}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-58,"servicedata":"582d341233dc00e5424b0b3fec","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.9,"tempf":73.22,"hum":66,"batt":75,"volt":2.879}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"rssi":-69,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-55,"servicedata":"582d341233dc00e4424d0b50ef","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.8,"tempf":73.04,"hum":66,"batt":77,"volt":2.896}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-54,"servicedata":"f73312342d58910937182c0b491205","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":24.49,"tempf":76.082,"hum":61.99,"batt":73,"volt":2.86}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"rssi":-67,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-55,"servicedata":"582d341233dc00e5424d0b50f2","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.9,"tempf":73.22,"hum":66,"batt":77,"volt":2.896}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-54,"servicedata":"8816e83c12342d580104dc00ad02020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22,"tempf":71.6,"hum":68.5,"batt":100}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-56,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
[... cut ...]
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-60,"servicedata":"582d341233dc00e6424a0b351c","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":23,"tempf":73.4,"hum":66,"batt":74,"volt":2.869}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-66,"servicedata":"8816e83c12342d580104dc00ad02020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22,"tempf":71.6,"hum":68.5,"batt":100}

```

Hope this helps

DigiH, 2022-10-16:
Great! Thanks a lot @hallard , this helps a lot!

No worries about the sorting, was quickly done here, and also showed me that for PVVX there are only three messages, whereas the stock firmware and ATC have between 13, respectively 18 messages. I take it they were all taken during the same time period, initially unsorted as above above as they came in. But could have to do something with the individual firmware reporting setting.

I'll tale a closer look and will amend the test decoder accordingly. Would be great if you could then give it another test run in the next days with all three devices with the firmware they have now :)

hallard, 2022-10-16:
@DigiH 

No problem I will leave them as is for few days, I've updated the logs above with more data

hallard, 2022-10-17:
Here few more with some not decoded 

```json
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-77,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-69,"servicedata":"582d341233dc00e3424d0b4ea5","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-69,"servicedata":"582d341233dc00e341490b29a8","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-70,"servicedata":"582d341233dc00e3424d0b4fab","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-68,"servicedata":"f73312342d58de08cd197e0b52f405","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"rssi":-73,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-69,"servicedata":"582d341233dc00e3424d0b50b2","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.7,"tempf":72.86,"hum":66,"batt":77,"volt":2.896}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-68,"servicedata":"f73312342d58e008cf198d0b54f605","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-69,"servicedata":"582d341233dc00e2424b0b3eb5","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-68,"servicedata":"f73312342d58e008e1198d0b54fa05","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-67,"servicedata":"f73312342d58de08db197e0b52fc05","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.7,"tempf":72.86,"hum":66.19,"batt":82,"volt":2.942}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-78,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-68,"servicedata":"f73312342d58db08f419770b510e05","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.67,"tempf":72.806,"hum":66.44,"batt":81,"volt":2.935}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-69,"servicedata":"582d341233dc00e3424d0b4fd5","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"rssi":-73,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-68,"servicedata":"f73312342d58df08ee198f0b541005","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"rssi":-71,"servicedata":"8816e83c12342d580104e3008602020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22.7,"tempf":72.86,"hum":64.6,"batt":100}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-69,"servicedata":"f73312342d58df08f519890b531205","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-69,"servicedata":"582d341233dc00e242490b2edc","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-68,"servicedata":"f73312342d58df08e619780b511405","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.71,"tempf":72.878,"hum":66.3,"batt":81,"volt":2.936}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-81,"servicedata":"8816e83c12342d580104e3008602020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22.7,"tempf":72.86,"hum":64.6,"batt":100}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-82,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-69,"servicedata":"582d341233dc00e342480b25e3","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-69,"servicedata":"f73312342d58d808fc19780b511705","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-73,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-69,"servicedata":"582d341233dc00e3424d0b4fec","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.7,"tempf":72.86,"hum":66,"batt":77,"volt":2.895}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-68,"servicedata":"f73312342d58da08fa198d0b541d05","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.66,"tempf":72.788,"hum":66.5,"batt":84,"volt":2.957}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-73,"servicedata":"8816e83c12342d580104e3009102020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22.7,"tempf":72.86,"hum":65.7,"batt":100}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-68,"servicedata":"f73312342d58dd08fc19800b521f05","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.69,"tempf":72.842,"hum":66.52,"batt":82,"volt":2.944}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-69,"servicedata":"582d341233dc00e3424d0b53f0","servicedatauuid":"0x181a"}
[... cut ...]
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-69,"servicedata":"f73312342d58da08fe19780b512405","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.66,"tempf":72.788,"hum":66.54,"batt":81,"volt":2.936}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"rssi":-80,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-69,"servicedata":"582d341233dc00e3424d0b4efa","servicedatauuid":"0x181a"}
```

DigiH, 2022-10-17:
Thanks @hallard,

I think we can safely say now that CGG1 decoder v2 (CGG1_ATC1441) and v3 (CGG1_PVVX) are working as expected, and are good for final release as they are.
The only undecoded messages from these two are when no name is recorded in a few transmissions every now and then, which we wouldn't want to do, due to the generic nature of `"servicedatauuid":"0x181a"`.

This leaves us with your undecoded shorter messages for v1 (stock firmware) along with the properly decoded messages.

Looking at these - undecoded with `"servicedatauuid":"0xfe95"` - the servicedata never changes in all your samples, in the middle there is the MAC address in reverse, but with no changes and no obvious similar temp/hum encodings as with the decoded `"servicedatauuid":"0xfdcd" ` I think these are non-relevant broadcasts to our decoding purpose.

With the decoded `"servicedatauuid":"0xfdcd" ` messages I still have one issue/question though. Is you battery really always at 100% in the `"58:2D:34:12:3C:E8"`device? 

Would you mind trying with the battery from one of the other devices, just to test my theory that the stock firmware actually doesn't send proper battery level information?

Once we have this sorted out I would say that the stock firmware, ATC1441 and PVVX are ok - at least for your devices, and thanks to your help and extensive testing!

hallard, 2022-10-17:
@DigiH the `58:2D:34:12:3C:E8` is the latest I used, just outboxed 2 days ago like the `58:2D:34:12:33:F7` at the same time and both ordered may be one year ago.

So yo need me to swap battery between `58:2D:34:12:3C:E8` and `58:2D:34:12:33:F7` correct? If so I will do that once at home.

DigiH, 2022-10-17:
> So yo need me to swap battery between `58:2D:34:12:3C:E8` and `58:2D:34:12:33:F7` correct? If so I will do that once at home.

That would be great, or the battery from `58:2D:34:12:33:DC` into `58:2D:34:12:3C:E8` - either one will do to see if `58:2D:34:12:3C:E8` with the stock firmware is still reporting 100% or if it has changed to more what ATC1441 and PVVX were reporting with the same battery.

Thanks

hallard, 2022-10-17:
> That would be great, or the battery from 58:2D:34:12:33:DC into 58:2D:34:12:3C:E8 

Here we go

```json
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-71,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-71,"servicedata":"f73312342d58eb08161b770b514005","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.83,"tempf":73.094,"hum":69.34,"batt":81,"volt":2.935}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-77,"servicedata":"582d341233dc00ec445c0bd8a1","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":23.6,"tempf":74.48,"hum":68,"batt":92,"volt":3.032}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-74,"servicedata":"8816e83c12342d580104e800a50202015a","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":23.2,"tempf":73.76,"hum":67.7,"batt":90}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-79,"servicedata":"582d341233dc00e9435c0bdca4","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":23.3,"tempf":73.94,"hum":67,"batt":92,"volt":3.036}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-77,"servicedata":"f73312342d58e808ff1a7c0b524405","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.8,"tempf":73.04,"hum":69.11,"batt":82,"volt":2.94}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-67,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-79,"servicedata":"582d341233dc00e8445c0bd8aa","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":23.2,"tempf":73.76,"hum":68,"batt":92,"volt":3.032}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-71,"servicedata":"f73312342d58e508081b950b554605","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.77,"tempf":72.986,"hum":69.2,"batt":85,"volt":2.965}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-75,"servicedata":"582d341233dc00e7445c0bd8ae","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":23.1,"tempf":73.58,"hum":68,"batt":92,"volt":3.032}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-70,"servicedata":"f73312342d58e608041b950b554805","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-72,"servicedata":"8816e83c12342d580104e800a50202015a","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":23.2,"tempf":73.76,"hum":67.7,"batt":90}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-72,"servicedata":"f73312342d58e708f91a960b554b05","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-78,"servicedata":"582d341233dc00e6445b0bcfb4","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":23,"tempf":73.4,"hum":68,"batt":91,"volt":3.023}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-67,"servicedata":"8816e83c12342d580104e500a30202015a","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22.9,"tempf":73.22,"hum":67.5,"batt":90}
{"id":"58:2D:34:12:33:F7","mac_type":0,"rssi":-69,"servicedata":"f73312342d58e608ef1a920b544d05","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-71,"servicedata":"f73312342d58e808e91a960b554f05","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.8,"tempf":73.04,"hum":68.89,"batt":85,"volt":2.966}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-77,"servicedata":"582d341233dc00e6445b0bcfba","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":23,"tempf":73.4,"hum":68,"batt":91,"volt":3.023}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-67,"servicedata":"8816e83c12342d580104e500a30202015a","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22.9,"tempf":73.22,"hum":67.5,"batt":90}
{"id":"58:2D:34:12:33:F7","mac_type":0,"name":"CGG_1233F7","rssi":-70,"servicedata":"f73312342d58e708e51a860b535105","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v3","model_id":"CGG1_PVVX","tempc":22.79,"tempf":73.022,"hum":68.85,"batt":83,"volt":2.95}
{"id":"58:2D:34:12:33:DC","mac_type":0,"rssi":-76,"servicedata":"582d341233dc00e5445d0bddbe","servicedatauuid":"0x181a"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"rssi":-71,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-65,"servicedata":"3058480b01e83c12342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:33:DC","mac_type":0,"name":"CGG_1233DC","rssi":-81,"servicedata":"582d341233dc00e4445c0bd8c8","servicedatauuid":"0x181a","brand":"Cleargrass/Qingping","model":"Round TH v2","model_id":"CGG1_ATC1441","tempc":22.8,"tempf":73.04,"hum":68,"batt":92,"volt":3.032}
{"id":"58:2D:34:12:3C:E8","mac_type":0,"name":"Qingping Temp & RH M","rssi":-67,"servicedata":"8816e83c12342d580104e500a30202015a","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH v1","model_id":"CGG1","tempc":22.9,"tempf":73.22,"hum":67.5,"batt":90}
```

DigiH, 2022-10-18:
Great, thanks a lot @hallard!

So newer CGG1s work fine with the stock firmware reporting the battery level, even though it seem it might only be in full 10% steps.

I have uploaded the changes to the STOCK, ATC1441 and PVVX decoders as we can view them as final now. 

I will have to look a bit more into why some old version with the stock firmware seem to report a never changing battery level, and pull some of the current v4/v5 together, but for the new CGG1 hardware versions all looks fine now, thanks to your help and very extensive testing and reporting.

You can go back to installing your preferred firmware on all your CGG1 again 👍  and if you want to rebuild OMG with the latest above mentioned changes please make sure to CLEAN/CLEAN ALL before a new build and upload with the current test decoder.

You might also want to open an issue in the Theengs Decoder repo, submit a PR or start a discussion there, about the inclusion of your **Puck.js** :grinning:

hallard, 2022-10-18:
> Great, thanks a lot @hallard!

your welcome, it will be cool to have them on my HA system now with OpenMQTTGateway

> You can go back to installing your preferred firmware on all your CGG1 again 👍 and if you want to rebuild OMG with the latest above mentioned changes please make sure to CLEAN/CLEAN ALL before a new build and upload with the current test decoder.

Yes I will use ATC, PVVX does fancy temperature on my homey plugins (not decoded I think)

> You might also want to open an issue in the Theengs Decoder repo, submit a PR or start a discussion there, about the inclusion of your Puck.js 😀

Yeah, need to find some spare time for that :-)

DigiH, 2022-10-18:
> PVVX does fancy temperature on my homey plugins (not decoded I think)

Just out of curiosity, doesn't Homey allow to receive MQTT data from OMG for example? Or are you talking about Homey (Bridge) directly receiving the BLE broadcasts - so a Theengs app/plug-in using the same decoders as OMG might be a preferred  option?

hallard, 2022-10-18:
> Just out of curiosity, doesn't Homey allow to receive MQTT data from OMG for example? 

Absolutely it can do that, but I need then to fire a flow when data arrive from MQTT to update virtual sensor values and as now I can't get raw data value in topic (need to parse json that can't be done for virtual device easily ) it's by far more complicated [see](https://community.openmqttgateway.com/t/vote-removing-simplereceiving-and-simplepublishing/1044/3)

> Or are you talking about Homey (Bridge) directly receiving the BLE broadcasts - so a Theengs app/plug-in using the same decoders as OMG might be a preferred option?

Nope it's a classic homey (old one) with this [plugins](https://github.com/Zsoltt-dom/MiTempHum-Homey) that works fine also but only ATC (and PVVX is wrong may be because triggered only with the classic uuid `0x181a`

DigiH, 2022-10-18:
Ok, I have to admit I don't know much and haven't got any experience with Homey, but recently saw the very interesting looking Homey Pro announcement.

> as now I can't get raw data value in topic (need to parse json that can't be done for virtual device easily ) it's by far more complicated [see](https://community.openmqttgateway.com/t/vote-removing-simplereceiving-and-simplepublishing/1044/3)

`simplePublishing` is still active in OMG, so you should be ok with receiving the simple individual messages, or are you still having issues with it?

https://github.com/1technophile/OpenMQTTGateway/blob/development/main/main.ino#L340

hallard, 2022-10-18:
> Ok, I have to admit I don't know much and haven't got any experience with Homey, but recently saw the very interesting looking Homey Pro announcement.

Trust me I used a lot (Fibaro, Eeedomus, Jeedom, ...) homey is the one I kept, the UX is so nice and good and it's so easy to make flow thru the smartphone no need to do any coding or script, that the goal isnt' it, place and use without any "oh something wrong let me fix it", it's just works, and fine. Worth it so much and WAF compliant :-)

> simplePublishing is still active in OMG, so you should be ok with receiving the simple individual messages, or are you still having issues with it?

yes exactly this one that I tried, didn't not succeeded to get it to work, maybe I missed something if it's still active it should work, I'll try agin tonight.

DigiH, 2022-10-18:
> yes exactly this one that I tried, didn't not succeeded to get it to work, maybe I missed something if it's still active it should work, I'll try agin tonight.

I just quickly tested it here with my BLE gateway, just added

`'-DsimplePublishing=true'`

to my `build_flags`, rebuilt, uploaded and then all the devices' messages were split into each individual property. Best monitored with some app like [MQTT-Explorer](https://github.com/thomasnordquist/MQTT-Explorer/releases/tag/v0.3.5).

Obviously all topics now get extended by each property name, so possibly that threw your Homey reception?

hallard, 2022-10-18:
I tried with `mosquitto_sub`, did not looked into MQTT Explorer that was running in background BTW because I was sure of my subscription (my bad)

```shell
mosquitto_sub -h 192.168.1.8 -t 'OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233DC' -t 'OMG/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/582D341233DC/#'
```

I'll try back tonight, that's perfect if it is working and I need to whitelist also

hallard, 2022-10-18:
Ok I tried back with M5StackCP plus but unable to get it work and thus far from stable this board, looks like I'm unable to flash it back, no time to investigate so I tried back to my ThingPulse ESP32 gateway rebuilding all, does works fine, sorry about that.

[image]

closed 2022-10-19 by hallard

hallard, 2022-11-05:
Hi There,

Received today 2 new Clear glass, one is the H version (Apple Home Kit) the most expensive and this one has no battery provided 

[image]

The other is classic CCG1 (with battery provided)

[image]

Stock firmware plugged the battery, got 2 OMG running one with debug message the other not, don't be surprised

```
json

{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0001000102868c3a3e","rssi":-65}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qin","manufacturerdata":"4c000631012bae0af558540a0001000102868c3a3e","rssi":-73}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0001000102868c3a3e","rssi":-64}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-67,"servicedata":"3058480b014e3112342d58280100","servicedatauuid":"0xfe95"}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0001000102868c3a3e","rssi":-72}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-64}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-83,"servicedata":"3058480b014e3112342d58280100","servicedatauuid":"0xfe95"}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-61}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-66,"servicedata":"08164e3112342d580104df00f001020164","servicedatauuid":"0xfdcd","brand":"Cleargrass/Qingping","model":"Round TH","model_id":"CGG1","tempc":22.3,"tempf":72.14,"hum":49.6,"batt":100}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-72}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-64}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-63}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-65,"servicedata":"3058480b014e3112342d58280100","servicedatauuid":"0xfe95"}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qin","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-74}
{"id":"58:2D:34:12:31:4E","mac_type":0,"rssi":-85,"brand":"Cleargrass/Qingping","model":"Round TH","model_id":"CGG1","tempc":22.3,"tempf":72.14,"hum":49.6,"batt":100}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-65,"servicedata":"3058480b014e3112342d58280100","servicedatauuid":"0xfe95"}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-74}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-71,"servicedata":"3058480b014e3112342d58280100","servicedatauuid":"0xfe95"}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-63}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-67,"brand":"Cleargrass/Qingping","model":"Round TH","model_id":"CGG1","tempc":22.3,"tempf":72.14,"hum":49.6,"batt":100}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-63}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-66,"servicedata":"3058480b014e3112342d58280100","servicedatauuid":"0xfe95"}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-66,"servicedata":"3058480b014e3112342d58280100","servicedatauuid":"0xfe95"}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qin","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-73}
{"id":"F8:1B:54:2B:DC:79","mac_type":1,"name":"Qingping Temp RH H","manufacturerdata":"4c000631012bae0af558540a0002000102868c3a3e","rssi":-58}
{"id":"58:2D:34:12:31:4E","mac_type":0,"name":"Qingping Temp & RH M","rssi":-84,"servicedata":"3058480b014e3112342d58280100","servicedatauuid":"0xfe95"}
```

Only the classic is decoded `58:2D:34:12:31:4E`

hallard, 2022-11-05:
> If not specifically bought for direct HomeKit compatibility, but mainly for using with OMG, Decoder or other similar BLE scanning software, not worth the extra expense.

Absolutely, but I wanted to give it a try and was pretty sure you would be interested to see frame :-)

hallard, 2022-11-05:
Worked ROOB on my iPhone

[image]

DigiH, 2022-11-05:
Hi @hallard , could you do some more logging of the CGG1H? Also with making sure that there will be different temperature reading, like putting it outside for a bit, holding it in your hands to warm up … also some manufacturerdata broadcasts while you get the correct temp and humidity from the app at the same time …  as currently I'm not seeing any obvious encoded data.

Thanks