1technophile/OpenMQTTGateway #1288: Switchbot detection

xanderificnl, 2022-10-16:
N: Scan begin
N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf850f049243","rssi":-63}
N: Found 0 devices, scan number 61 end
N: Scan begin
N: Found 1 devices, scan number N62 en: d
Device detected: C6:C7:52:66:3C:BF

**Describe the bug**
OpenMQTTGateway isn't recognizing my Switchbot Meter Plus properly. 

**To Reproduce**
- Flash several different ESP32 

**Expected behavior**
For it to recognize my Switchbot Meter Plus.

**Environment (please complete the following information):**
- esp32dev-ble from binary (site)
- esp32dev-ble-cont from binary (site)
- esp32dev-ble self-built

latest versions

**Additional context**

Pinging @DigiH / issue #20 

The switchbot is C6:C7:52:66:3C:BF

In essense the monitor just says this:

```
Device detected: C6:C7:52:66:3C:BF
N: Scan begin
N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf850f049243","rssi":-58}
N: Found 0 devices, scan number 373 end
```

and when I press the bluetooth button on the meter:

```
N: Device detected: C6:C7:52:66:3C:BF
N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf880b06924c","rssi":-32,"servicedata":"4980e406924c","servicedatauuid":"0xfd3d"}
N: Found 1 devices, scan number 441 end

N: Device detected: 5E:3C:86:96:DD:A7
N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf8d0f099242","rssi":-37}
N: Send on /BTtoMQTT/5E3C8696DDA7 msg {"id":"5E:3C:86:96:DD:A7","mac_type":1,"manufacturerdata":"4c0010055f1c12f757","rssi":-60,"txpower":12}
N: Found 0 devices, scan number 527 end

-snip-

N: Device detected: C6:C7:52:66:3C:BF
N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf8e0f099245","rssi":-26,"servicedata":"49c0e4099245","servicedatauuid":"0xfd3d"}
N: Found 1 devices, scan number 590 end

N: Scan begin
N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf8f0f099242","rssi":-25}
N: Found 0 devices, scan number 612 end
```

```
N: Send on /SYStoMQTT msg {"uptime":700,"version":"v0.9.16","freemem":134236,"mqttport":"1883","mqttsecure":false,"freestack":4780,"rssi":-37,"SSID":"XiFi","BSSID":"08:55:31:D6:79:49","ip":"192.168.88.231","mac":"34:86:5D:FD:4F:08","lowpowermode":0,"btqblck":0,"btqsum":382,"btqsnd":330,"btqavg":1.157576,"interval":0,"scanbcnct":10,"scnct":592,"modules":["BT"]}
```
On MQTT:

```
# home/OpenMQTTGateway_ESP32_BLE_C/BTtoMQTT/C6C752663CBF
{
  "id": "C6:C7:52:66:3C:BF",
  "mac_type": 1,
  "manufacturerdata": "6909c6c752663cbf8d0f099242",
  "rssi": -36,
  "servicedata": "49c0e4099242",
  "servicedatauuid": "0xfd3d"
}
```

xanderificnl, 2022-10-16:
Here we go, the first actual scan (I saw) with data:

```
N: Device detected: 71:3A:F0:36:95:D0
N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-25,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":19,"tempf":66.2,"hum":66,"batt":100}
N: Send on /BTtoMQTT/713AF03695D0 msg {"id":"71:3A:F0:36:95:D0","mac_type":1,"manufacturerdata":"4c0010056e1cc98704","rssi":-91,"txpower":8}
N: Found 0 devices, scan number 802 end
```

xanderificnl, 2022-10-16:
Might the issue be that I assumed the meter readings should stay whereas they are only advertised when announced by the switchbot?

DigiH, 2022-10-16:
If you

xanderificnl, 2022-10-16:
It was a pebkac (problem exists between keyboard and computer). Clearly, my assumptions made me blind for what was right in front of me, i.e. this project merely mirroring signals from bluetooth to MQTT.

I do apologize for wasting your time @DigiH. And I do want to express my gratitude to all maintainers and developers for making an awesome project. Hopefully not all users are so blind as me haha.

Can we close this?

DigiH, 2022-10-16:
Please see my message above, while you might have mistaken a few things a bit, you are very correct in stating that not all the messages are decoded which could/should be decoded, due to the fast that your SwitchBot Meter might be a very recent purchase with a slightly newer format which isn't recognised by the Decoder yet.

So thanks for pointing this out!

xanderificnl, 2022-10-16:
Thanks. I feel a tad less stupid now!

So the servicedata was a pretty common occurence, and while I looked through your code last night - I did pick up on it containing the readings. That's why I figured there was something wrong, I also tried compiling with your decoder library but that made no difference.

I'm guessing the "add" mode is when I'm pressing the button? For what it's worth, the other project* has been running since yesterday and has been updating my broker very regularly. 

What can I do to help with the new format? 

\* https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32

DigiH, 2022-10-16:
> What can I do to help with the new format?

If you are compiling your builds with Platformio, you can just change the Decoder URL in the platformio.ini file from

`decoder = https://github.com/theengs/decoder.git#v0.6.2`

to

`decoder = https://github.com/DigiH/decoder#SBMT-test`

to test the decoder changes I just uploaded.

You should not see any undecoded messages now which include "`servicedata"`, and as you said, they were quite a few of those before, they should now all be properly decoded messages, and therefore more frequent.

xanderificnl, 2022-10-16:
Adding the device ID _has worked_. Here's what's happening now:

- Turn on the ESP32
- The switchbot gets picked up automatically
- It'll get picked up everyonce in awhile - a quick glance at MQTT explorer _suggests_ it picks up sensor readings at least once every 30 seconds - sometimes (such as shown below) quicker. 

[image]

This is sweet! I _think_ this resolves the issue and now I have _every excuse I need_ to get working on OMG <-> Homie integration ;-)

Thank you @DigiH, much appreciated and such speed!

DigiH, 2022-10-16:
> This is sweet! I _think_ this resolves the issue and now I have _every excuse I need_ to get working on OMG <-> Homie integration ;-)

Great, I'll will be looking forward to your Homie integration!

> Thank you @DigiH, much appreciated and such speed!

Thank you for pointing this out, and helping along to pinpoint and verify this different ID! As this is not documented on the official SwitchBot API pages (yet).

I have merged this into the official Theengs Decoder development branch, and will delete my test branch soon. So if you have to rebuild for any reason please use

decoder = https://github.com/theengs/decoder

for any future builds.

Thanks again!

xanderificnl, 2022-10-16:
> > This is sweet! I _think_ this resolves the issue and now I have _every excuse I need_ to get working on OMG <-> Homie integration ;-)
> 
> Great, I'll will be looking forward to your Homie integration!

I'm almost looking forward to fighting with the compiler! I do love the language but man, compilers and I never C eye to eye ;)

> 
> > Thank you @DigiH, much appreciated and such speed!
> 
> Thank you for pointing this out, and helping along to pinpoint and verify this different ID! As this is not documented on the official SwitchBot API pages (yet).

You're most welcome!  

> 
> I have merged this into the official Theengs Decoder development branch, and will delete my test branch soon. So if you have to rebuild for any reason please use
> 
> decoder = https://github.com/theengs/decoder
> 
> for any future builds.

Just rebuilded to switch back from *-cont to esp32-dev-ble and used that repository.

> 
> Thanks again!
> 
> Please also close this OMG issue, if you're happy with the changes. Thanks

I am happy with the solution. I'm going to put on some monitoring and have it run over night, see how many times it published sensor data - and hopefully this can be the start of a beautiful array of BLE devices broadcasting sensors for OMG to pick up.

Cheers!

closed 2022-10-16 by xanderificnl

xanderificnl, 2022-10-21:
After @DigiH fixed the issue, I recompiled to esp32-dev-ble (verified it used the fix) and had two ESP32's running side by side.

One with OMG and another with [SwitchBot-MQTT-BLE-ESP32](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32). Both with the defaults + WiFi configuration. 

I monitored them via a tmux-background window and _mosquitto-sub_, i.e.:

```
mosquitto_sub -h 172.16.32.2 -t home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/C6C752663CBF | ax -o json | tee log/openmqttgateway.log
mosquitto_sub -h 172.16.32.2 -t switchbot/switchbot-relay/meter/one/attributes | ax -o json | tee log/switcbot-mqtt-ble-esp32.log 
```

The log files are here: [switcbot-mqtt-ble-esp32.log](https://github.com/1technophile/OpenMQTTGateway/files/9841795/switcbot-mqtt-ble-esp32.log) and  [openmqttgateway.log](https://github.com/1technophile/OpenMQTTGateway/files/9841798/openmqttgateway.log)

Here are some interesting bits:

```
$ wc -l ./openmqttgateway.log 
5812 ./openmqttgateway.log
$ grep 'tempc' ./openmqttgateway.log | wc -l
170
$ wc -l ./switcbot-mqtt-ble-esp32.log
897 ./switcbot-mqtt-ble-esp32.log
```

In short: 

- OMG picked up the Switchbot a staggering amount of times but unfortunately only reported sensor data about 170 times;
- Switchbot-mqtt reported 897 times sensor data. It should be noted it's status on MQTT is 'passive scanning' and it does have inherit knowledge of the Switchbot's mac address.

_I'm going to restart the *-cont version to see how often it'll report sensor data. I do recall it showing 10+ sensor readings in a small window_

@DigiH can you advise? I'm guessing 170 sensor readings over 5 days is very low. Am I missing something?

DigiH, 2022-10-22:
Hi @xanderificnl 

> I'm guessing 170 sensor readings over 5 days is very low. Am I missing something?

Yes, absolutely, and great that you made these logs. The **openmqttgateway.log** clearly shows that most of the messages received from the SwitchBot Meter have **manufacturerdata** only in them, missing the **servicedata** required for Theengs Decpder decoding.

> I'm going to restart the *-cont version to see how often it'll report sensor data.

Did you see any changes with the *-cont version, more specifically, was more data with **servicedata** being picked up and was therefor decoded?

I just looked at the official SwitchBot BLE API again and realised that there are two formats for the Meters:

(Old) Broadcast Message - Manufacturer data
(New) Broadcast Message - Service data

https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/blob/latest/devicetypes/meter.md#meter-broadcast-message-format

If and why your Meter seems to send out both formats, and if the manufacturerdata contains all the information - I'll have to have a closer look, only saw that the last byte looks like being the humidity.

Which firmware version does your Meter have? Might there be a firmware update for it in the SwitchBot app?

Any chance of resetting your Meter, to see how it behaves after a default resetting/new start?

Generally I'd expect the Meter to be picked up roughly every minutes, as it already is now with the default OMG BLE binary,  but with it transmitting data which can/is being decoded, so correct data from the Meter every minute.

DigiH, 2022-10-22:
As I don't have a SwitchBot Meter myself I just checked with a colleague, and his logging shows the opposite of yours. In general the messages received by his SwitchBot Meter do contain **both** the **manufacturerdata** and **servicedata**, with the **servicedata** correctly being decoded and the wanted information included in the messages. 

Only now and then his Meter messages also only contain the **manufacturerdata** and are not being decoded, which unfortunately is the majority with your device.

That, together with your initial **49** vs. **69** ID issue makes me wonder if it is due to an earlier firmware or also possibly adjustable with a device reset.

Could you also try logging with the **esp32dev-ble-datatest** environment/binary, so that we also get the raw data included in the decoded messages.

Thanks

xanderificnl, 2022-10-22:
Hi.

Hopefully I haven't missed answering a question.

1. Cheers & Thanks so much. Appreciate you working with me to figure this out!
2. The *-cont version was ran via `rcom -d /dev/ttyUSB0 | grep SwitchBot` w/o storing any logs. It had about 3 results in 30 minutes.
3. App states firmware v0.6. There was no firmware update I could find in the app.
4. Unpaired the device from my phone and I've turned off the Switchbot-MQTT ESP.
5. Followed [these instructions to reset](https://support.switch-bot.com/hc/en-us/articles/360037754854-How-to-reset-SwitchBot-Meter-Meter-Plus-) the switchbot.
6. Flashed via Platform.io to esp32dev-ble from _development_ branch. Configuration below.
7. Currently logging with `TZ=UTC rcom -d /dev/ttyUSB0 | tee serial.log`. See below.
8. I'll publish a longer log after a couple of hours - can you confirm this is the kind of data you're looking for?

#### Config

```
[platformio]
; default_envs = esp32dev-ble
default_envs = esp32dev-ble-datatest

[libraries]
decoder = https://github.com/theengs/decoder
; #development
; decoder = https://github.com/DigiH/decoder#SBMT-test
;decoder = https://github.com/theengs/decoder

[env:esp32dev-ble-datatest]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager32}
  ${libraries.ble}
  ${libraries.decoder}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=2'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DpubBLEManufacturerData=true'
  '-DpubKnownBLEServiceData=true'
  '-DpubBLEServiceUUID=true'
  '-DGateway_Name="OpenMQTTGateway_ESP32_BLE"'
  '-DMQTT_SERVER="172.16.32.2"'
  '-DMQTT_USER="username"'
  '-DMQTT_PASS="pwd"'
  '-Dwifi_ssid="XiFi"'
  '-Dwifi_password="l3ttersandnumbers"'
```

#### Log

At 16:03 (UTC) I pressed the bluetooth button to see what would happen, the bluetooth indicator lighted and it got picked up.  So there may be a bit of a nuance in the data around that timestamp.

```
$ TZ=UTC rcom -d /dev/ttyUSB0 | tee serial.log

[15:57:03.0016] rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:57:03.0066] configsip: 0, SPIWP:0xee
[15:57:03.0086] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:57:03.0145] mode:DIO, clock div:2
[15:57:03.0166] load:0x3fff0018,len:4
[15:57:03.0187] load:0x3fff001c,len:1044
[15:57:03.0218] load:0x40078000,len:10124
[15:57:03.0236] load:0x40080400,len:5828
[15:57:03.0256] entry 0x400806a8
[15:57:03.6081] N: 
[15:57:03.6086] ************* WELCOME TO OpenMQTTGateway **************
[15:57:03.6136] N: OpenMQTTGateway Version: version_tag
[... 122 lines cut ...]
[16:05:56.7341] N: Device detected: E4:8F:4C:26:CE:EE
[16:05:56.7783] N: Send on /BTtoMQTT/E48F4C26CEEE msg {"id":"E4:8F:4C:26:CE:EE","mac_type":1,"manufacturerdata":"4c0012020000","rssi":-33}
[16:05:58.3342] N: Device detected: C6:C7:52:66:3C:BF
[16:05:58.3680] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf060b02924a","rssi":-30,"servicedata":"6980e402924a","servicedatauuid":"0xfd3d","brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.2,"tempf":64.76,"hum":74,"batt":100}
[16:06:03.6935] N: Found 3 devices, scan number 9 end
[16:06:03.6973] N: Device detected: 3C:91:80:C9:28:58
[16:06:03.7003] N: Send on /BTtoMQTT/3C9180C92858 msg {"id":"3C:91:80:C9:28:58","mac_type":0,"rssi":-81}
[16:06:59.2570] N: Scan begin
[16:07:08.1807] N: Send on /SYStoMQTT msg {"uptime":604,"version":"version_tag","freemem":138900,"mqttport":"1883","mqttsecure":false,"freestack":3444,"rssi":-48,"SSID":"XiFi","BSSID":"08:55:31:D6:79:49","ip":"192.168.88.231","mac":"34:86:5D:FD:4F:08","lowpowermode":0,"btqblck":0,"btqsum":19,"btqsnd":19,"btqavg":1,"interval":55555,"scanbcnct":10,"scnct":9,"modules":["BT"]}
[16:07:09.2667] N: Found N: Device detected: C6:C7:52:66:3C:BF
[16:07:09.2713] 2 devices, scan number 10 end
[16:07:09.2743] N: BLE Connect begin
[16:07:09.2753] N: BLE Connect end
[16:07:09.3107] N: Device detected: 50:CB:2D:D9:62:25
[16:07:09.3542] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf060b02924a","rssi":-29}
[16:07:09.3663] N: Send on /BTtoMQTT/50CB2DD96225 msg {"id":"50:CB:2D:D9:62:25","mac_type":1,"manufacturerdata":"4c0010054e1c3b4881","rssi":-90,"txpower":12}
[16:08:04.8319] N: Scan begin
[16:08:14.8325] N: Device detected: E4:8F:4C:26:CE:EE
[16:08:14.8412] N: Found 3 devices, scan number 11 end
[16:08:14.8766] N: Device detected: 3C:91:80:C9:28:58
[16:08:14.8803] N: Device detected: C6:C7:52:66:3C:BF
[16:08:14.9207] N: Send on /BTtoMQTT/E48F4C26CEEE msg {"id":"E4:8F:4C:26:CE:EE","mac_type":1,"manufacturerdata":"4c0012020000","rssi":-28}
[16:08:14.9313] N: Send on /BTtoMQTT/3C9180C92858 msg {"id":"3C:91:80:C9:28:58","mac_type":0,"rssi":-80}
[16:08:14.9394] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf060b02924a","rssi":-33}
```

DigiH, 2022-10-22:
Hi,

The 'fix' I implemented previously, which allows for the 49… servicedata to be decoded, should actually be taken out again, as 49… servicedata indicates being in ADD MODE, which last for 120 seconds, as can be seen in your

> At 16:03 (UTC) I pressed the bluetooth button to see what would happen, the bluetooth indicator lighted and it got picked up. So there may be a bit of a nuance in the data around that timestamp.

> 8\. I'll publish a longer log after a couple of hours - can you confirm this is the kind of data you're looking for?

Yes, the configuration looks fine, the steps you follow sound fine, but why so many manufacturerdata only messages is still a mystery to me.

If the longer log will be not serial, but as the above ones with _mosquitto-sub_, that will be great.

Just out of curiosity - you haven't set **ActiveBLEScan** in **config_BT.h** to _false_ by any chance, possibly for some other environment/test?

Looking forward to your longer logs.

xanderificnl, 2022-10-22:
Hi,

Before filing this issue, I repulled 0.9.16 and had a clean tree. Today I moved to the development branch. Git reports my tree is clean (gitignore ignores env files). So ActiveBLEScan is default, in config_BT.h:

```
$ grep 'define ActiveBLEScan' main/config_BT.h 
#  define ActiveBLEScan true // Set active scanning, this will get more data from the advertiser.
```

**I've started a subscriber for mosquitto-sub**

Things do appear to be going much better so far. I've had more sensor readings by OMG than I've had in the past 10 days. 

Here's a preview of the formatted logs so far:

```
{"@timestamp":"2022-10-22T19:26:56+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf110b09914b","rssi":-29}
{"@timestamp":"2022-10-22T19:29:07+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf110b09914b","rssi":-29}
{"@timestamp":"2022-10-22T19:31:18+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf110b09914b","rssi":-33}
{"@timestamp":"2022-10-22T19:32:24+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf110b09914b","rssi":-33}
{"@timestamp":"2022-10-22T19:33:28+02:00","batt":100,"brand":"SwitchBot","hum":75,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf120b09914b","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-30,"servicedata":"6980e409914b","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T19:34:35+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf120b09914b","rssi":-32}
{"@timestamp":"2022-10-22T19:35:41+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf120b09914b","rssi":-29}
{"@timestamp":"2022-10-22T19:36:36+02:00","batt":100,"brand":"SwitchBot","hum":75,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf120b09914b","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-30,"servicedata":"6980e409914b","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T19:37:43+02:00","batt":100,"brand":"SwitchBot","hum":75,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf120b09914b","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-31,"servicedata":"6980e409914b","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
```

DigiH, 2022-10-22:
The `[env:esp32dev-ble-datatest]` environment looks fine. Could you just add the build_flags?

```
'-DTimeBtwRead=0'
'-DScan_duration=1000'
'-DAttemptBLECOnnect=false'
```

to make it a `env:esp32dev-ble-datatest-cont`. Logging for ten minutes or so should be more than enough then to see what the different broadcast ratios are.

Thanks

xanderificnl, 2022-10-22:
For completeness: it reported sensor readings 43 times in approximately 2 hours. A lot more than previously - but still less than what your friend sees. 

--- 
switched to the new build flags, and here's the log: 

```
{"@timestamp":"2022-10-22T19:59:08+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T19:59:13+02:00","batt":100,"brand":"SwitchBot","hum":74,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-31,"servicedata":"6980e409914a","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T19:59:25+02:00","batt":100,"brand":"SwitchBot","hum":74,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-31,"servicedata":"6980e409914a","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T19:59:31+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-33}
{"@timestamp":"2022-10-22T19:59:33+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-33}
{"@timestamp":"2022-10-22T19:59:37+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-32}
{"@timestamp":"2022-10-22T19:59:39+02:00","batt":100,"brand":"SwitchBot","hum":74,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-33,"servicedata":"6980e409914a","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T19:59:45+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-81}
{"@timestamp":"2022-10-22T19:59:47+02:00","batt":100,"brand":"SwitchBot","hum":74,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-33,"servicedata":"6980e409914a","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T19:59:49+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T19:59:53+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T19:59:55+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-62}
{"@timestamp":"2022-10-22T20:00:03+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:00:05+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-33}
{"@timestamp":"2022-10-22T20:00:13+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:00:19+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:00:23+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-32}
{"@timestamp":"2022-10-22T20:00:27+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:00:29+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:00:33+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:00:35+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:00:41+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:00:43+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-33}
{"@timestamp":"2022-10-22T20:00:57+02:00","batt":100,"brand":"SwitchBot","hum":74,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-31,"servicedata":"6980e409914a","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T20:01:02+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-33}
{"@timestamp":"2022-10-22T20:01:04+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:01:10+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:01:11+02:00","batt":100,"brand":"SwitchBot","hum":74,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-33,"servicedata":"6980e409914a","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T20:01:18+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
{"@timestamp":"2022-10-22T20:01:26+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf150b09914a","rssi":-30}
[... cut ...]
{"@timestamp":"2022-10-22T20:09:09+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf160b09914a","rssi":-33}
{"@timestamp":"2022-10-22T20:09:11+02:00","batt":100,"brand":"SwitchBot","hum":74,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf160b09914a","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-31,"servicedata":"6980e409914a","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}
{"@timestamp":"2022-10-22T20:09:15+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf160b09914a","rssi":-33}
```

xanderificnl, 2022-10-22:
Wow. On a brand new ESP, with these opts:

```
[env:esp32dev-ble-datatest]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager32}
  ${libraries.ble}
  ${libraries.decoder}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=2'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DpubBLEManufacturerData=true'
  '-DpubKnownBLEServiceData=true'
  '-DpubBLEServiceUUID=true'
  '-DGateway_Name="OpenMQTTGateway_ESP32_BLE"'
  '-DMQTT_SERVER="172.16.32.2"'
  '-DMQTT_USER="username"'
  '-DMQTT_PASS="pwd"'
  '-Dwifi_ssid="XiFi"'
  '-Dwifi_password="snip"'
  '-DTimeBtwRead=0'
  '-DScan_duration=1000'
  '-DAttemptBLECOnnect=false'
  '-DESPWifiManualSetup=true'
```

This is happening:

```
{"@timestamp":"2022-10-22T20:41:13+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","rssi":-27}
{"@timestamp":"2022-10-22T20:41:22+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-33,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:41:26+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-30,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:41:34+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-30,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:41:36+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-30,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:41:42+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-33,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:41:46+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","rssi":-28}
{"@timestamp":"2022-10-22T20:41:56+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-33,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:41:58+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-32,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:42:00+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-32,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:42:06+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-32,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:42:08+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-29,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:42:16+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-33,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:42:22+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","rssi":-29}
{"@timestamp":"2022-10-22T20:42:32+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-32,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T20:42:38+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","rssi":-31}
{"@timestamp":"2022-10-22T20:42:40+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","rssi":-27}
{"@timestamp":"2022-10-22T20:42:54+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-28,"servicedata":"698064009249","servicedatauuid":"0xfd3d","te [... cut ...]
{"@timestamp":"2022-10-22T20:43:41+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf1d0b009249","rssi":-31}
```

#### So perhaps the other 2 ESP32s need a full reflash?

So something weird happened. I grabbed an ESP32 that was lying around, I pressed 'upload' and it wasn't able to connect to WiFi. FIgured it was a fluke, and unpacked a new one. Same problem.

Added the `'-DESPWifiManualSetup=true'` flag, and it did work. A flag I had used before on the other 2 ESPs.

So apparently the other 2 ESPs still have dangling settings running afoul? 

#### New ESP: 5 minutes & 36 sensor readings....

DigiH, 2022-10-22:
> Just unpacked a new ESP for testing; and it -won't- connect to WiFi with the _exact same_ configuration, i.e. all I did was press "upload". Starting to wonder if there may be remnants of a previous configuration dangling around on the other 2 ESPs.

Best to erase the Flash before uploading a new configuration to a new or well used ESP32. Worth a try with these two.

> (I did believe the "upload" on platformio did that.)

`esptool.py --port /dev/ttyUSB0 erase_flash`

in the platformio terminal will do the trick.

> #### time (in seconds) between sensor readings:
> 
> ```
> 5 12 14 8 70 14 16 8 6 26 2 4 14 72 9 20 18 4 42 10 12 38 68 28 18 16 2 6 10 31 14 104 22 24 8 6 6 10 38 20 8 8 7 1 13 20 16 51 26 10 22 19 32 6 8 8 54 8 
> ```

Well, more regular decoded readings than the previous `switcbot-mqtt-ble-esp32.log` now, undecoded 'breaks' not more than roughly a minute apart. 

Still very curious as to why there are so many more manufacturerdata only messages than decoded ones including the servicedata, when it should be the other way round.

But this clearly shows that with the default 

```
'-DTimeBtwRead=55000'
'-DScan_duration=10000'
```

the 10 second scan window every 55 seconds could/would easily fall into several manufacturerdata only broadcasts, continuously only showing these for a long time, even if correctly decoded servicedata messages had been broadcast during the TimeBtwRead. What you saw before with the default environment/binary.

Is/was the SwitchBot Meter linked to a SwitchBot Hub at all?

Any improvements after clearing the flash before a new upload?

DigiH, 2022-10-22:
> Added the `'-DESPWifiManualSetup=true'` flag, and it did work. A flag I had used before on the other 2 ESPs.

if you use manual WiFi and MQTT broker credentials, as in your environment above
```
  '-DMQTT_SERVER="172.16.32.2"'
  '-DMQTT_USER="username"'
  '-DMQTT_PASS="pwd"'
  '-Dwifi_ssid="XiFi"'
  '-Dwifi_password="snip"'
```

you will always also have to set 
`'-DESPWifiManualSetup=true'`

as otherwise the above should be recognised. As it is indicated in the above environment.

xanderificnl, 2022-10-22:
Yeah, so I did start off with the binary release - and that didn't give me sensor data. So I moved on to building myself and fiddling with settings. So presumably increasing the window might solve the issue?

I don't have or own a hub. 

So on the brand new ESP is off and I've wiped one of the previous ESPs, cleaned the environment, built and uploaded it. These are the logs:

```
{"@timestamp":"2022-10-22T21:08:54+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf200b009249","rssi":-25}
{"@timestamp":"2022-10-22T21:09:01+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-25}
{"@timestamp":"2022-10-22T21:09:02+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-24}
{"@timestamp":"2022-10-22T21:09:06+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-23}
{"@timestamp":"2022-10-22T21:09:08+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-24}
{"@timestamp":"2022-10-22T21:09:10+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-25}
{"@timestamp":"2022-10-22T21:09:12+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-23}
{"@timestamp":"2022-10-22T21:09:16+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-24}
{"@timestamp":"2022-10-22T21:09:18+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-27,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T21:09:20+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-26}
{"@timestamp":"2022-10-22T21:09:26+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-23}
{"@timestamp":"2022-10-22T21:09:28+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-26}
{"@timestamp":"2022-10-22T21:09:30+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-26}
{"@timestamp":"2022-10-22T21:09:32+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-24}
{"@timestamp":"2022-10-22T21:09:36+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-24}
{"@timestamp":"2022-10-22T21:09:42+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-26}
{"@timestamp":"2022-10-22T21:09:44+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-24}
{"@timestamp":"2022-10-22T21:09:50+02:00","batt":100,"brand":"SwitchBot","hum":73,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-26,"servicedata":"698064009249","servicedatauuid":"0xfd3d","tempc":18,"tempf":64.4}
{"@timestamp":"2022-10-22T21:09:52+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf210b009249","rssi":-23}
```

So the brand new ESP actually flooded the logs with sensor data but that also returned to manufacturerdata. So it was mostlikely a fluke. 

Presumably changing/increase the window would solve it?

xanderificnl, 2022-10-22:
with regards to the window: I wasn't reading correctly. You meant with the regular release, not the adjusted settings we've been using, correct?

> you will always also have to set `'-DESPWifiManualSetup=true'`

Yes, but with the ESP I had been testing on, I hadn't set that flag for the datatest environment. It still connected to WiFi whereas 2 other ESPs failed to do so. So there must have been dangling resources/configuration going on. 

-
I'm inclinded to let the new ESP run for a night and see how it fares. So far it's been doing so much better even though it does seem to get flooded with manufacturer-only messages everyonce in a while.

DigiH, 2022-10-22:
> Presumably changing/increase the window would solve it?

Yes, I would think having a go with the `TimeBtwRead` and `Scan_duration` should give you a usable result, as the continuous scanning showed that you can catch a decoded message at least every minute.

Just to show that something is still amiss is the comparison between your -cont log and the colleague's -cont log

[image]

That right side would also be the ideal for your SwichBot Meter as well! 

Why there seems to be a majority of manufacturerdata only messages is beyond me at this stage, but I think we found, that with nicely adjusted `TimeBtwRead` and `Scan_duration` settings you can get a workable OMG setup, with at least one correct decoded reading every minute.

xanderificnl, 2022-10-22:
That's a huge difference. With the new ESP, I've been seeing only manufacturer data for more than 5 minutes now and that's with `esp32dev-ble-datatest`. 

I'm going to google around some more and see if I can find a project that allows me to use an ESP32 as a BLE-receiver and inspect that data in Wireshark. Presumably, we'd get a little bit more raw information about what's going on. 

Something tells me that my Switchbot is looking for a hub, or it maybe found a hub (that isn't mine) somewhere near me. Or that it's looking for registration or something along those lines. 

Thanks for the help. It's really nice, I'll update when I get the chance!

1technophile, 2022-10-23:
Hello,

Do you mind sharing a picture of the back of your sensor.
Here is mind that report constantly data:
[image]

The goal is to check for any model difference.

xanderificnl, 2022-10-23:
[image]

Mine seems to have an extra IC-field (next to the FCC-ID) but besides that I see no informational differences

DigiH, 2022-10-24:
Filtering out the decoded messages only there still are occasional gaps of around 15 minutes without any servicedata/decoding ☹️ 

If and why this is happening with your SwitchBot Meter is beyond me, as there are only slight differences with the model printed information above.

Lastly, to account for possible reception issues of your ESP32s - which I don't really think is the problem here - you could also try running [Theengs Gateway](https://github.com/theengs/gateway) on your laptop/desktop PC, just to rule out that last possible parameter on the receiving side.

xanderificnl, 2022-10-24:
It's running and similar to [when I flashed a newly unpacked ESP32](https://github.com/1technophile/OpenMQTTGateway/issues/1288#issuecomment-1287881767) I'm only seeing sensor readings.

At least 1 reading per 10 seconds so far - I'll let it run for a bit.

xanderificnl, 2022-10-24:
This should be enough time: _every single publications to MQTT contained sensor readings_, approximately 16 readings per minute. 

So I actually cleared the flash from an ESP and flashed esp32dev-ble (regular) and all `BTtoMQTT` topics before I started Theengs and the ESP. This way, I could see them side by side. 

##### Here are some observations.

top data is esp32dev-ble.
[image]

[image]

#### My interpretation (so far)

* The window and at least the interval can account for _some of the difference_ in observed mac addresses/devices.  The rest is accounted by the ESP picking up between -100 and -30 whereas Theengs/bluetooth adapter only picks up between -80 and -40.
* The ESP saw and decoded a Windows PC and a Razer Stereo device.
* Theengs didn't see either of those, but did decode an OLED TV and the Switchbot.

So Theeng received about 16 sensor readings _per minute_ and there wasn't a single manufacturer only reading. The brand spanking newly unpacked ESP was able to see 36 readings in 5 minutes - still less than Theengs but looking at MQTT, we can clearly see the ESP is picking up a lot of stuff - and it's able to decode the Switchbot's data.

##### Some debug from Theengs gw

```
DEBUG:BLEGateway:C6:C7:52:66:3C:BF RSSI:-63 AdvertisementData(manufacturer_data={2409: b'\xc6\xc7Rf<\xbf9\x0b\x05\x91K'}, service_data={'0000fd3d-0000-1000-8000-00805f9b34fb': b'i\x80\xe4\x05\x91K'}, rssi=-63)
INFO:BLEGateway:publishing device `{'id': 'C6:C7:52:66:3C:BF', 'rssi': -63, 'brand': 'SwitchBot', 'model': 'Meter (Plus)', 'model_id': 'THX1/W230150X', 'tempc': 17.5, 'tempf': 63.5, 'hum': 75, 'batt': 100}
```

binary resolved:

```
manufacturer_data={2409: 'c6c752663cbf390b05914b'}
service_data={'0000fd3d-0000-1000-8000-00805f9b34fb': '6980e405914b'}
```

The only difference in the manufacturer data is 6909 vs. 2409; presuming the key and value are concatenated. 
```
# from omg
{"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf390b05914b","rssi":-51}
```

_I'm clearly no expert on BLE, OMG, etc. seeing as this process has been my first interaction with both, but is it possible the signal is getting lost/cut off or that the ESP just can't keep up with all the broadcasts going on?_

1technophile, 2022-10-24:
Thanks for this extensive testing, could you try with the last development version, in particular I would like to check if [this PR](https://github.com/1technophile/OpenMQTTGateway/commit/c612169ba80f201ce8a162f683f42c5d65237d72) improved BLE reception by decreasing wifi/MQTT usage.

Alternatively, you could try [this](https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/examples/NimBLE_Scan_Continuous/NimBLE_Scan_Continuous.ino) also and see if you constantly get servicedata and manufacturerdata for the sswitchbot.

xanderificnl, 2022-10-24:
#### with the PR

* clean builds
* esptool erase flash
* build [c612169ba80f201ce8a162f683f42c5d65237d72](https://github.com/1technophile/OpenMQTTGateway/commit/c612169ba80f201ce8a162f683f42c5d65237d72)

Mostly manufacturer only readings

#### esphome

The nimble example gave me a bit of grieve so I figured: let's try ESPHome's `esp32_ble_tracker` module. 

Default tracker config

```
[00:07:23.6600] [C][esp32_ble_tracker:787]: BLE Tracker:
[00:07:23.6644] [C][esp32_ble_tracker:788]:   Scan Duration: 300 s
[00:07:23.6706] [C][esp32_ble_tracker:789]:   Scan Interval: 320.0 ms
[00:07:23.6765] [C][esp32_ble_tracker:790]:   Scan Window: 30.0 ms
[00:07:23.6814] [C][esp32_ble_tracker:791]:   Scan Type: ACTIVE
[00:07:23.6866] [C][esp32_ble_tracker:792]:   Continuous Scanning: True
```

Excerpt from the serial console:

```
[00:20:36.0707] [VV][esp32_ble_tracker:575]: Parse Result:
[00:20:36.0753] [VV][esp32_ble_tracker:592]:   Address: C6:C7:52:66:3C:BF (RANDOM)
[00:20:36.0823] [VV][esp32_ble_tracker:594]:   RSSI: -27
[00:20:36.0877] [VV][esp32_ble_tracker:595]:   Name: ''
[00:20:36.0913] [VV][esp32_ble_tracker:603]:   Ad Flag: 6
[00:20:36.0963] [VV][esp32_ble_tracker:609]:   Manufacturer data: C6.C7.52.66.3C.BF.61.0B.07.91.4A (11)
[00:20:36.1056] [VV][esp32_ble_tracker:620]:   Service data:
[00:20:36.1104] [VV][esp32_ble_tracker:621]:     UUID: 0xFD3D
[00:20:36.1162] [VV][esp32_ble_tracker:622]:     Data: 69.80.E4.07.91.4A (6)
[00:20:36.1213] [VV][esp32_ble_tracker:625]: Adv data: 02.01.06.0E.FF.69.09.C6.C7.52.66.3C.BF.61.0B.07.91.4A.09.16.3D.FD.69.80.E4.07.91.4A (28)
[00:20:36.1334] [D][ble_adv:012]: SwitchBot
[00:20:36.1373] [V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.06 s).
[00:20:36.1463] [V][component:200]: Components should block for at most 20-30ms.

[00:25:31.8750] [VV][esp32_ble_tracker:575]: Parse Result:
[00:25:31.8805] [VV][esp32_ble_tracker:592]:   Address: C6:C7:52:66:3C:BF (RANDOM)
[00:25:31.8866] [VV][esp32_ble_tracker:594]:   RSSI: -28
[00:25:31.8916] [VV][esp32_ble_tracker:595]:   Name: ''
[00:25:31.8967] [VV][esp32_ble_tracker:603]:   Ad Flag: 6
[00:25:31.9029] [VV][esp32_ble_tracker:609]:   Manufacturer data: C6.C7.52.66.3C.BF.66.0B.03.92.57 (11)
[00:25:31.9096] [VV][esp32_ble_tracker:620]:   Service data:
[00:25:31.9145] [VV][esp32_ble_tracker:621]:     UUID: 0xFD3D
[00:25:31.9196] [VV][esp32_ble_tracker:622]:     Data: 69.80.E4.03.92.57 (6)
[00:25:31.9256] [VV][esp32_ble_tracker:625]: Adv data: 02.01.06.0E.FF.69.09.C6.C7.52.66.3C.BF.66.0B.03.92.57.09.16.3D.FD.69.80.E4.03.92.57 (28)
[00:25:31.9376] [V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.05 s).
[00:25:31.9469] [V][component:200]: Components should block for at most 20-30ms.

[00:25:41.1333] [VV][esp32_ble_tracker:575]: Parse Result:
[00:25:41.1386] [VV][esp32_ble_tracker:592]:   Address: C6:C7:52:66:3C:BF (RANDOM)
[00:25:41.1455] [VV][esp32_ble_tracker:594]:   RSSI: -23
[00:25:41.1495] [VV][esp32_ble_tracker:595]:   Name: ''
[00:25:41.1549] [VV][esp32_ble_tracker:603]:   Ad Flag: 6
[00:25:41.1595] [VV][esp32_ble_tracker:609]:   Manufacturer data: C6.C7.52.66.3C.BF.68.0B.03.92.50 (11)
[00:25:41.1676] [VV][esp32_ble_tracker:620]:   Service data:
[00:25:41.1729] [VV][esp32_ble_tracker:621]:     UUID: 0xFD3D
[00:25:41.1777] [VV][esp32_ble_tracker:622]:     Data: 69.80.E4.03.92.50 (6)
[00:25:41.1846] [VV][esp32_ble_tracker:625]: Adv data: 02.01.06.0E.FF.69.09.C6.C7.52.66.3C.BF.68.0B.03.92.50.09.16.3D.FD.69.80.E4.03.92.50 (28)

```

It reads about once a minute w/ service data. The component warning happens after _some devices_, but not all. Interestingly to note is that the manufacturer data changes whereas with OMG it stays the same, i.e. from one of my `omg.log`'s

```
$ jq '.manufacturerdata' ~/omg.log | wc -l
46
$ jq '.manufacturerdata' ~/omg.log | sort -u
"6909c6c752663cbf290b079149"
"6909c6c752663cbf2a0b069149"
```

So far ESPHome's output is comparable with Theengs, compared the data into ESPHome's dotted format for readability i.e.

~ | manufacturer data |
-|-
ESPHOME | C6.C7.52.66.3C.BF.68.0B.03.92.50
Theengs |     C6.C7.52.66.3C.BF.39.0B.05.91.4B
OMG |          69.09.C6.C7.52.66.3C.BF.2A.0B.06.91.49

Why do you think OMG's is prefixed by 69.09 instead of the mac address such as the others?

h2zero, 2022-10-25:
> The nimble example gave me a bit of grieve

Could you elaborate on this?

> Why do you think OMG's is prefixed by 69.09 instead of the mac address such as the others?

I'll look into this.

xanderificnl, 2022-10-25:
> > The nimble example gave me a bit of grieve
> 
> Could you elaborate on this?

I was in a bit of a rush (had a few moments in between work), so I figured I'd quickly install Arduino's IDE; but found Arduino's web IDE instead. So I imported the library and added the example, but it wouldn't compile. It ran into an issue looking for a config.h file.  I figured, I'd look into it a little bit later. So I did, and then it ran into missing FreeRTOS (iirc) and so I added that, but that didn't pan out. So again, I figured I'd look into it a bit later and in the meantime I just flashed ESPHome w/ some config to get some data.

So _nimble example gave me a bit of grieve_ really means _I'm still very inexperienced in the ESP* ecosystem and haven't found my way around yet_ :)

Truth be told, I probably overlooked something obvious; or read over something while it was staring me in the face. 

> 6909 in little endian = 0x0969 = 2409 decimal, which represents the manufacturer ID in the BLE manufacturer data type.

Ah, that makes sense. I'll be giving the nimble example another try - hopefully later today - to see if I can confirm the observations made with ESPHome. 

--

I do want to express my gratitude to all of you, and especially @DigiH for all his time and effort, for helping me out and looking into this!

DigiH, 2022-10-25:
> Interestingly to note is that the manufacturer data changes whereas with OMG it stays the same

Not quite the case ;) as with OMG/Decoder the manufacturerdata also changes, as the last 3 bytes do actually contain the temperature and humidity information (no battery). Easily visible when both manufacturerdata and servicedata are present

{"@timestamp":"2022-10-22T20:07:20+02:00","batt":100,"brand":"SwitchBot","hum":74,"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf160b**09914a**","model":"Meter (Plus)","model_id":"THX1/W230150X","rssi":-33,"servicedata":"6980e4**09914a**","servicedatauuid":"0xfd3d","tempc":17.9,"tempf":64.22}

Unfortunately when it is a manufacturerdata only transmission there is no additional clear identification as with the  servicedata's device type and the servicedatauuid to uniquely identify this device to be able to use manufacturerdata only for a clear decoding.

{"@timestamp":"2022-10-22T20:08:51+02:00","id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf160b**09914a**","rssi":-33}

xanderificnl, 2022-10-26:
Here are the logs of the NimBLE example:

[c6:c7:52:66:3c:bf = the switchbot]

```
[15:09:07.0065] Scanning...
[15:09:07.2946] Advertised Device: Name: HwZ_fb113f1ab1002b1915, Address: 00:a0:50:63:16:55 
[15:09:07.3965] Advertised Device: Name: , Address: 64:6f:d4:4f:e1:57, manufacturer data: 4c001005441c8bb550, txPower: 12 
[15:09:12.1448] Advertised Device: Name: , Address: f5:e4:04:f9:c7:e8, manufacturer data: 4c0012020003 
[15:09:24.4545] Advertised Device: Name: , Address: d4:5d:76:96:ee:4c, manufacturer data: 4c0012020000 
[15:09:26.2919] Advertised Device: Name: , Address: c6:c7:52:66:3c:bf, manufacturer data: 6909c6c752663cbfc10b01924d
[15:09:26.3007] Service Data:
[15:09:26.3027] UUID: 0xfd3d, Data: i���M 
[15:10:07.2678] Advertised Device: Name: HwZ_fb113f1ab1002b1915, Address: 00:a0:50:63:16:55 
[15:10:08.1237] Advertised Device: Name: , Address: 64:6f:d4:4f:e1:57, manufacturer data: 4c001005441c8bb550, txPower: 12 
[15:10:16.1883] Advertised Device: Name: , Address: f5:e4:04:f9:c7:e8, manufacturer data: 4c0012020003 
[15:10:26.4362] Advertised Device: Name: , Address: c6:c7:52:66:3c:bf, manufacturer data: 6909c6c752663cbfc10b01924d
[15:10:26.4456] Service Data:
[15:10:26.4465] UUID: 0xfd3d, Data: i���M 
[15:11:07.2196] Advertised Device: Name: HwZ_fb113f1ab1002b1915, Address: 00:a0:50:63:16:55 
[15:11:12.5512] Advertised Device: Name: , Address: c6:c7:52:66:3c:bf, manufacturer data: 6909c6c752663cbfc10b01924d
[15:11:12.5604] Service Data:
[15:11:12.5614] UUID: 0xfd3d, Data: i���M 
[15:11:14.6790] Advertised Device: Name: , Address: 64:6f:d4:4f:e1:57, manufacturer data: 4c001005441c8bb550, txPower: 12 
[15:11:16.2264] Advertised Device: Name: , Address: f5:e4:04:f9:c7:e8, manufacturer data: 4c0012020003 
[15:12:07.1737] Advertised Device: Name: , Address: 64:6f:d4:4f:e1:57, manufacturer data: 4c001005441c8bb550, txPower: 12 
[15:12:10.2758] Advertised Device: Name: , Address: f5:e4:04:f9:c7:e8, manufacturer data: 4c0012020003 
[15:12:25.1998] Advertised Device: Name: , Address: 58:8e:c6:2a:9a:7f, manufacturer data: 4c0002151ca92e23f0874df7b9a2fd4b716a4bf60084000003 
[15:12:30.7349] Advertised Device: Name: , Address: c6:c7:52:66:3c:bf, manufacturer data: 6909c6c752663cbfc10b01924d
[15:12:30.7441] Service Data:
[15:12:30.7451] UUID: 0xfd3d, Data: i���M 
[15:13:07.1300] Advertised Device: Name: HwZ_fb113f1ab1002b1915, Address: 00:a0:50:63:16:55 
[15:13:08.0068] Advertised Device: Name: , Address: 64:6f:d4:4f:e1:57, manufacturer data: 4c001005441c8bb550, txPower: 12 
[15:13:08.3238] Advertised Device: Name: , Address: f5:e4:04:f9:c7:e8, manufacturer data: 4c0012020003 
[15:13:08.7941] Advertised Device: Name: , Address: c6:c7:52:66:3c:bf, manufacturer data: 6909c6c752663cbfc10b01924d
[15:13:08.8030] Service Data:
[15:13:08.8050] UUID: 0xfd3d, Data: i���M 
[15:14:07.0962] Advertised Device: Name: HwZ_fb113f1ab1002b1915, Address: 00:a0:50:63:16:55 
[15:14:09.6395] Advertised Device: Name: , Address: 64:6f:d4:4f:e1:57, manufacturer data: 4c001005441c8bb550, txPower: 12 
[15:14:14.3629] Advertised Device: Name: , Address: f5:e4:04:f9:c7:e8, manufacturer data: 4c0012020003 
[15:14:16.9680] Advertised Device: Name: , Address: c6:c7:52:66:3c:bf, manufacturer data: 6909c6c752663cbfc10b01924d
[15:14:16.9769] Service Data:
[15:14:16.9789] UUID: 0xfd3d, Data: i���M 
```

> > The nimble example gave me a bit of grieve
> 
> Could you elaborate on this?

@h2zero The problem was either with me and/or the web ide. After installing Arduino's IDE locally, I got the example running in less than 5 minutes flat w/o any fuzz.

1technophile, 2022-10-27:
Let me try to run it on my side with my switchbot, and share what I see

1technophile, 2022-10-31:
Could you try with this sketch please with NimBLEArduino v1.3.8 and share your logs:
```cpp
/** Example of continuous scanning for BLE advertisements.
 * This example will scan forever while consuming as few resources as possible
 * and report all advertisments on the serial monitor.
 *
 * Created: on January 31 2021
 *      Author: H2zero
 *
 */

#include "NimBLEDevice.h"

NimBLEScan* pBLEScan;

std::string convertServiceData(std::string deviceServiceData) {
  int serviceDataLength = (int)deviceServiceData.length();
  char spr[2 * serviceDataLength + 1];
  for (int i = 0; i < serviceDataLength; i++) sprintf(spr + 2 * i, "%.2x", (unsigned char)deviceServiceData[i]);
  spr[2 * serviceDataLength] = 0;
  return spr;
}

class MyAdvertisedDeviceCallbacks: public NimBLEAdvertisedDeviceCallbacks {
    void onResult(NimBLEAdvertisedDevice* advertisedDevice) {
      Serial.println("-----------------------------------------------------------------------------------");
      Serial.printf("Advertised Device mac: %s \n", advertisedDevice->getAddress().toString().c_str());
      Serial.printf("Advertised Device name: %s \n", (char*)advertisedDevice->getName().c_str());
      if (advertisedDevice->haveManufacturerData()) {
         char* manufacturerdata = BLEUtils::buildHexData(NULL, (uint8_t*)advertisedDevice->getManufacturerData().data(), advertisedDevice->getManufacturerData().length());
         Serial.printf("Advertised Device manufacturer data: %s \n", manufacturerdata);
         free(manufacturerdata);
      }
      if (advertisedDevice->haveServiceData()) {
          int serviceDataCount = advertisedDevice->getServiceDataCount();
           Serial.printf("Get services data number: %d \n", serviceDataCount);
          for (int j = 0; j < serviceDataCount; j++) {
            std::string service_data = convertServiceData(advertisedDevice->getServiceData(j));
            Serial.printf("Service data: %s \n", service_data.c_str());
            std::string serviceDatauuid = advertisedDevice->getServiceDataUUID(j).toString();
            Serial.printf("Service data UUID: %s \n", (char*)serviceDatauuid.c_str());
          }
      }
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

/** *Optional* Sets the filtering mode used by the scanner in the BLE controller.
 *
 *  Can be one of:
 *  CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE (0) (default)
 *  Filter by device address only, advertisements from the same address will be reported only once.
 *
 *  CONFIG_BTDM_SCAN_DUPL_TYPE_DATA (1)
 *  Filter by data only, advertisements with the same data will only be reported once,
 *  even from different addresses.
 *
 *  CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE (2)
 *  Filter by address and data, advertisements from the same address will be reported only once,
 *  except if the data in the advertisement has changed, then it will be reported again.
 *
 *  Can only be used BEFORE calling NimBLEDevice::init.
*/
  NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE);

/** *Optional* Sets the scan filter cache size in the BLE controller.
 *  When the number of duplicate advertisements seen by the controller
 *  reaches this value it will clear the cache and start reporting previously
 *  seen devices. The larger this number, the longer time between repeated
 *  device reports. Range 10 - 1000. (default 20)
 *
 *  Can only be used BEFORE calling NimBLEDevice::init.
 */
  NimBLEDevice::setScanDuplicateCacheSize(200);

  NimBLEDevice::init("");

  pBLEScan = NimBLEDevice::getScan(); //create new scan
  // Set the callback for when devices are discovered, no duplicates.
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), false);
  pBLEScan->setActiveScan(true); // Set active scanning, this will get more data from the advertiser.
  pBLEScan->setInterval(97); // How often the scan occurs / switches channels; in milliseconds,
  pBLEScan->setWindow(37);  // How long to scan during the interval; in milliseconds.
  pBLEScan->setMaxResults(0); // do not store the scan results, use callback only.
}

void loop() {
  // If an error occurs that stops the scan, it will be restarted here.
  if(pBLEScan->isScanning() == false) {
      // Start scan with: duration = 0 seconds(forever), no scan end callback, not a continuation of a previous scan.
      pBLEScan->start(0, nullptr, false);
  }

  delay(2000);
}
```

xanderificnl, 2022-11-01:
Here's the initial log:

```
[17:46:38.2958] ets Jul 29 2019 12:21:46
[17:46:38.2988] 
[17:46:38.2988] rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:46:38.3037] configsip: 0, SPIWP:0xee
[17:46:38.3056] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:46:38.3117] mode:DIO, clock div:1
[17:46:38.3136] load:0x3fff0030,len:1344
[17:46:38.3167] load:0x40078000,len:13864
[17:46:38.3188] load:0x40080400,len:3608
[17:46:38.3207] entry 0x400805f0
[17:46:38.5201] Scanning...
[17:46:51.0383] -----------------------------------------------------------------------------------
[17:46:51.0462] Advertised Device mac: d1:85:2e:a5:f8:a7 
[17:46:51.0492] Advertised Device name:  
[17:46:51.0522] Advertised Device manufacturer data: 4c0012020000 
[17:46:52.5724] -----------------------------------------------------------------------------------
[17:46:52.5804] Advertised Device mac: c6:c7:52:66:3c:bf 
[17:46:52.5834] Advertised Device name:  
[17:46:52.5866] Advertised Device manufacturer data: 6909c6c752663cbf700f069246 
[17:46:52.5914] Get services data number: 1 
[17:46:52.5945] Service data: 69c0e4069246 
[17:46:52.5964] Service data UUID: 0xfd3d 
[17:47:00.4284] -----------------------------------------------------------------------------------
[17:47:00.4354] Advertised Device mac: c6:c7:52:66:3c:bf 
[17:47:00.4394] Advertised Device name:  
[17:47:00.4413] Advertised Device manufacturer data: 6909c6c752663cbf710f069245 
[17:47:00.4473] Get services data number: 1 
[17:47:00.4503] Service data: 69c0e4069245 
[17:47:00.4523] Service data UUID: 0xfd3d 
[17:47:04.2898] -----------------------------------------------------------------------------------
[17:47:04.2978] Advertised Device mac: 7e:38:4c:ef:28:31 
[17:47:04.3008] Advertised Device name:  
[17:47:04.3028] Advertised Device manufacturer data: 4c001005561cfe9edf 
[17:47:05.5530] -----------------------------------------------------------------------------------
[17:47:05.5610] Advertised Device mac: 3c:91:80:c9:28:58 
[17:47:05.5640] Advertised Device name:  
[17:47:45.0912] -----------------------------------------------------------------------------------
[17:47:45.0988] Advertised Device mac: d1:85:2e:a5:f8:a7 
[17:47:45.1028] Advertised Device name:  
[17:47:45.1047] Advertised Device manufacturer data: 4c0012020000 
[17:47:55.5469] -----------------------------------------------------------------------------------
[17:47:55.5550] Advertised Device mac: 7f:b1:fd:c6:cd:c6 
[17:47:55.5579] Advertised Device name:  
[17:47:55.5599] Advertised Device manufacturer data: 4c001006141e6ba53556 
[17:48:28.3918] -----------------------------------------------------------------------------------
[17:48:28.3998] Advertised Device mac: dd:bf:df:a5:be:c7 
[17:48:28.4028] Advertised Device name:  
[17:48:28.4058] Advertised Device manufacturer data: 4c0012020003 
[17:48:30.4981] -----------------------------------------------------------------------------------
[17:48:30.5050] Advertised Device mac: c6:c7:52:66:3c:bf 
[17:48:30.5090] Advertised Device name:  
[17:48:30.5110] Advertised Device manufacturer data: 6909c6c752663cbf710f069245 
[17:48:30.5171] Get services data number: 1 
[17:48:30.5201] Service data: 69c0e4069245 
[17:48:30.5220] Service data UUID: 0xfd3d 
[17:48:42.5310] -----------------------------------------------------------------------------------
[17:48:42.5385] Advertised Device mac: c6:c7:52:66:3c:bf 
[17:48:42.5424] Advertised Device name:  
[17:48:42.5444] Advertised Device manufacturer data: 6909c6c752663cbf710f069245 
[17:48:42.5504] Get services data number: 1 
[17:48:42.5524] Service data: 69c0e4069245 
[17:48:42.5555] Service data UUID: 0xfd3d 
[17:48:44.3752] -----------------------------------------------------------------------------------
[17:48:44.3827] Advertised Device mac: dd:bf:df:a5:be:c7 
[17:48:44.3867] Advertised Device name:  
[17:48:44.3887] Advertised Device manufacturer data: 4c0012020003 
[17:48:57.1616] -----------------------------------------------------------------------------------
[... 275 lines cut ...]
[18:04:45.0232] Advertised Device mac: c6:c7:52:66:3c:bf 
[18:04:45.0262] Advertised Device name:  
[18:04:45.0282] Advertised Device manufacturer data: 6909c6c752663cbf730f079245 
[18:04:45.0342] Get services data number: 1 
[18:04:45.0372] Service data: 69c0e4079245 
[18:04:45.0392] Service data UUID: 0xfd3d 
[18:04:55.5739] -----------------------------------------------------------------------------------
[18:04:55.5813] Advertised Device mac: 74:7c:b1:c6:68:c4 
[18:04:55.5853] Advertised Device name:  
[18:04:55.5873] Advertised Device manufacturer data: 4c001005451ce77763 
[18:05:00.7919] -----------------------------------------------------------------------------------
[18:05:00.7988] Advertised Device mac: 3c:91:80:c9:28:58 
[18:05:00.8028] Advertised Device name:  
[18:06:02.1029] -----------------------------------------------------------------------------------
[18:06:02.1102] Advertised Device mac: 66:6a:c4:07:6e:85 
[18:06:02.1142] Advertised Device name:  
[18:06:02.1162] Advertised Device manufacturer data: 4c001005421cfd1e00 
[18:06:05.2001] -----------------------------------------------------------------------------------
[18:06:05.2076] Advertised Device mac: c6:c7:52:66:3c:bf 
[18:06:05.2115] Advertised Device name:  
[18:06:05.2135] Advertised Device manufacturer data: 6909c6c752663cbf730f079245 
[18:06:05.2195] Get services data number: 1 
[18:06:05.2215] Service data: 69c0e4079245 
[18:06:05.2245] Service data UUID: 0xfd3d 
```

xanderificnl, 2022-11-01:
Attached is the log.

Notably, the switchbot got picked up _with the service data_ a total of 113 times. Somehow, only OMG is picking up manufacturer-only data on various ESP32s. So why is OMG seeing manufacturer only data and nothing else is? 

I've run 3 different nimble continuous scanners (the default, yours and mine), ESPHome and Theengs Gateway (on my laptop). 

[serial.log](https://github.com/1technophile/OpenMQTTGateway/files/9913116/serial.log)

1technophile, 2022-11-01:
> Can we get OMG to dump what it sees to serial before it starts processing it?

Yes you would need to set the debug log level to `LOG_LEVEL_TRACE`  [here](https://github.com/1technophile/OpenMQTTGateway/blob/b2a757f28bbc8325e4e4ee77117edecc1ad501ef/main/User_config.h#L496)

Or add this into your pio env `build_flags`:
`  '-DLOG_LEVEL=LOG_LEVEL_TRACE'`

xanderificnl, 2022-11-02:
Cheers, thanks!

---

You can skip forward to *[interesting bit]* below.

---

The final tally of the library functions: 289 broadcasts - all including service data.

Steps:
1. PIO: clean all
2. PIO: clean all all (I'm not sure why that's an option, but sure - why not.)
3. GIT: reset
4. GIT: pull origin/development (clean tree)
5. ESP: erase flash.
6. OMG: Added log level to config. See below for full config.
7. PIO: build (all dependencies are fetched etc.)
8. PIO: upload / flash
9. RCOM: open serial log & tee serial.log.
10. ESP: restart (so we can see OMG boot).

Full config:
```ini
[platformio]
default_envs =
  esp32dev-ble-cont

;esp32 cnt
[env:esp32dev-ble-cont]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager32}
  ${libraries.ble}
  ${libraries.decoder}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=2'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DGateway_Name="kfwnlnfkwl"'
  '-DAttemptBLEConnect=false'
  '-DMQTT_SERVER="172.16.32.2"'
  '-Dwifi_ssid="XiFi"'
  '-Dwifi_password="-snip-"'
  '-DESPWifiManualSetup=true'
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
upload_port = /dev/ttyUSB0
```

----

#### interesting bit

[serial.log](https://github.com/1technophile/OpenMQTTGateway/files/9914591/serial.log) <-- LOG _before my edit_

After not seeing anything standing out, I modified `ZgatewayBT.ino` to include:

```
Log.trace(F("%s" CR), advertisedDevice->toString().c_str());
```

Which yielded the familiar: `T: Name: , Address: c6:c7:52:66:3c:bf, manufacturer data: 6909c6c752663cbfa00f029243`

1. I re-ran the BLE example against NimBLE 1.3.8 (the one used by OMG). The example still returned continuous service data.
2. I bumped OMG's NimBLE to 1.4.1., but that again made no difference. 
3. Made OMG's settings look like those from NimBLE:

```
// config_BT.h
 #ifndef Scan_duration
-#  define Scan_duration 10000 //define the time for a scan
+#  define Scan_duration 0 //define the time for a scan
 #endif
 #ifndef BLEScanInterval
-#  define BLEScanInterval 52 // How often the scan occurs / switches channels; in milliseconds,
+#  define BLEScanInterval 97 // How often the scan occurs / switches channels; in milliseconds,
 #endif
 #ifndef BLEScanWindow
-#  define BLEScanWindow 30 // How long to scan during the interval; in milliseconds.
+#  define BLEScanWindow 37 // How long to scan during the interval; in milliseconds.
```

and this starts to happen:

```
[02:07:38.0405] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-48,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":66,"batt":100}
[02:08:14.4896] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-49,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
[02:09:48.7264] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-48,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
- 02:10 nothing.
[02:11:42.9423] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-48,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
[02:12:31.0727] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-55,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
[02:13:35.2240] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-55,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
- 02:14 nothing
- 02:15 nothing
[02:16:23.5750] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-50,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
- 02:17 nothing
[02:18:34.5287] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-47,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
- 02:19
[02:20:44.8493] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-54,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
- 02:21
[02:22:49.1084] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-49,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
[02:23:29.1938] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-48,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":18.5,"tempf":65.3,"hum":65,"batt":100}
```

and I'm starting to see this (previously unseen):

```
[02:24:25.4183] T: Device mac C6:C7:52:66:3C:BF
[02:24:25.4212] T: Device already discovered or that doesn't require discovery C6:C7:52:66:3C:BF
-truncated-
[02:29:10.5493] N: Device detected: C6:C7:52:66:3C:BF
[02:29:10.5533] T: getDeviceByMac C6:C7:52:66:3C:BF
[02:29:10.5923] T: getDeviceByMac C6:C7:52:66:3C:BF
[02:29:10.5953] T: update C6:C7:52:66:3C:BF
```

At this point, restarting the ESP may result in the Switchbot being seen or not at all (missing the window permanently, it seems). What's strange is that Theengs' Gateway was about to get 16 readings per minute from the Switchbot [via my laptop].

So I'll:

1. re-run Theengs' Gateway from my laptop - to confirm it's still getting lots of readings per minute;
2. re-run the Example using the default values from OMG;
3. fiddle some more with the window & intervals.

Any advice and suggestions are appreciated.

1technophile, 2022-11-02:
I think it is progressing well, thanks to your thorough testing.

If you could try OMG only with this modification:
`+#  define Scan_duration 0 //define the time for a scan`

It would enable pinpointing the continuous scan versus the windows/interval modification as a potential cause.

xanderificnl, 2022-11-02:
So I `git reset'd` and cleaned flash, etc... all the usual steps.

And this is happening (it's 19:11 here atm):

```
$ tail -f ~/serial.log | grep 'Send on /BTtoMQTT/C6C752663CBF msg' --text
[19:06:36.8231] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-23,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":17.6,"tempf":63.68,"hum":63,"batt":100}
[19:07:20.7559] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-25,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":17.6,"tempf":63.68,"hum":63,"batt":100}
[19:08:04.8407] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-23,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":17.6,"tempf":63.68,"hum":63,"batt":100}
[19:08:58.9555] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-23,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":17.6,"tempf":63.68,"hum":63,"batt":100}
[19:09:55.0747] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-25,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":17.6,"tempf":63.68,"hum":63,"batt":100}
[19:10:59.2287] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-25,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":17.6,"tempf":63.68,"hum":63,"batt":100}
[19:11:17.2785] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-23,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":17.6,"tempf":63.68,"hum":63,"batt":100}
```

1technophile, 2022-11-02:
Cool, seems `Scan_duration  0` solve your issue. This could make us reconsider the existing parameters of [esp32dev-ble-cont](https://github.com/1technophile/OpenMQTTGateway/blob/b2a757f28bbc8325e4e4ee77117edecc1ad501ef/platformio.ini#L470).

But doesn't explain why we have missing service data with `esp32dev-ble`

xanderificnl, 2022-11-03:
Running with the default esp32dev-ble at the moment with trace on. 

One thing that stands out is that _good readings_ are processed before the scan ends, and readings without service data are processed after the scan ends. Is it possible that OMG is freeing the data before it's retrieved?

Here's a clean reading for reference:

```
[05:13:25.3536] N: Scan begin
[05:13:27.3417] T: Creating BLE buffer
[05:13:27.3443] N: Device detected: 52:37:AE:7D:14:92
[05:13:27.3473] T: getDeviceByMac 52:37:AE:7D:14:92
[05:13:27.3853] T: No device found 
[05:13:27.3873] N: Send on /BTtoMQTT/5237AE7D1492 msg {"id":"52:37:AE:7D:14:92","mac_type":1,"manufacturerdata":"4c0010050a1c557eb6","rssi":-86,"txpower":12}
[05:13:27.4003] T: jsonPubl - ON
[05:13:27.4013] T: [ OMG->MQTT ] topic: home/esp32-dev-ble-fkenwlnflwe/BTtoMQTT/5237AE7D1492 msg: {"id":"52:37:AE:7D:14:92","mac_type":1,"manufacturerdata":"4c0010050a1c557eb6","rssi":-86,"txpower":12} 
[05:13:31.1258] T: Creating BLE buffer
[05:13:31.1277] N: Device detected: C6:C7:52:66:3C:BF
[05:13:31.1317] T: getDeviceByMac C6:C7:52:66:3C:BF
[05:13:31.1348] T: Get services data number: 1
[05:13:31.1377] T: Converted service data (6) to 69c0e406903f
[05:13:31.1417] T: Service data: 69c0e406903f
[05:13:31.1437] T: Service data UUID: 0xfd3d
[05:13:31.1668] T: Decoder found device: THX1/W230150X
[05:13:31.1708] T: getDeviceByMac C6:C7:52:66:3C:BF
[05:13:31.1737] T: update C6:C7:52:66:3C:BF
[05:13:31.1757] T: Removing servicedatauuid
[05:13:31.1787] T: Removing servicedata
[05:13:31.1807] T: Removing manufacturerdata
[05:13:31.1828] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-25,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":16.6,"tempf":61.88,"hum":63,"batt":100}
[05:13:31.2007] T: jsonPubl - ON
[05:13:31.2029] T: [ OMG->MQTT ] topic: home/esp32-dev-ble-fkenwlnflwe/BTtoMQTT/C6C752663CBF msg: {"id":"C6:C7:52:66:3C:BF","mac_type":1,"rssi":-25,"brand":"SwitchBot","model":"Meter (Plus)","model_id":"THX1/W230150X","tempc":16.6,"tempf":61.88,"hum":63,"batt":100} 
[05:13:32.0078] T: Creating BLE buffer
[05:13:32.0098] N: Device detected: 52:37:AE:7D:14:92
[05:13:32.0138] T: getDeviceByMac 52:37:AE:7D:14:92
[05:13:32.0519] T: No device found 
[05:13:32.0538] N: Send on /BTtoMQTT/5237AE7D1492 msg {"id":"52:37:AE:7D:14:92","mac_type":1,"manufacturerdata":"4c0010050a1c557eb6","rssi":-86,"txpower":12}
[05:13:32.0658] T: jsonPubl - ON
[05:13:32.0678] T: [ OMG->MQTT ] topic: home/esp32-dev-ble-fkenwlnflwe/BTtoMQTT/5237AE7D1492 msg: {"id":"52:37:AE:7D:14:92","mac_type":1,"manufacturerdata":"4c0010050a1c557eb6","rssi":-86,"txpower":12} 
[05:13:35.3618] N: Found 2 devices, scan number 24 end
[05:13:35.3652] T: Process BLE stack free: 3068
[05:13:35.3682] T: macAdr C6:C7:52:66:3C:BF
[05:13:35.3702] T: macType 1
[05:13:35.3712] T: isDisc 1
[05:13:35.3722] T: isWhtL 0
[05:13:35.3732] T: isBlkL 0
[05:13:35.3752] T: connect 0
[05:13:35.3762] T: sensorModel_id 39
[05:13:35.3772] T: macAdr 6D:50:C9:F8:3B:43
[05:13:35.3802] T: macType 1
[05:13:35.3812] T: isDisc 0
[05:13:35.3822] T: isWhtL 0
[05:13:35.3832] T: isBlkL 0
[05:13:35.3842] T: connect 0
[05:13:35.3852] T: sensorModel_id 55
[05:14:16.6461] T: retrieving value of system characteristics Uptime (s):1569
```

Here's a bad reading. Ever since starting to look into this problem, I've seen garbled serial output, and I assumed it was simply the serial connection. Yet, as far as I can tell, proper readings are outputted nice and tidy and bad readings are generally accompanied by garbled output.

```
Scan begins [05:16:42.1104] N: Scan begin
[05:16:43.1864] T: Creating BLE buffer
[05:16:43.1884] N: Device detected: DF:AE:6F:EF:AA:86
[05:16:43.1924] T: getDeviceByMac DF:AE:6F:EF:AA:86
[05:16:43.2304] T: No device found 
[05:16:43.2324] N: Send on /BTtoMQTT/DFAE6FEFAA86 msg {"id":"DF:AE:6F:EF:AA:86","mac_type":1,"manufacturerdata":"4c0012020003","rssi":-56}
[05:16:43.2434] T: jsonPubl - ON
[05:16:43.2444] T: [ OMG->MQTT ] topic: home/esp32-dev-ble-fkenwlnflwe/BTtoMQTT/DFAE6FEFAA86 msg: {"id":"DF:AE:6F:EF:AA:86","mac_type":1,"manufacturerdata":"4c0012020003","rssi":-56} 
[05:16:48.4070] T: Creating BLE buffer
[05:16:48.4090] N: Device detected: 7F:14:32:27:45:54
[05:16:48.4130] T: getDeviceByMac 7F:14:32:27:45:54
[05:16:48.4510] T: No device found 
[05:16:48.4530] N: Send on /BTtoMQTT/7F1432274554 msg {"id":"7F:14:32:27:45:54","mac_type":1,"manufacturerdata":"4c001005021c03f543","rssi":-86,"txpower":12}
[05:16:48.4659] T: jsonPubl - ON
[05:16:48.4670] T: [ OMG->MQTT ] topic: home/esp32-dev-ble-fkenwlnflwe/BTtoMQTT/7F1432274554 msg: {"id":"7F:14:32:27:45:54","mac_type":1,"manufacturerdata":"4c001005021c03f543","rssi":-86,"txpower":12} 
[05:16:52.1188] T: Creating BLE buffer
Scan ends: [05:16:52.1214] N: Found 4 devices, scan number 27 end
First Switchbot mention [05:16:52.1244] N: Device detected: C6:C7:52:66:3C:BF
Garbled (T:) [05:16:52.1274] T: getDeT: viceByMac C6:C7:52:66:3C:BF 
Garbled (T:) [05:16:52.1314] Process BLE stack free: 3068
[05:16:52.1336] T: macAdr C6:C7:52:66:3C:BF
[05:16:52.1364] T: macType 1
[05:16:52.1376] T: isDisc 1
[05:16:52.1384] T: isWhtL 0
[05:16:52.1394] T: isBlkL 0
[05:16:52.1404] T: connect 0
[05:16:52.1414] T: sensorModel_id 39
[05:16:52.1434] T: macAdr 6D:50:C9:F8:3B:43
[05:16:52.1454] T: macType 1
[05:16:52.1464] T: isDisc 0
[05:16:52.1483] T: isWhtL 0
[05:16:52.1494] T: isBlkL 0
[05:16:52.1504] T: connect 0
[05:16:52.1514] T: sensorModel_id 55
[05:16:52.1634] T: No device found 
[05:16:52.1654] T: Creating BLE buffer
[05:16:52.1675] N: Device detected: 3C:91:80:C9:28:58
[05:16:52.1714] T: getDeviceByMac 3C:91:80:C9:28:58
[05:16:52.1745] T: No device found 
Sensorless reading [05:16:52.1754] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf970f07903f","rssi":-24}
```

h2zero, 2022-11-03:
Very interesting from the log:
```
[90m[23:15:56.0996][0m N: Scan begin
[90m[23:16:06.1060][0m T: Creating BLE buffer
[90m[23:16:06.1086][0m N: Found 5 devices, scan number N: D7evice detected:  C6:C7:52:66:3C:BF
[90m[23:16:06.1146][0m end
[90m[23:16:06.1146][0m T: getDeviceByMac C6:C7:52:66:3C:BF
[90m[23:16:06.1176][0m T: Process BLE stack free: 3068
```
Looks like the switchbot was only just detected as the scan ended and results in missing data. What I suspect is going on is that the switchbot advertisement was detected but no scan response had been received when the scan ended. When this occurs NimBLE will invoke the callback with whatever data it does have, complete or not, which in this case it was not. 

Since the device advertises about every 10 seconds, I would adjust the scan duration to 15 seconds to allow more time to get the scan response data.

xanderificnl, 2022-11-03:
Currently running w/   '-DScan_duration=15000'

Results in:

```
[19:59:32.7484] T: Creating BLE buffer
[19:59:32.7504] N: Device detected: C6:C7:52:66:3C:BF
[19:59:32.7534] T: getDeviceByMac C6:C7:52:66:3C:BF
[19:59:32.7573] N: Found 6 devices, scan number 1 end
[19:59:32.7603] T: Process BLE stack free: 3068
[19:59:32.7633] N: BLE Connect begin
[19:59:32.7643] N: BLE Connect end
[19:59:32.7938] T: No device found 
```

I am fairly sure the Switchbot advertises a bit more than once per 10 seconds. Theengs' Gateway on my laptop picked up 16 sensor readings a minute, so that's about once per 3-4 seconds.

h2zero, 2022-11-03:
Thanks for that, could you try one more thing? I'm curious what happens if you set the scan duration to 6 seconds instead. It's possible that this is caused by NimBLE when seeing the device multiple times during the scan.

xanderificnl, 2022-11-03:
Scan_duration to 6000

Results:

```
[20:14:57.9009] N: Found 3 devices, scan number 1 end
[20:14:57.9049] N: Device detected: 64:F6:5B:4E:54:2B
[20:14:57.9079] T: getDevT: Process BLE stack freie: ceByMac 64:F6:5B:4E:54:2B
[20:14:57.9129] 3068
[20:14:57.9139] N: BLE Connect begin
[20:14:57.9159] N: BLE Connect end
[20:14:57.9459] T: No device found 
[20:14:57.9478] T: Creating BLE buffer
[20:14:57.9499] N: Device detected: C6:C7:52:66:3C:BF
[20:14:57.9529] T: getDeviceByMac C6:C7:52:66:3C:BF
[20:14:57.9894] T: No device found 
```

With 2 seconds:

```
[20:16:45.2162] T: Creating BLE buffer
[20:16:45.2187] N: Found 3 devices, scan number 1 end
[20:16:45.2217] N: Device detected: 64:F6:5B:4E:54:2B
[20:16:45.2247] T: getDeviceByMac 64:F6:5B:4E:54:2B
[20:16:45.2287] T: Process BLE stack free: 3068
[20:16:45.2308] N: BLE Connect begin
[20:16:45.2327] N: BLE Connect end
[20:16:45.2607] T: No device found 
[20:16:45.2628] T: Creating BLE buffer
[20:16:45.2647] N: Device detected: C6:C7:52:66:3C:BF
[20:16:45.2688] T: getDeviceByMac C6:C7:52:66:3C:BF
[20:16:45.3052] T: No device found 
```

---
FWIW, a comparison of OMG & Theengs from MQTT:

[image]

h2zero, 2022-11-03:
Hmm, not really what I expected but here is a NimBLE branch to try: https://github.com/h2zero/NimBLE-Arduino/tree/adv-data-fix

xanderificnl, 2022-11-03:
To confirm:

```
Removing unused dependencies...
Library Manager: Removing NimBLE-Arduino @ 1.3.8+sha.4a7529e
Library Manager: NimBLE-Arduino@1.3.8+sha.4a7529e has been removed!
Library Manager: Installing git+https://github.com/h2zero/NimBLE-Arduino.git#adv-data-fix

# and

|-- NimBLE-Arduino @ 1.4.1+sha.4502d2b # sum of adv-data-fix commit
```

Running with 10s:

```
# 1
[21:21:07.0103] N: Scan begin
[21:21:14.5211] T: Creating BLE buffer
[21:21:14.5237] N: Device detected: DF:8B:8A:97:46:24
[21:21:14.5267] T: getDeviceByMac DF:8B:8A:97:46:24
[21:21:14.5657] T: No device found 
[21:21:14.5677] N: Send on /BTtoMQTT/DF8B8A974624 msg {"id":"DF:8B:8A:97:46:24","mac_type":1,"manufacturerdata":"4c0012020003","rssi":-54}
[21:21:14.5776] T: jsonPubl - ON
[21:21:14.5797] T: [ OMG->MQTT ] topic: home/reee/BTtoMQTT/DF8B8A974624 msg: {"id":"DF:8B:8A:97:46:24","mac_type":1,"manufacturerdata":"4c0012020003","rssi":-54} 
[21:21:17.0205] T: Creating BLE buffer
[21:21:17.0229] N: Found 5 devices, scan number 1 end
[21:21:17.0260] N: DeviceT: Process BLE stack fre e: detect3100
[21:21:17.0300] ed: C6:C7:52:66:3C:BF
[21:21:17.0319] TN: : BgetDeviceByMac C6:C7:52:66:3C:BF
[21:21:17.0359] LE Connect begin
[21:21:17.0369] N: BLE Connect end
[21:21:17.0670] T: No device found 
[21:21:17.0690] T: Creating BLE buffer
[21:21:17.0710] N: Device detected: 64:D5:24:BE:5B:6E
[21:21:17.0740] T: getDeviceByMac 64:D5:24:BE:5B:6E
[21:21:17.1110] T: No device found 
[21:21:17.1129] T: Creating BLE buffer
[21:21:17.1149] N: Device detected: 7B:24:FD:A4:6D:89
[21:21:17.1180] T: getDeviceByMac 7B:24:FD:A4:6D:89
[21:21:17.1544] T: No device found 
[21:21:17.1570] T: Creating BLE buffer
[21:21:17.1590] N: Device detected: 68:99:7A:56:0E:B5
[21:21:17.1620] T: getDeviceByMac 68:99:7A:56:0E:B5
[21:21:17.1990] T: Decoder found device: IBEACON
[21:21:17.2019] T: getDeviceByMac 68:99:7A:56:0E:B5
[21:21:17.2050] T: add 68:99:7A:56:0E:B5
[21:21:17.2079] T: Removing servicedatauuid
[21:21:17.2100] T: Removing servicedata
[21:21:17.2120] T: Removing manufacturerdata
[21:21:17.2150] T: Device mac 68:99:7A:56:0E:B5
[21:21:17.2169] T: Device already discovered or that doesn't require discovery 68:99:7A:56:0E:B5
[21:21:17.2240] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf250f099043","rssi":-28}
[21:21:17.2359] T: jsonPubl - ON
[21:21:17.2380] T: [ OMG->MQTT ] topic: home/reee/BTtoMQTT/C6C752663CBF msg: {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf250f099043","rssi":-28} 
[21:21:17.2519] N: Send on /BTtoMQTT/64D524BE5B6E msg {"id":"64:D5:24:BE:5B:6E","mac_type":1,"manufacturerdata":"4c001005071cbe93a6","rssi":-90,"txpower":12}
[21:21:17.2639] T: jsonPubl - ON
[21:21:17.2660] T: [ OMG->MQTT ] topic: home/reee/BTtoMQTT/64D524BE5B6E msg: {"id":"64:D5:24:BE:5B:6E","mac_type":1,"manufacturerdata":"4c001005071cbe93a6","rssi":-90,"txpower":12} 
[21:21:17.2800] N: Send on /BTtoMQTT/7B24FDA46D89 msg {"id":"7B:24:FD:A4:6D:89","mac_type":1,"manufacturerdata":"0106010690caae1b75","rssi":-79}
[21:21:17.2910] T: jsonPubl - ON
[21:21:17.2930] T: [ OMG->MQTT ] topic: home/reee/BTtoMQTT/7B24FDA46D89 msg: {"id":"7B:24:FD:A4:6D:89","mac_type":1,"manufacturerdata":"0106010690caae1b75","rssi":-79} 
[21:21:17.3060] N: Send on /BTtoMQTT/68997A560EB5 msg {"id":"68:99:7A:56:0E:B5","mac_type":1,"rssi":-86,"brand":"GENERIC","model":"iBeacon","model_id":"IBEACON","mfid":"4c00","uuid":"08dd9d6bc15e630f4ec5c1d452d21005","major":23068,"minor":22952,"txpower":-48}
[21:21:17.3270] T: jsonPubl - ON
[21:21:17.3290] T: [ OMG->MQTT ] topic: home/reee/BTtoMQTT/68997A560EB5 msg: {"id":"68:99:7A:56:0E:B5","mac_type":1,"rssi":-86,"brand":"GENERIC","model":"iBeacon","model_id":"IBEACON","mfid":"4c00","uuid":"08dd9d6bc15e630f4ec5c1d452d21005","major":23068,"minor":22952,"txpower":-48} 

# 2
[21:22:12.5977] N: Scan begin
[21:22:22.6073] T: Creating BLE buffer
[21:22:22.6103] N: Found 1 devices, scan number 2 end
[21:22:22.6133] N: Device detected: C6:C7:52:66:3C:BF
[... 118 lines cut ...]
[21:29:28.3549] T: getDeviceByMac 76:1E:17:B5:02:BE
[21:29:28.3579] T: Process BLE stack free: 3084
[21:29:28.3609] T: macAdr 68:99:7A:56:0E:B5
[21:29:28.3629] T: macType 1
[21:29:28.3639] T: isDisc 0
[21:29:28.3649] T: isWhtL 0
[21:29:28.3659] T: isBlkL 0
[21:29:28.3669] T: connect 0
[21:29:28.3679] T: sensorModel_id 55
[21:29:28.3699] T: macAdr C6:C7:52:66:3C:BF
[21:29:28.3729] T: macType 1
[21:29:28.3739] T: isDisc 1
[21:29:28.3749] T: isWhtL 0
[21:29:28.3759] T: isBlkL 0
[21:29:28.3769] T: connect 0
[21:29:28.3779] T: sensorModel_id 39
[21:29:28.3899] T: No device found 
[21:29:28.3919] N: Send on /BTtoMQTT/761E17B502BE msg {"id":"76:1E:17:B5:02:BE","mac_type":1,"manufacturerdata":"4c001005071caa5cb1","rssi":-88,"txpower":12}
[21:29:28.4039] T: jsonPubl - ON
[21:29:28.4059] T: [ OMG->MQTT ] topic: home/reee/BTtoMQTT/761E17B502BE msg: {"id":"76:1E:17:B5:02:BE","mac_type":1,"manufacturerdata":"4c001005071caa5cb1","rssi":-88,"txpower":12} 
```

Does NimBLE have an underlying flag to enable trace logs? From outside tests (nimble examples, Theengs', ESPhome, etc.) we know the service data is always transmitted; but it'd be nice if we can literally tell the data was picked up by OMG at some point.

Can we dump a representation of NimBLEAdvertisedDevice as it is created? Life would be so much simpler if we could debug with gdb.

h2zero, 2022-11-03:
Thanks for testing, you can enable NimBLE logs with `'-DCONFIG_NIMBLE_CPP_LOG_LEVEL=5'.

xanderificnl, 2022-11-03:
Thanks for all the hard work you've given to the community and the help you're providing me. It's much appreciated :)

Initially I inserted this into OMG's callback:

```c#
    char* payload = BLEUtils::buildHexData(NULL, (uint8_t*)advertisedDevice->getPayload(), advertisedDevice->getPayloadLength());
    Log.trace(F("[payload] %s:  %s" CR), advertisedDevice->getAddress().toString().c_str(), payload);
```

That resulted in:

```
[22:16:14.4896] T: [payload] c6:c7:52:66:3c:bf:  0201060eff6909c6c752663cbf300f009142
```

At this point, I figured I could just copy an example by technophile, and insert that:
_I did adjust the servicedata to use buildHexData but I left the serial.println & prinft's since I figured they don't really matter_
```c#
    Serial.println("begin-----------------------------------------------------------------------------------");
    Serial.printf("Advertised Device mac: %s \n", advertisedDevice->getAddress().toString().c_str());
    Serial.printf("Advertised Device name: %s \n", (char*)advertisedDevice->getName().c_str());
    if (advertisedDevice->haveManufacturerData()) {
      char* manufacturerdata = BLEUtils::buildHexData(NULL, (uint8_t*)advertisedDevice->getManufacturerData().data(), advertisedDevice->getManufacturerData().length());
      Serial.printf("Advertised Device manufacturer data: %s \n", manufacturerdata);
      free(manufacturerdata);
    }
    if (advertisedDevice->haveServiceData()) {
      int serviceDataCount = advertisedDevice->getServiceDataCount();
      Serial.printf("Get services data number: %d \n", serviceDataCount);
      for (int j = 0; j < serviceDataCount; j++) {
        std::string service_data = NimBLEUtils::buildHexData(nullptr, (uint8_t*)advertisedDevice->getServiceData().data(), advertisedDevice->getServiceData().length());
        Serial.printf("Service data: %s \n", service_data.c_str());
        std::string serviceDatauuid = advertisedDevice->getServiceDataUUID(j).toString();
        Serial.printf("Service data UUID: %s \n", (char*)serviceDatauuid.c_str());
      }
    }
    Serial.println("end-----------------------------------------------------------------------------------");
```

also updated the queue sender to tell us when that happens, i.e.

```c#
    if (xQueueSend(BLEQueue, &advertisedDevice, 0) != pdTRUE) {
      Log.error(F("BLEQueue full" CR));
    } else {
      Log.trace(F("[xQueueSend] %s" CR), advertisedDevice->getAddress().toString().c_str());
    }
```

And here's a good reading:

```
[22:18:38.6682] I NimBLEScan: New advertiser: c6:c7:52:66:3c:bf
[22:18:38.6728] I NimBLEScan: Updated advertiser: c6:c7:52:66:3c:bf
[22:18:38.6777] begin-----------------------------------------------------------------------------------
[22:18:38.6848] Advertised Device mac: c6:c7:52:66:3c:bf 
[22:18:38.6888] Advertised Device name:  
[22:18:38.6907] Advertised Device manufacturer data: 6909c6c752663cbf310f009143 
[22:18:38.6968] Get services data number: 1 
[22:18:38.6998] Service data: 69c0e4009143 
[22:18:38.7018] Service data UUID: 0xfd3d 
[22:18:38.7038] end-----------------------------------------------------------------------------------
[22:18:38.7118] T: Creating BLE buffer
garbled [22:18:38.7138] T: [xQueueSend] Nc6:c7:52:66:3c:bf
[22:18:38.7168] : Device detected: C6:C7:52:66:3C:BF
[22:18:38.7198] T: getDeviceByMac C6:C7:52:66:3C:BF
[22:18:38.7228] T: Get services data number: 1
[22:18:38.7258] T: Converted service data (6) to 69c0e4009143
[22:18:38.7298] T: Service data: 69c0e4009143
[22:18:38.7328] T: Service data UUID: 0xfd3d
[... 30 lines cut ...]
[22:19:41.6809] T: [ OMG->MQTT ] topic: home/reee/BTtoMQTT/46DC327156FA msg: {"id":"46:DC:32:71:56:FA","mac_type":1,"manufacturerdata":"4c0010050a1cd1cf54","rssi":-90,"txpower":12} 
[22:19:44.5466] D NimBLEScan: discovery complete; reason=0
[22:19:44.5512] begin-----------------------------------------------------------------------------------
[22:19:44.5592] Advertised Device mac: c6:c7:52:66:3c:bf 
[22:19:44.5622] Advertised Device name:  
[22:19:44.5642] Advertised Device manufacturer data: 6909c6c752663cbf310f009143 
[22:19:44.5702] end-----------------------------------------------------------------------------------
[22:19:44.5782] T: Creating BLE buffer
garbled [22:19:44.5801] TN: : Device det[ected: xQuC6:C7:52:66:3C:BF
garbled [22:19:44.5841] eT: getDeviceByMac C6:C7:52:66:3C:BF
garbled [22:19:44.5872] ueSend] c6:c7:52:66:3c:bf
[22:19:44.5892] N: Found 2 devices, scan number 2 end
[22:19:44.5921] T: Process BLE stack free: 3084
[22:19:44.5952] T: macAdr C6:C7:52:66:3C:BF
[22:19:44.5982] T: macType 1
[22:19:44.5992] T: isDisc 1
[22:19:44.6002] T: isWhtL 0
[22:19:44.6012] T: isBlkL 0
[22:19:44.6022] T: connect 0
[22:19:44.6032] T: sensorModel_id 39
[22:19:44.6187] T: No device found 
[22:19:44.6212] N: Send on /BTtoMQTT/C6C752663CBF msg {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf310f009143","rssi":-37}
[22:19:44.6332] T: jsonPubl - ON
[22:19:44.6342] T: [ OMG->MQTT ] topic: home/reee/BTtoMQTT/C6C752663CBF msg: {"id":"C6:C7:52:66:3C:BF","mac_type":1,"manufacturerdata":"6909c6c752663cbf310f009143","rssi":-37} 
```

h2zero, 2022-11-03:
You're welcome 😀 and thanks for your efforts as well, that's some great detailed troubleshooting!

Now for the issue at hand.

From the good reading:
```
[22:18:38.6682] I NimBLEScan: New advertiser: c6:c7:52:66:3c:bf  -- advertisement data stored here (manufacturer data is in this)
[22:18:38.6728] I NimBLEScan: Updated advertiser: c6:c7:52:66:3c:bf  -- scan response data is stored here (this is where the missing service data comes from) 
```

From the bad reading:
```
[22:19:40.8088] I NimBLEScan: New advertiser: c6:c7:52:66:3c:bf  -- advertisement data, all good here
************************************ missing scan response data here *********************************
[22:19:41.5864] I NimBLEScan: New advertiser: 46:dc:32:71:56:fa -- other device advertisement data
[22:19:41.5909] I NimBLEScan: Updated advertiser: 46:dc:32:71:56:fa -- other device scan response data
```

As to why this is happening, this would be incredibly hard to know for certain with logs alone. This could be due to a number of factors such as the device simply didn't receive the scan response request, or went to sleep. On our side there could have been an interruption in the receiver at the moment it was sent (think WiFi, due to shared antenna), radio interference, BLE stack bugs or bad timing.

The interesting thing about the applications that do work is that they are not simultaneously processing WiFi packets while scanning BLE, it would not surprise me if this was ultimately the root cause. Potential improvement might be provided by setting `BLEScanInterval` and  `BLEScanWindow` to the same value (something like 37), this will cause the BLE stack to consume the maximum amount of antenna time, but may cause a WiFi disconnect.

xanderificnl, 2022-11-03:
For what it's worth, the ESP and the switchbot are within 20 cm of each other. 

I'll try the window + interval shortly. We might also try disabling wifi fully (not sure how easy that'd be with OMG) since we can still see all the data over serial.

Thanks :)

h2zero, 2022-11-03:
I'm not sure about fully disabling WiFi, could be quite difficult. Hopefully there is some improvement with those settings though, let me know if it helps.

xanderificnl, 2022-11-04:
I've just realized you may have missed that message. Technophile and I discovered that a scan_interval of 0 with the default OMG settings yielded at least one reading a minute, and sometimes two. During that run there were max 5 missing readings in a timeframe of a couple of hours. See [this message](https://github.com/1technophile/OpenMQTTGateway/issues/1288#issuecomment-1301034227).

Anyway, setting the scan interval & window to 37 does seem to work as well. I've seen a couple serviceless data so far, but mostly readings. So scan interval 0 still seems to pick up more and at slightly higher intervals (i.e. it takes about 60-70 seconds for each reading w/ these settings). 

I've yet to see a wifi or mqtt disconnect. If there's anything else you'd like me to try/run, I'm more than happy to narrow down this issue further. 

With the way OMG's output gets garbled, I've also wondered if maybe the multithreading may be causing things to run amok - but yesterday (or earlier) I already disabled the 2 lines switching/disabling/enabling cores and that made no difference.

h2zero, 2022-11-04:
Glad those settings made an improvement! Yes, setting the scan duration to 0 for continuous scanning would work as well simply due to the scan having more opportunities to receive all the data.

I'm not sure how much better we can get than this, if I think of anything else I'll let you know.

xanderificnl, 2022-11-05:
Cheers, thanks everyone!

[2 more replies]