1technophile/OpenMQTTGateway #1293: Puck JS integration

hallard, 2022-10-19:
Hi @DigiH 

As asked, just sorted out my puck.js it's an espruino device, and need to be programmed (in js) with an application. 
So what app you want me to try?

There are so [much](https://espruino.github.io/EspruinoApps/) may be `Bluetooth Advertiser - Everything! `

hallard, 2022-10-20:
Installed this app and here are results

Source code of the app is [here](https://github.com/espruino/EspruinoApps/tree/master/apps/bleeverything) it may help for encoding format :-)

```json
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-71,"servicedata":"303030","servicedatauuid":"0x183d"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-43,"servicedata":"00","servicedatauuid":"0x183c"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-77,"servicedata":"4a","servicedatauuid":"0x1809"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-30,"servicedata":"303030","servicedatauuid":"0x183d"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-82,"brand":"GENERIC","model":"Service data","model_id":"ServiceData","batt":100}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-36,"servicedata":"303030","servicedatauuid":"0x183d"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-29,"servicedata":"303030","servicedatauuid":"0x183d"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-77,"servicedata":"302e3234","servicedatauuid":"0x182a"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-28,"servicedata":"303030","servicedatauuid":"0x183d"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 6","rssi":-85,"servicedata":"302e3231303136353032333830","servicedatauuid":"0x182b"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-28,"servicedata":"303030","servicedatauuid":"0x183d"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-81,"servicedata":"303030","servicedatauuid":"0x183d"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-35,"servicedata":"00","servicedatauuid":"0x183c"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-78,"servicedata":"00","servicedatauuid":"0x183c"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-30,"servicedata":"00","servicedatauuid":"0x183c"}
{"id":"FD:52:F6:D9:66:F9","mac_type":1,"name":"Puck.js 66f9","rssi":-80,"servicedata":"00","servicedatauuid":"0x183c"}
```

DigiH, 2022-10-20:
Thanks @hallard , will have a look at these and the app source. Once we got this **everything** sorted maybe we can look at the others with their individual details, to see if they are just a subset of the encodings, or if they have their own encoding rules.

hallard, 2022-10-20:
BTW as far as I remember battery life duration is not awesome, but depends a lot of application used and advertisement frequency, I'll see how long this one live.

DigiH, 2022-10-20:
@hallard Could you use the `esp32dev-ble-datatest` environment/binary again for logging for the Puck.js?

The battery seems to already be recognised by the standard ServiceData decoder, possibly some of the others also follow the Bluetooth SIG service conventions, although `0x1809` (Health Thermometer) with **4a** -> 74º would only make sense if you had the Puck.js set to report in Fahrenheit - or in the oven ;)

Will have to look at the app source though for a better understanding.

DigiH, 2022-10-21:
> `0x1809` (Health Thermometer) with **4a** -> 74º

So yes, the **Everything** app broadcasts the temperature in Fahrenheit by default. So that's a first. And only in the **Everything** app, it seems, with the temp only app broadcasting in Celsius - hmm.

And all the others seem pretty clear as well, but would need more testing and logging.

DigiH, 2022-10-21:
@hallard - Looking more at the different apps, it seems clear that they are more example apps to follow for creating your own, than ready to go for general use apps. The Fahrenheit temperature broadcast being one example here, also using the Health Thermometer service uuid instead of the Temperature one makes us believe that it is not really worth creating a decoder for what are just example apps.

If you feel like creating a ready for release app to work with the different Puck.js versions, adhering to the standard service uuids, it might be a different case, if you made that publicly available for use by other Puck.js users.

DigiH, 2022-10-26:
Hi @bouton,

while it will be unlikely, for the above mentioned reasons of most everyone with a Puck.js having their own applications on them, to include an official [Decoder](https://github.com/theengs/decoder) for use with OpenMQTTGateway, your iBeacon Bangle should already be detected and decoded as an iBeacon for presence detection.

For you individual Puck.js you could define your own decoder by following the [Theengs Decoder documentation](https://decoder.theengs.io/participate/adding-decoders.html) to include in an individual build for OpenMQTTGateway.

If you are actually looking to reduce your devices, you are using Home Assistant and you are mainly interested in Bluetooth device recognition, then [Theengs Gateway](https://github.com/theengs/gateway) might be the way to go for your use case - even more easily integrated in HA with the [Theengs Gateway Add-on](https://github.com/theengs/addon-theengsgw).

While it is impossible to include a generic Puck.js decoder, we are trying to implement more generally used standard Bluetooth SIG services/characteristics, which many devices use. So far this only includes the 0x180F/0x2A19 - battery level.

I hope this explains the situation a bit and help you in finding the most suitable solution for your use case.

bouton, 2022-10-26:
Thanks. I looked into Theengs and was a bit overwhelmed. Will look again,
taking a deep breath.
As for Bluetooth SIG services/characteristics that seems an obvious
addition. Looking forward to all/some additions.

Thanks

alfista2600, 2023-06-05:
getting back to my own puck.js as I have a particular use case. I've been able to send it bluetooth commands using EspruinoHub on a Pi Zero but I'd rather send it commands through an OMG device. Is this possible with a relatively little amount of effort?

DigiH, 2023-06-05:
Hi @alfista2600 - [sending MQTT commands is already possible](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt) with OpenMQTTGateway, so you should also be able to send these through your OMG gateway, unless sending the commands requires a PIN, which is not yet implemented.

Hope this helps.

closed 2023-10-03 by github-actions[bot]