1technophile/OpenMQTTGateway #1294: How to push data to BLE device?

Leadrive, 2022-10-20:
Hi, the BLE device can send BTtoMQTT to MQTT.

Now, I want to push data to BLE device from MQTT, how to do?
thank you.

1technophile, 2022-10-20:
Hello,

You have more information below:
https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt-esp32-only

Leadrive, 2022-10-20:
@1technophile 
Thanks your reply.
Yes, I test the esp32, it can get MQTTtoBT command.

"ble_write_value":"TEST",

I want to know the esp32 how to tranfer the data TEST to BLE device?

Leadrive, 2022-10-20:
I use the BLE device PHY6252 PB-03 module, it have a USART connector to send or receive command.
I want the BLE device can send and receive the MQTT message(user define data) throught Gateway.

1technophile, 2022-10-20:
I don't know this BLE module, but it seems that you will need to identify the service and characteristics you want to write in (if they exist). Either program the module with your firmware and define the services and characteristics.

1technophile, 2022-10-21:
Closing this, if you have other questions feel free to jump in the [forum](https://community.openmqttgateway.com/)

closed 2022-10-21 by 1technophile