1technophile/OpenMQTTGateway #1302: LYWSD03MMC_ATC wrong values

antibill51, 2022-10-25:
**Describe the bug**
I have value spikes, and false devices with LYWSD03MMC_ATC

**To Reproduce**
```
[env:esp32dev-ble-cont-perso]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
  ${libraries.wifimanager32}
  ${libraries.decoder}

build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=0'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DESPWifiManualSetup=true'
  '-DMQTT_USER="xx"'
  '-DMQTT_PASS="xx"'
  '-DMQTT_SERVER="xx"'
  '-Dwifi_ssid="xx"'
  '-Dwifi_password="xx"'
  '-DOMG_VERSION="0.9.16beta"'
  '-DTimeBtwRead=0'
  '-DAttemptBLECOnnect=true'
  '-DScan_duration=5000'
  '-DActiveBLEScan=true'
  '-DHassPresence=true'
  '-DGateway_Name="Garage"'
upload_port = 192.168.2.24
upload_protocol = espota
upload_flags =
  --auth=OTAPASSWORD
  --port=8266
upload_speed = 512000
monitor_speed = 115200
```

**Expected behavior**
A filter on non-standard values?

**Screenshots**
[image]

**Environment (please complete the following information):**
OpenMQTTGateway development version Oct 14, 2022
* Library : 
TheengsDecoder   0.9.0
NimBLE-Arduino 1.3.8
* LYWSD03MMC_ATC B1.9 firmware :
 pvvx/ATC_MiThermometer ATC_v38.bin in ATC441 mode.

**Additional context**
Nothing

1technophile, 2022-10-25:
Thanks for this detailed report, I see that the v38 is more than [one year old](https://github.com/atc1441/ATC_MiThermometer/releases/tag/38), on the ATC side, and the v3.8 doesn't exist on PVVX side https://github.com/pvvx/ATC_MiThermometer/releases. Not sure of the version you are using, do you have a link?

DigiH, 2022-10-25:
Hi @antibill51,

would it also be possible for you to activate the publishing of the raw servicedata, by sending

`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"pubKnownServiceData":true}'`

then monitor the OMG output with [MQTT-Explorer](https://github.com/thomasnordquist/MQTT-Explorer/releases/tag/v0.3.5) for a while, and copy and paste the offending spiked values with their servicedata here for further analysis?

Thanks

antibill51, 2022-10-30:
Hi @DigiH,

The first is good, the second and the third are wrong :

```
{
"id":"A4:C1:38:B3:B3:50",
"mac_type":0,
"name":"Salon",
"rssi":-89,
"distance":21.51847,
"servicedata":"a4c138b3b35000d8462e0a3c7b",
"brand":"Xiaomi",
"model":"LYWSD03MMC",
"model_id":"LYWSD03MMC_ATC",
"tempc":21.6,
"tempf":70.88,
"hum":70,
"batt":46,
"volt":2.62
}
```

```
{
'id': 'A4:C1:38:B3:B3:50', 
'mac_type': 0, 
'rssi': -79, 
'distance': 8.651812, 
'servicedata': 'a4c138b3b37856adbaa13bfe3f', 
'brand': 'Xiaomi', 
'model': 'LYWSD03MMC', 
'model_id': 'LYWSD03MMC_ATC', 
'tempc': 2218.9, 
'tempf': 4026.02, 
'hum': -70, 
'batt': -95, 
'volt': 15.358
}
```

```
{
'id': 'A4:C1:38:B3:B3:50', 
'mac_type': 0, 
'rssi': -82, 
'distance': 11.4949, 
'servicedata': 'a4c138b3b37856adba753afe3f', 
'brand': 'Xiaomi', 
'model': 'LYWSD03MMC', 
'model_id': 'LYWSD03MMC_ATC', 
'tempc': 2218.9, 
'tempf': 4026.02, 
'hum': -70, 
'batt': 117, 
'volt': 15.102
}
```

False positive device MAC address :
3A89BAAD5678
2C2138BAAD56
38543FFE31B5

Real device MAC address:
A4C138B3B350
A4C1383C8895
A4C138477FFA
all start with A4C138?

DigiH, 2022-10-30:
Thanks @antibill51 , this will help to pin down the problem with the wrong recognitions, unfortunately I forgot to also ask for you to activate the servicedatauuid as well, could you also

`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"pubServiceDataUUID":true}'`

for some additional messages like the above which include the servicedatauuid?

Thanks a lot.

DigiH, 2022-10-30:
> False positive device MAC address :
> 3A89BAAD5678
> 2C2138BAAD56
> 38543FFE31B5
> 
> Real device MAC address:
> A4C138B3B350
> A4C1383C8895
> A4C138477FFA
> all start with A4C138?

Yes, the first three bytes of a MAC address always indicate the vendor, so it's usually the same for devices by the same manufacturer.

Not quite sure how you found the false positive MAC addresses though, as in the above examples I can only see the issue with
'id': 'A4:C1:38:B3:B3:50', 
vs.
servicedata': 'a4c138b3b3**78**…

with the 78 byte definitely not being part of the recognised device's MAC address, and being the messages which contain the false values.

Since the **servicedata** is only being picked up my OMG as it is being broadcast, and not changed in any way on the OMG receiving end, I wonder how these wrong servicedata broadcasts are being produced on the ATC firmware side, as the ATC protocol states

Byte 5-10 MAC in correct order

so there should never exist such servicedata broadcasts from the ATC firmware. Could it still be an issue with an older firmware? Hopefully we will also see more with messages which include the above stated pubServiceDataUUID.

antibill51, 2022-10-30:
The 3 false addresses have nothing to do with the data sent, they are sensors that do not exist, detected, but a priori no value assigned. I was also having the same issue with fake TPMS sensors, but these haven't been showing up since a few updates.

Debug data from Home Assistant:
```
LYWSD03MMC_ATC-tempc (sensor.lywsd03mmc_atc_tempc)
MQTT discovery data:
Topic: homeassistant/sensor/31683FFE3139-tempc/config
Payload
state_topic: +/+/BTtoMQTT/31683FFE3139
device_class: temperature
unit_of_measurement: °C
name: LYWSD03MMC_ATC-tempc
unique_id: 31683FFE3139-tempc
value_template: '{{ value_json.tempc | is_defined }}'
state_class: measurement
device:
  identifiers:
    - 31683FFE3139
  connections:
    - - mac
      - 31683FFE3139
  manufacturer: Xiaomi
  model: LYWSD03MMC_ATC
  name: LYWSD03MMC-FE3139
  via_device: Salon
platform: mqtt
Subscribed topics:
+/+/BTtoMQTT/31683FFE3139
0 dernier(s) message(s) reçu(s)
Transmitted messages:
LYWSD03MMC_ATC-hum (sensor.lywsd03mmc_atc_hum)
MQTT discovery data:
Topic: homeassistant/sensor/31683FFE3139-hum/config
Payload
state_topic: +/+/BTtoMQTT/31683FFE3139
device_class: humidity
unit_of_measurement: '%'
name: LYWSD03MMC_ATC-hum
unique_id: 31683FFE3139-hum
value_template: '{{ value_json.hum | is_defined }}'
state_class: measurement
device:
  identifiers:
    - 31683FFE3139
  connections:
    - - mac
      - 31683FFE3139
  manufacturer: Xiaomi
  model: LYWSD03MMC_ATC
  name: LYWSD03MMC-FE3139
  via_device: Salon
platform: mqtt
Subscribed topics:
+/+/BTtoMQTT/31683FFE3139
0 dernier(s) message(s) reçu(s)
Transmitted messages:
LYWSD03MMC_ATC-batt (sensor.lywsd03mmc_atc_batt)
MQTT discovery data:
Topic: homeassistant/sensor/31683FFE3139-batt/config
Payload
state_topic: +/+/BTtoMQTT/31683FFE3139
device_class: battery
unit_of_measurement: '%'
name: LYWSD03MMC_ATC-batt
unique_id: 31683FFE3139-batt
value_template: '{{ value_json.batt | is_defined }}'
state_class: measurement
device:
  identifiers:
    - 31683FFE3139
  connections:
    - - mac
      - 31683FFE3139
  manufacturer: Xiaomi
  model: LYWSD03MMC_ATC
  name: LYWSD03MMC-FE3139
  via_device: Salon
platform: mqtt
Subscribed topics:
+/+/BTtoMQTT/31683FFE3139
0 dernier(s) message(s) reçu(s)
Transmitted messages:
LYWSD03MMC_ATC-volt (sensor.lywsd03mmc_atc_volt)
MQTT discovery data:
Topic: homeassistant/sensor/31683FFE3139-volt/config
Payload
state_topic: +/+/BTtoMQTT/31683FFE3139
device_class: voltage
unit_of_measurement: V
name: LYWSD03MMC_ATC-volt
unique_id: 31683FFE3139-volt
value_template: '{{ value_json.volt | is_defined }}'
state_class: measurement
device:
  identifiers:
    - 31683FFE3139
  connections:
    - - mac
      - 31683FFE3139
  manufacturer: Xiaomi
  model: LYWSD03MMC_ATC
  name: LYWSD03MMC-FE3139
  via_device: Salon
platform: mqtt
Subscribed topics:
+/+/BTtoMQTT/31683FFE3139
0 dernier(s) message(s) reçu(s)
Transmitted messages:
```

A new message with sevicedataUUID, but without servicedata (reactivated now).
```
'id': 'A4:C1:38:47:7F:FA',
'mac_type': 0,
'rssi': -82,
'distance': 11.4949,
'servicedatauuid': '0x181a',
'brand': 'Xiaomi',
'model': 'LYWSD03MMC',
'model_id': 'LYWSD03MMC_ATC',
'tempc': 2218.9,
'tempf': 4026.02,
'hum': -70,
'batt': -67,
'volt': 15.358
```

The custom name was added from "Set New Name"  button  on [this page](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

Thanks

antibill51, 2022-10-31:
A new messages with  false values : 
```
'id': 'A4:C1:38:47:7F:FA',
'mac_type': 0,
'rssi': -80,
'distance': 9.521558,
'servicedata': 'a4c138477f7856adbae937fe3f',
'servicedatauuid': '0x181a',
'brand': 'Xiaomi',
'model': 'LYWSD03MMC',
'model_id': 'LYWSD03MMC_ATC',
'tempc': 2218.9,
'tempf': 4026.02,
'hum': -70,
'batt': -23,
'volt': 14.334
```

DigiH, 2022-11-01:
Hi @antibill51 we tried to reproduce your issue here, but with other LYWSD03MMC devices - admittedly with different hardware versions as your B1.9 - we could never get any broadcasts where the servicedata starts with a wrong MAC address in the last byte, also indicating the wrong decoded values.

Would you mind testing with a fresh upload, with the default settings, i.e. also not giving your device a custom name in the firmware?

@pvvx - have you ever seen issues like this when using the **atc1441** format, where the mac address reference in the service data has a wrong last byte, this also indicating wrong values when decoding to the **atc1441** format standard?

> 'id': 'A4:C1:38:3C:88:**95**', 
> 'mac_type': 0, 
> 'rssi': -95, 
> 'distance': 35.511, 
> 'servicedata': 'a4c1383c88**78**56adba553bfe3f', 
> 'servicedatauuid': '0x181a', 
> 'brand': 'Xiaomi', 
> 'model': 'LYWSD03MMC', 
> 'model_id': 'LYWSD03MMC_ATC', 
> 'tempc': 2218.9, 
> 'tempf': 4026.02, 
> 'hum': -70, 
> 'batt': 85, 
> 'volt': 15.358

So far this always seems to be a wrong **78**, from what I have seen.

Thanks

pvvx, 2022-11-01:
Always the same data: '7856adba553bfe3f' 7856adba-553bfe3f
Looks like an ESP8266 stack entry.
553bfe3f -> 0x3ffe3ba1 - ESP8266 Memory Map:  _0x3FFE8000..0x3FFEC000: User data RAM. Available to applications._

h2zero, 2022-11-03:
Yep, looks like memory error to me too.

Address: 0x3ffe3b55 == 0xbaad5678

The question is why.

antibill51, 2022-11-23:
Hi.
It doesn't seem to be happening since I commented :

```
  '-DTimeBtwRead=0'
  '-DScan_duration=5000'
```

I use 5 devices on OpenMQTTGateway, I wonder if the problem would not come from simultaneous connection attempts on the  same LYWSD03MMC ?

DigiH, 2023-01-24:
@animalillo thanks for reporting this. 

Could you turn on showing additional data with
`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"pubKnownServiceData":true}'`
and post some MQTT message here with sub-zero temperature?

1technophile, 2023-02-04:
Closing, feel free if there are other questions

closed 2023-02-04 by 1technophile