1technophile/OpenMQTTGateway #1305: C1011 and Zgateway/RF reception issues

crasu, 2022-10-28:
**Describtion**
I am using an C1101 with a couple of GS-WDS07 433Mhz door sensors. The sensors are about 5m from the receiver. I used to have reception issues on some days (every 5th to 10th message used to get lost). The issues seem to be suboptimal AGC parameters for the C1101 in the RF lib you are using [here](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/b8c6af4c7c2214cd77a4e9b2e2cb37b24b393605/ELECHOUSE_CC1101_SRC_DRV.cpp#L1046).

When changing these values to the recommended values from TI the reception dramtically improves:
https://www.ti.com/lit/an/swra215e/swra215e.pdf?ts=1666996146234&ref_url=https%253A%252F%252Fwww.google.com%252F

This is the patch I am using:
```
--- a/main/ZgatewayRF.ino
+++ b/main/ZgatewayRF.ino
@@ -118,6 +118,9 @@ void setupRF() {
   }
 
   ELECHOUSE_cc1101.Init();
+  ELECHOUSE_cc1101.SpiWriteReg(CC1101_AGCCTRL2, 0x03);
+  ELECHOUSE_cc1101.SpiWriteReg(CC1101_AGCCTRL1, 0x00);
+  ELECHOUSE_cc1101.SpiWriteReg(CC1101_AGCCTRL0, 0x91);
   ELECHOUSE_cc1101.SetRx(receiveMhz);
 #  endif
 #  ifdef RF_DISABLE_TRANSMIT
```

If others are seeing similar issues I could open a pr for this or make this configurable.

Btw:
Tried settings from cul as well. They work better as the original settings as well (guess they are optimized for 868mhz).
https://github.com/heliflieger/a-culfw/blob/f4305ea7ca9aba2ace6978c9c29e2645072e5c66/culfw/clib/mbus/smode_rf_settings.h#L72

**Expected behavior**
No package loss

1technophile, 2022-10-29:
Hello,

It could be interesting to compare with [the version 1.0.0](https://github.com/1technophile/OpenMQTTGateway/releases/tag/v1.0.0), it uses [RadioLib ](https://github.com/jgromes/RadioLib)instead.

crasu, 2022-10-29:
Ah just saw the pr a couple of days ago. Nice that this is merged now.
However this is only used for the rtl433 receiver, which for me gave even worse receive performance.

Just checked - radiolib+rtl433 is still not working very well.

1technophile, 2022-10-29:
> Just checked - radiolib+rtl433 is still not working very well.

Could you detail please?

crasu, 2022-10-29:
IMHO radio lib is not initializing any agc settings on the C1101 as it [seems](https://github.com/jgromes/RadioLib/blob/0b5cd51828d2bf4846bca9ba6a8df5b2e4532459/src/modules/CC1101/CC1101.cpp#L12).

NorthernMan54, 2022-10-29:
@crasu Are you sure - https://github.com/NorthernMan54/rtl_433_ESP/blob/c7158f5bc3e9bd4be80ae6c567063b06e25f2219/src/rtl_433_ESP.cpp#L194

But with that said, I still want to review your recommend settings versus what I have been using in a real world test, give me 24 hours + to run something.

crasu, 2022-10-29:
> @crasu Are you sure - https://github.com/NorthernMan54/rtl_433_ESP/blob/c7158f5bc3e9bd4be80ae6c567063b06e25f2219/src/rtl_433_ESP.cpp#L194
You are right - I only looked into radio lib itself not int the OpenMQTTGateway source. 

> But with that said, I still want to review your recommend settings versus what I have been using in a real world test, give me 24 hours + to run something.
I am unsure if my settings work well for the rtl433 gateway.

crasu, 2022-10-29:
> > Just checked - radiolib+rtl433 is still not working very well.
> 
> Could you detail please?

I cannot receive messages comming from the GS-WDS07 reliably with the rf433 receiver. It only works if I put the device very close to the antenna (under 50cm).

NorthernMan54, 2022-10-29:
First thing

This is 10 minutes with the current release AGC Settings

[image]

This is 10 minutes with the suggested AGC settings

[image]

Reception seems worse with some devices not being received at all

Second with the CC1101 I found that reception is some cases can be problematic.  With different boards having different results ( possibly knock off boards ), and if the antenna is placed in a vertical orientation that reception is better.

crasu, 2022-10-30:
Thank you for looking into this.

I  tried different C1101 but both were cheap modules from the same seller on ebay. Do you have a good source for these modules in europe?

The antenna is in a fixed position and I get good reception values with gnuradio/limeSDR. I do not have any other devices to test with.

NorthernMan54, 2022-11-01:
I'm in North America, and usually just order from Aliexpress for parts etc.

Looking deeper at the results, I did my testing with rtl_433_ESP, which is using a flavour of carrier sense to determine start and end of signal, while RF uses just signal pulses to determine start and end of signal.   Give me a day or so to redo the test case with RF.

NorthernMan54, 2022-11-01:
I just completed an A/B test with my SONOFF iFan03 remote - This is detected as protocol 1 under RF

This test was completed with my CC1101 about 40 to 50 feet away from my test setup ( several rooms away ).  And with the current code that uses the LSATAN CC1101 Library I'm able to receive a signal, and when I add your recommendations for AGC I can no longer receive a signal.

So for me in my environment, these settings reduce the signal reception range.

crasu, 2023-03-21:
I am still using my settings but that seems only to work for me ;-(

closed 2023-03-21 by crasu