1technophile/OpenMQTTGateway #1306: [FR] LYWSD02 NTP clock sync

nonflammable, 2022-10-29:
Feature request:
LYWSD02 internal clock is not precise. After few weeks time can be shitf over minute.
A good solution would be to synchronize the time with a NTP server via OMG.

1technophile, 2022-10-29:
Good catch, for info Theengs Gateway does it automatically thanks to @koenvervloesem.

But agreed, could be considered with OMG

nonflammable, 2022-10-29:
> Good catch, for info Theengs Gateway does it automatically thanks to @koenvervloesem. https://github.com/theengs/gateway
> 
> But agreed, could be considered with OMG

exactly
https://gateway.theengs.io/use/use.html#time-synchronization

nonflammable, 2022-10-29:
Alternatively I can try send NTP time via MQTT to BLE write ... but how? ... what should the message look like?
(https://github.com/1technophile/OpenMQTTGateway/pull/988)

nonflammable, 2023-09-05:
Maybe it's time to refresh request :innocent:

netroy, 2023-10-30:
I'm able to do this using the `commands/MQTTtoBT` topic with a payload like this
```json
{
  "ble_write_address":"[[LYWSD02 MAC ADDRESS]]",
  "ble_write_service":"ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6",
  "ble_write_char":"ebe0ccb7-7a0a-4b0c-8a1a-6ff2997da3a6",
  "ble_write_value":"09c93f6501",
  "value_type":"HEX",
  "immediate":true
}
```

I'm calculating the `ble_write_value` using this JS code in my browser console
```javascript
(() => {
  const timeZone = 1; // CET
  const buffer = new ArrayBuffer(5);
  new DataView(buffer).setUint32(0, Math.floor(Date.now()/1000), true);
  new DataView(buffer).setUint8(4, timeZone);
  return [...new Uint8Array(buffer)]
    .map(x => x.toString(16).padStart(2, '0'))
    .join('')
})();
```

So, theoretically it should be possible to add this 
1. in `main/ZgatewayBT.ino`, as a GATT characteristic write right after reading temperature/humidity (which seems too frequent), or 
2. as a button using `createDiscovery("button"` in `main/ZmqttDiscovery.ino` to let users press a button to trigger the sync via mqtt, or
3. somehow add keep track of all clocks that need syncing, and then sync them in `syncNTP` once every hour or so.

nonflammable, 2023-12-29:
bash version:
`
timestamp=`date +%s|xargs printf %x`
czas=$(printf %s "$timestamp" | dd conv=swab 2> /dev/null | rev)

mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT -m '{
  "ble_write_address":"AA:BB:CC:DD:EE:FF",
  "ble_write_service":"ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6",
  "ble_write_char":"ebe0ccb7-7a0a-4b0c-8a1a-6ff2997da3a6",
  "ble_write_value":"'$czas'01",
  "value_type":"HEX",
  "immediate":true
}'
`
be sure to change:
MQTT TOPIC: `mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT`
MAC:  `"ble_write_address":"AA:BB:CC:DD:EE:FF"`
TIMEZONE: `"ble_write_value":"'$czas'01"` - 01=CET