1technophile/OpenMQTTGateway #1312: Unable to make SRX882S ( v2.0 ) to work with ESP8266 or ESP32

Vendo232, 2022-11-05:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Setting up SRX882S to receive RF with ESP8266 or ESP32, changed pins in Arduino per recommendation and device works I can see it online and connected but no RF reception. 

**To Reproduce**
Steps to reproduce the behavior:
1. Publish to '...'
2. Subscribe on '....'
3. See error

[image]
No RTL_433toMQTT topics when 433MHz device triggered in close proximity. The RF device is functional and verified with orther working instance using CC1101 ( which works OK ) 

I expect to receive topics for door/motion sensors as in this pictures. ( using ESP32 and CC1101 ) but for SRX882S nothing is recieved.
[image]

[image]

[image]

[image]

used 1.0 , 0.9.15 , 0.9.16 no difference

This is second set of SRX882 , after first was not working, ordered second set from different vendor and it is not working.

1technophile, 2022-11-05:
Hello

> No RTL_433toMQTT topics when 433MHz device triggered in close proximity. 

RTL_433 decoding will work only [with CC1101 and SX127X modules](https://docs.openmqttgateway.com/use/rf.html#rtl-433-device-decoders), with SRX882S you can use RF, RF2 or Pilight gateways.

Also, you would need devices that are compatible with those gateways. Your best guess is to try first RF.

Vendo232, 2022-11-05:
> Your best guess is to try first RF

thank you Sir for comming back to me. I forgot to mention above that the the FW I`m using is the RF for ESP8266

[image]

I have tried 3 other ESP boards, all work OK, I can setup WIFI , MQTT etc.. but there is something between the board and SRX882S that does not work correctly, I`m not getting anything. As mentioned above also ordered from different vendor second set and nothing.

I was hoping that one could tell me I`m doing something fundamentally wrong in my setup for RF and SRX882C.

( I have second instance ESP32 + CC1101 + RTL433 and that is working fine )

Vendo232, 2022-11-05:
I guess my question would be how would the MQTT topics show up in MQTT Explorer if I use RF FW and SRX882C?

DigiH, 2022-11-05:
Hi @Vendo232 , is it just me, or is the VCC not also connected to the data pin of the SRX882S on the second picture above?

Vendo232, 2022-11-05:
hello @DigiH I do not have VCC connected to the Data pin I`m using this schematics

[image]

DigiH, 2022-11-05:
Sorry, yeah I meant the CS pin, just looks like it on the picture, that both blue (VCC and D1) wires are both going into the same CS. Could very well just be the angle of the photo.

I'm using the same setup, admittedly though with an SRX882 and not SRX882S.

DigiH, 2022-11-05:
Could you change the power connections from the VIN (which is mainly used for input, also usually but not always also gives a nice output) of the ESP8266 to the actual V3 output?

Vendo232, 2022-11-05:
@DigiH  I have tried that, measured 3.3V on the receiver directly, also tried to supply 4.8V from other source again measured directly on the receiver. 

the SC is connected to the D1 pin and nothing. 

When I trigger 433Mhz door sensor right next to it I get readings on my ( ESP32 + C1101 + RTL433 ) upstairs but no readings in MQTT Explorer under this device.

DigiH, 2022-11-05:
> When I trigger 433Mhz door sensor right next to it I get readings on my ( ESP32 + C1101 + RTL433 ) upstairs but no readings in MQTT Explorer under this device.

If your door sensor is being picked up by RTL433 and therefore recognised and decoded by RTL433 it won't show up on an RF gateway with SRX882S using RF, RF2 or Pilight. 

So the door sensor cannot be used as a test device to see if your RF gateway is working. Do you have any other 433 RF devices, which are not being picked up by RTL433?

Vendo232, 2022-11-05:
@DigiH thank you for confirming that the SRX882 is working for you. Perhaps there is something esle in the "S" version different

[image]

Vendo232, 2022-11-05:
> 
> If your door sensor is being picked up by RTL433 and therefore recognised and decoded by RTL433 it won't show up on an RF gateway with SRX882S using RF, RF2 or Pilight.

Should I disconnect the RTL433/ESP32 device?

DigiH, 2022-11-05:
> Should I disconnect the RTL433/ESP32 device?

That wpn't change anything, what I meant was that if a device is recognised by the RTL433 library it will not be picked up by any of the other RF libraries, i.e. RF, RF2 or Pilight.

So your door sensor will only ever be recognised and picked up by an OMG RTL433 gateway, with such a gateway on ly working with a CC1101 or SX127X module.

So your RF gateway with the SRX882S might be working perfectly. You just don't seem to have any device which it could pick up to report via MQTT.

Vendo232, 2022-11-05:
> So your door sensor will only ever be recognised and picked up by an OMG RTL433 gateway, with such a gateway n ly working with a CC1101 or SX127X module

@DigiH thank you, I did not know it. the RTL433 with ESP32 picks up all my devices just fine except my neighborhood weather stations ( probably due to weak antenna )

I will place the ESP8266 + RF + SRX882S somewhere higher in my house and see if it picks up something else.

With my Linux + Docker + RTL433 + DVB dongle I get some TPMS readings plus few weather stations, was hoping the ESP8266 would give me better readings.

DigiH, 2022-11-05:
With the recent changes to OMG including the new radiolib and also supporting SX127X modules, it might be worth trying such a module, as many tests seem to indicate better RTL433 reception than with a CC1101, so you might be able to pick up your neighbourhood RTL433 weather stations like that.

Testing and trying different external antennas for a CC1101 or SX127X, and also positioning them upright could help with RTL433 ESP32 reception.

Admittedly though, a DVB dongle running RTL433 on linux will probably most of the time have the widest reception. 

All the best

Vendo232, 2022-11-05:
@DigiH  thank you for your time! would you be able to share an Aliexpress linkn for recommended SX127X module?

DigiH, 2022-11-05:
> @DigiH thank you for your time! would you be able to share an Aliexpress linkn for recommended SX127X module?

Most tests have been done by @NorthernMan54 - the maintainer of the rtl_433_ESP library -  and OMG's @1technophile with dedicated **ESP32 HELTEC LORA32 v2** which is an all in one ESP32 with in built SX127X and a screen.

https://de.aliexpress.com/item/1000006969786.html

This might be an option for you as well. 

I opted for a separate SX127X module to hook up to an ESP32

https://de.aliexpress.com/item/1005004532397555.html

but it only arrived yesterday and I haven't hooked it up and tested it yet ;)

DigiH, 2022-11-05:
@Vendo232 , one last question, before you go off an order some of the above SX127X devices.

What are these weather stations recognised as with your DVB dongle running RTL433?

Vendo232, 2022-11-05:
I have not tried Transition via 433 before, it was always just to receive signal in my case. I have tried to send msg via MQTT and I got the "433toMQTT" topic for SRX882S. Not sure though if it is just internal signal or real RF received.

Interestingly the ESP32+RTL433+C1101 picked up the signal and it is very interesting how it is decoding the signal.

[image]

Vendo232, 2022-11-05:
> What are these weather stations recognised as with your DVB dongle running RTL433?

@DigiH I do not have the instance running now, but will try to get readings from one station and get back to you.

Vendo232, 2022-11-05:
if I send via RF RXS882S msg {"value":1315160}

this is how ESP32+RTL433+C1101 receives it
[image]

the ESP8266+RSX882S will just get 
[image]

it is not what I was expecting.

DigiH, 2022-11-07:
> this is how ESP32+RTL433+C1101 receives it [image]
> 
> the ESP8266+RSX882S will just get [image]

Yes, false positives are not that uncommon with rtl_433. They can be safely ignored, unless of course you have some of these devices and rely on only proper readings for them. To completely remove them from being recognised at all **rtl_433_esp** has the option to define and only _allow_ MY_DEVICES.

hallard, 2022-11-28:
Sorry to hijack this thread but I spent time to investigate why my SRX882S was not working, with ESP32 and official wiring such as:

- VCC : 3.3V
- GND : GND
- DATA : GPIO27

When I measured CS it was 0V so my guess was that my SRX882S CS pin has pulldown while SRX882 has pull up

Confirmed by this post (just saw it after my own investigation)
[image]

So to get to working I connected CS to 3.3V and now it's working fine

- VCC : 3.3V
- CS : 3.3V
- GND : GND
- DATA : GPIO27

Worth having a note on documentation :-)

Vendo232, 2022-11-28:
> Sorry to hijack this thread but I spent time to investigate why my SRX882S was not working, with ESP32 and official wiring such as:
> 
> * VCC : 3.3V
> * GND : GND
> * DATA : GPIO27
> 
> When I measured CS it was 0V so my guess was that my SRX882S CS pin has pulldown while SRX882 has pull up
> 
> Confirmed by this post (just saw it after my own investigation) [image]
> 
> So to get to working I connected CS to 3.3V and now it's working fine and you can apply this also to SRX882 V1 it does not hurts and make documentation consistent for all SRX882 version.
> 
> * VCC : 3.3V
> * CS : 3.3V
> * GND : GND
> * DATA : GPIO27
> 
> Worth having a note on documentation :-)

It worked sir! Should be noted in wiki this detail!

1technophile, 2023-04-09:
Docs updated

closed 2023-04-09 by 1technophile