1technophile/OpenMQTTGateway #1326: TFA Drop Rain Gauge 30.3233.01

kimi1983, 2022-11-20:
Hello,

Hardware used : Heltec WiFi LoRa 32

Problem : 

I am able to receive the signal from my TFA Drop but the id of the device is changing very often.
When you change battery of the device I can live with this behavior.
But today the id of the device changed on its own during the day.

example : 
{"model":"TFA-Drop","id":721248,"battery_ok":1,"rain_mm":2.286,"mic":"CHECKSUM","protocol":"TFA Drop Rain Gauge 30.3233.01","rssi":-59,"duration":301996}

in my case the id is also used as topic name which is not good when it changes very often because I need to adapt also the topic id in my homeautomation software.

do you have any tip for me ?

best regards,
Kim

kimi1983, 2022-11-20:
The TFA Drop is showing up like this : 

home/OpenMQTTGateway433/RTL_433toMQTT/TFA-Drop/721248

it would be fine with only TFA-Drop and if the ID would remaine the same. In this case 721248

NorthernMan54, 2022-11-20:
The ID number is being sent as part of the signal by the device and decoded by the rtl_433_ESP library, so no options with the software.

Have you tried using a wildcard on the topic ?

home/OpenMQTTGateway433/RTL_433toMQTT/TFA-Drop/+ or home/OpenMQTTGateway433/RTL_433toMQTT/TFA-Drop/#

DigiH, 2022-11-20:
Hi @kimi1983 while I don't know why the ID of your TFA Drop keeps changing - does it maybe reset itself frequently for some reason ?? - could you not subscribe to its messages with a wildcard, as in

`home/OpenMQTTGateway433/RTL_433toMQTT/TFA-Drop/#`

in your home automation software to still get all its messages even with changing IDs?

Unless there is another TFA Drop broadcasting in your area this should remove the need to take the ID into account.

NorthernMan54, 2022-11-20:
Also is your device / batteries having an issue, and triggering the ID changes.

kimi1983, 2022-11-20:
thank you for your quick replies.

in my case I am using Jeedom ass Backend Software.
I am already subscribing to this topic : home/OpenMQTTGateway433/RTL_433toMQTT/TFA-Drop/#

In this case I am sure to capture every signal coming from my TFA Drop even when the ID changes.
But I am unable to react to an unknown ID with jeedom. Every ID Change is ading a new command to me jeedom config.

kimi1983, 2022-11-20:
> Also is your device / batteries having an issue, and triggering the ID changes.

Would the normal behavior of the TFA Drop be to hold the same ID until the batteries are empty ?

NorthernMan54, 2022-11-20:
I don't have a TFA Drop in my collection, but all my other devices ( LaCrosse and Acurite ) do maintain the same ID until the batteries are changed. It's how their display's know what sensor to display.

kimi1983, 2022-11-20:
> I don't have a TFA Drop in my collection, but all my other devices ( LaCrosse and Acurite ) do maintain the same ID until the batteries are changed. It's how their display's know what sensor to display.

OK, this behavior would be fine for me.
So maybe my device is having an hardware error.

Will monitor the behavior and otherwise buy another one.

Thank you.

DigiH, 2022-11-20:
Could it be that some rain water got into the electronics/battery parts to cause resets/restarts?

I would keep it indoors to dry out for a while to see if the ID then stays the same. If that is the case possibly open it to see if any sealing got compromised.

kimi1983, 2022-11-20:
actually the device is new..
I checked once the battery compartment which was dry.
maybe there is a hardware failure on my device.

will monitor the next days and see if the id is changing again....

NorthernMan54, 2022-11-20:
For reference, this is the issue from rtl_433 where they did the work on that device, and no mention of the ID numbers changing

https://github.com/merbanan/rtl_433/issues/1240

Maybe low power from the batteries? triggering the reset when power drops below a particular level

HanSyt, 2023-04-12:
Same problem here, TFA drop changes its ID and I have a new device in HA. Quite annoying behaviour. I know it is changing with batteries replaced and the original receiver has no problems with this. Issue is with both RTL dongle en RFXCom.

I think I have to replace this device (or modify it with a Zigbee version like this: https://community.home-assistant.io/t/diy-zigbee-rain-gauge/255379
Or maybe convert the TFA to it.

1technophile, 2023-04-12:
oh, that's not OpenMQTTGateway integration, can't help with RFXCom unfortunately

HanSyt, 2023-04-14:
I did choose for the easiest way, replace the TFA Drop for a netatmo system, this also replace my noisy home made CO2 meter (fan).
Maybe I convert it tot the DIY Zigbee solution

closed 2023-10-01 by github-actions[bot]

FlyingMG, 2024-01-08:
Hi, I recently added the TFA Drop 30.3233.01 to my weather station. I am acquiring the data using rtl_433 and plotting it in grafana from an influxdb. But the value for the rain in mm (rain_mm) is always incremented and not set to zero again. Is there a problem with my sensor or do I misunderstand the way I should interpret the data I am receiving ?