1technophile/OpenMQTTGateway #1350: prebuild images with DS18b20 and DHT22 support?

ozett, 2022-12-07:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
i flashed a lolin-lite board with your prebuild binary and the webinstaller 
i jumperd a dht22 and a ds18b20 sensonr to the default pins
dht22 (GPIO16) (ds18b20 (gpio2) (also VCC and GND for both
an can not find an sign of life from thi sensor.
no mqtt topic

**To Reproduce**
Steps to reproduce the behavior:
[image]

**Expected behavior**
i exptected to find some information about htis sensors on the serial log
i expected to find new mqtt-topics added ../DHTtoMQTT/#
i expected to find new mqtt-topics added ../CLIMAtoMQTT/#
found nothing.

**Screenshots**
used this pin-numbering
[image]

**Additional context**
hard to debug, the log is somehow short.
how can i debug pin errors?
how to debug the discovery of the sensos

maybe the prebuild images does not supoort this sensors? 
i cannot find some informatione.
at the moment i assume, that they are somehow automatically discovered and activated by the firmware...
or not?

1technophile, 2022-12-07:
If I understand correctly, you are looking to have BLE gateway + DHT22 + ds18b20  on ESP32 ?

1technophile, 2022-12-07:
Do you think you could install platformio and upload a build ?
https://docs.openmqttgateway.com/upload/builds.html

ozett, 2022-12-08:
as a last resort i prepared myself to do my own buld.
but i wanted not to install all the stuff.

are your prebuild binaries without sensors?

tasmata has a binary with all sensor for the webinstaller.
https://tasmota.github.io/docs/Firmware-Builds/#firmware-variants
hope to find the same here.

i can build a binary with arduino, or visual studio/plattform-io. 
everything is running great, hoped only to flash a sensor-binary. 
do you want me to do all the build myself (on weekend) ?
and upload afterwards?

1technophile, 2022-12-08:
> are your prebuild binaries without sensors?

If you need a combination of sensors/gateway you need indeed to build yourself, only the most common builds are proposed out of the box

> tasmata has a binary with all sensor for the webinstaller.

We are taking a different approach with OMG, make available only what is necessary in the code, this is why there is binaries dedicated to one board or another. The goal is to leverage the full capabilities of the board without extra code that the board will never use and that could cause instability.
The downside of this is that for particular use case you would need to build yourself the board, but if you put your hands into PlatformIO you will see that this is not so difficult:

```ini
[env:esp32dev-ble-dht-ds]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
  ${libraries.wifimanager32}
  ${libraries.dht}
  ${libraries.unifiedsensor}
  ${libraries.decoder}
build_flags = 
  ${com-esp.build_flags}
  '-DOMG_VERSION="v1.1.1"'
  '-DZgatewayBT="BT"'
  '-DZsensorDHT="DHT"'
  '-DZsensorDS1820="DS1820"'
  '-DTimeBtwRead=100'
  '-DMQTT_USER="username"'
  '-DMQTT_PASS="password"'
  '-DMQTT_SERVER="192.168.1.55"'
```

ozett, 2022-12-12:
i found some time to install platformio.

DHT and 1820 sensors are now enabled ,
in platformio.ini the lolin-lite board is actived,

but how do i know what was also in your default binary?
I need the switchbot discovery and mqtt. 

is this default enabled? or what to enable - and where?

[image]

DigiH, 2022-12-12:
Hi @ozett,

Have a look at the default Lolin Lite environment in platformio.ini - `[env:esp32-lolin32lite-ble]`

There you see the default definition which is also being used for the pre-built binaries.

You can duplicate this environment definition for further customisation, or add the additional required libraries and build-flags definitions to it directly, as shown by @1technophile's example above, i.e. the 

```
${libraries.dht}
${libraries.unifiedsensor}
```

libraries, and the 

```
'-DZsensorDHT="DHT"'
'-DZsensorDS1820="DS1820"'
```

build-flags.

Discovery and MQTT will be activated for all the environments, unless specifically undefined for individual environments or globally. 

This way you can customise any build environments to suit individual needs, then build and upload them to your board, without having to uncomment/set the definitions in the individual config files, then having to comment/unset them again for other board builds.

Maybe you can show us your final environment definition.

ozett, 2022-12-12:
thanks for quick reply.

my direct changes seems enough (as a dirty hack ) to get this two sensors into the build? 
i will try to build this firmware and check if switchbot and sensors are discovered.

i have still hopes that someone provide also "all-sensor"-binaries , like tasmota, 
with all the risks for lazy people.  if something goes wrong with them, one could build custom binaries at the end.
i am only a normal user...

thanks for quick help, i will try now..

DigiH, 2022-12-12:
> my direct changes seems enough (as a dirty hack ) to get this two sensors into the build?

I'm not sure, as you only seem to have the `build_flags` defined in `User_config.h`, but the relevant libraries are not added to the  `lolin32lite-ble` environment.

Let us know how you get on.

ozett, 2022-12-12:
first try,
more libs and embedded packages coming to my workstation. i am not happy ... but going further
[image]

ozett, 2022-12-12:
did not help, googling throws a lot of this error.
maybe this helps? https://community.platformio.org/t/git-library-download/17000/12
(all i wanted at start was 2 little sensors...)

ozett, 2022-12-12:
now a lib is missing 🧑‍🔬 . what to do ?

[image]

ozett, 2022-12-12:
reinstall of platformio did not help with github,
restart of visualstudiocode help with github-client.

no i followed the docs, and will test

[image]

ozett, 2022-12-12:
lib missing. what now?

[image]

DigiH, 2022-12-12:
Yes, creating your own `*_env.ini `file with your custom environment definitions is the best way to go.

Also have a look at the left side of VSC and select the Platformio icon. There, under **Project Tasks** you can see all the environments, including your custom environment. Expanding your environment there you will the the BUILD and UPLOAD options.

Just two pinters:
• You seem to have taken the esp32dev environment as a template for your board. Since you have a :olil you should take the previously mentioned `lolin32lite-ble` as a template, which has the different board definition.
• Also customise the name of your environment slightly. This will make it easier to pick out in the above mentioned Project Tasks list.

ozett, 2022-12-12:
i resetted user_config.h to defaults, that helped with the lib (screenshot below)

>• You seem to have taken the esp32dev environment as a template for your board. Since you have a :olil you should take the previously mentioned lolin32lite-ble as a template, which has the different board definition.
• Also customise the name of your environment slightly. This will make it easier to pick out in the above mentioned Project Tasks list.

i took all you pointed out straight [from the developers first aid](https://github.com/1technophile/OpenMQTTGateway/issues/1350#issuecomment-1342721422). i am not that expert to see this little differences at first glance.

thanks for all the hints. (will try another one)

( lib-reset like this:)
[image]

DigiH, 2022-12-12:
> lib missing. what now?

I just created the same environemnt here, and yes, the 

`${libraries.dallastemperature}`

also needs to be included for the DS18b20 for a correct build.

ozett, 2022-12-12:
thanks, willl add this.
but as i edited for lolin-lite, the pubsub client is gone. why this?

[image]

DigiH, 2022-12-12:
So in your own custom *_env.ini file you should have your custom environment like this
```

[env:esp32-lolin32lite-ble-CUSTOM]
platform = ${com.esp32_platform}
board = lolin32
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager32}
  ${libraries.ble}
  ${libraries.decoder}
  ${libraries.unifiedsensor}
  ${libraries.dallastemperature}
  ${libraries.dht}
build_flags =
  ${com-esp.build_flags}
  '-DZsensorDHT="DHT"'
  '-DZsensorDS1820="DS1820"'
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=22'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DGateway_Name="OMG_LOLIN32LITE_BLE_DHT22_DS18b20"'
```

You can also add the relevant WiFi and MQTT credentials there if you don't want to go through the AP gateway set up, but need to fully define all the relevant required build_flags, for them to fully work and to avoid errors.

```
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="YOURNETWORK
  '-Dwifi_password="PASSWORD"'
  '-DMQTT_SERVER="192.168.14.204"'
  '-DMQTT_USER=""'
  '-DMQTT_PASS=""'
  '-DMQTT_PORT="1883"'
```

Then selecting this environment in the Platformio Project Tasks, you will see

[image]

where you can select to Build and Upload.

ozett, 2022-12-12:
looks good.  🥇 
will try to flash...

thanks for all the hints and help! 🏅 

will report later if flashing was success and if its running..

[image]

ozett, 2022-12-12:
upload with visual-studio-code had errors,
using the docs found a broken link to the bootloader...

what a hack... 

[image]

ozett, 2022-12-12:
having the ds1820 sensor plugged into PIN2 wont let me flash.
i unplugged only this cable from pin2 and it worked flashing from vscode.
but everytime i say "upload" it compiles again... annoying...
(and maybe this pin-default "2" should change for noobs?)

[image]

ozett, 2022-12-12:
i dont get sensor-topics on mqtt.
how do i check, if the cabling is wrong?
or if the sensors are not configured for sending over mqtt?
how to debug some OMG configuration issues?
any method to check this things?

[image]

ozett, 2022-12-12:
to check the pinout and cabling, i flashed tasmota.
everything works on this pins without any change.

how do i check OMG for reading the same sensors on the same pins?

[image]

DigiH, 2022-12-12:
The config/pinouts options for the DHT22 are in `config_DHT.h`, and for the DS1820 in `config_DS1820.h`.

Since I've never connected any sensors to my gateways, let alone a DS1820, I'm not sure if it might not also need the onewire library, but assuming it does.

Could you add
`
${libraries.onewire}`

to your environment?.

Are you getting the DHT22 values correctly in the MQTT messages?

ozett, 2022-12-13:
> it might not also need the onewire library
think it was installed as dependency, but can configure explicity.

🏃  on the same pins the sensors work fine with tasmota
💀 on the same pins i dont get any sign of life on OMG from the same sensors/pins💀

is there a way to debug/log anything about the DHT or the DS1820 sensor?

ozett, 2022-12-13:
uncomment of the define of the sensors in `User_config.h `
is necesseray to make them work? or an alternative statement like that in my custom.ini?

i am not a developer, but try my best in fiddling aroound..

[image]

ozett, 2022-12-13:
error with userconfig.h, 
but maybe thats blocks activating the sensors?
[image]

ozett, 2022-12-13:
cannot find DHT lib in the platform.io registry.
What to do?

[image]

DigiH, 2022-12-13:
The library which is being used for the DHT family is

https://github.com/adafruit/DHT-sensor-library

with version 1.3.2, as the newer version actually caused issues with the DHT22 sensor not being recognised.

https://community.openmqttgateway.com/t/dht22-sensor-not-showing/2045/21

Let's concentrate on the DHT22 sensor first, as I also have this one and it was working fine for me during the above issue testing with the reverted library.

The only necessary steps to include it should be the added libraries and build_flags in your environment

${libraries.unifiedsensor}
${libraries.dht}
…
'-DZsensorDHT="DHT"'

no other changes in any other files. Just to make sure, you are using the latest version of OpenMQTTGateway 1.1.1?

Could you copy and past your full environment definition again, commenting out your WiFi and MQTT credetials, to make sure the environment set up is fine?

You can also use the Project Task **Upload and Monitor** to see the serial monitoring in the Platformio terminal to see what is going on .

ozett, 2022-12-13:
if it will ever work, it will be the same difficulty to keep up with development,
i am afraid... 

the part from my custom platform.io.ini

```
[env:esp32-lolin32lite-ble-dht-ds1820]
platform = ${com.esp32_platform}
board = lolin32
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ;${env.lib_deps}           ; Inherit all the library dependencies from [env]
  ${libraries.wifimanager32}
  ${libraries.ble}
  ${libraries.decoder}
  ${libraries.dht}
  ${libraries.unifiedsensor}
  ${libraries.dallastemperature}
  ${libraries.onewire}
  
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=22'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DGateway_Name="OMG_LOLIN32LITE_BLE"'
  '-DOMG_VERSION="v1.1.1"'
  '-DZsensorDHT="DHT"'
  '-DZsensorDS1820="DS1820"'
  '-DTimeBtwRead=100'
  '-DMQTT_USER=""'
  '-DMQTT_PASS=""'
  '-DMQTT_SERVER="192.168.14.204"'
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="wifi"'
  '-Dwifi_password="passkey"'

``` 

looks by names, that the adafruit lib is likely not build, but the one-wire-lib is. 
but maybe the dht-lib has another name 

[image]

DigiH, 2022-12-13:
> '-DTimeBtwRead=100'

Could you comment out this line or delete it and rebuild. As this is only for the BLE connection to define the time between the BLE scanning, I'm wondering if 100 ms are enough to get proper sensor readings.

ozett, 2023-01-01:
i put integrating sensors on hold.
i use another cheap board with tasmota for sensors. very comfortable.
i try to make use only with OMG mainly for the BT data @ the moment
https://github.com/1technophile/OpenMQTTGateway/issues/1372

closed 2023-09-30 by github-actions[bot]