1technophile/OpenMQTTGateway #1368: Missing esp32C3 programming documentation

erwann2000, 2022-12-16:
Hello,
I take few time to boot success with ESP32C3.
I found solution with arduino-ide log during programming time.

In log, the difference between ESP32 and ESP32C3 is :
The bootloader binary need to be write at 0x0000 instead of 0x1000

1technophile, 2022-12-16:
Hello,

Did you try the web uploader with this board ?
https://github.com/1technophile/OpenMQTTGateway/blob/827724f079b9cd6cee6ce69735aee5bb011dced5/platformio.ini#L1436

erwann2000, 2022-12-16:
I prefer use esptool. I test with online tool but I had the following message: "Your Unknown Chip board is not supported.
OpenMQTTGateway: Unknown Chip"

[image]

erwann2000, 2022-12-16:
In https://docs.openmqttgateway.com//firmware_build/esp32c3-dev-m1-ble.manifest.json
I can see :
        { "path": "/firmware_build/bootloader_dio_80m.bin", "offset": 4096 }
This offset 4096 = 0x1000 

Do all esp32c3 start at 0x0000 or just mine ?

paccerdk, 2022-12-22:
I also couldn't get the uploader to work, got same error (unknown chip)
I managed to flash the **esp32c3-dev-m1-ble-*** firmware with the following command:
`python esptool.py --chip esp32c3 --port "COM7" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0000 esp32c3-dev-m1-ble-bootloader.bin 0x8000 esp32c3-dev-m1-ble-partitions.bin 0xe000 boot_app0.bin 0x10000 esp32c3-dev-m1-ble-firmware.bin`

Flashing went well, and initially, it did run.
However, after configuring the WiFi/Mqtt, it boot loops and i get the following serial output:

```
[   254][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selecESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x42088c56
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
[   248][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (256) gpio: gpio_set_level(226): GPIO output gpio_num error
[  2378][E][Preferences.cpp:483] getString(): nvs_get_str len fail: BTConfig NOT_FOUND
[  3454][E][esp32-hal-misc.c:128] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403824d4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
[   248][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (256) gpio: gpio_set_level(226): GPIO output gpio_num error
[  2377][E][Preferences.cpp:483] getString(): nvs_get_str len fail: BTConfig NOT_FOUND
[  3452][E][esp32-hal-misc.c:128] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT
```
etc.

1technophile, 2022-12-23:
Give me some days to dig this.

h2zero, 2022-12-30:
I dug into this a bit, something in the arduino core/idf version is bugged with the watchdog timer causing frequent resets. Also a random assert in the BLE controller is triggered occasionally, which is more catastrophic because it doesn't trigger a reset at all/

xe5700, 2023-02-13:
I use PlatformIO to flash.

closed 2023-09-30 by github-actions[bot]