1technophile/OpenMQTTGateway #1374: Every random MAC generates a new Topic. When does this stop?

ozett, 2022-12-27:
**Is your feature request related to a problem? Please describe.**
i have around 20 BT devices sending data, OMG makes for every random MAC (change every 15 minutes?) a new topic on MQTT.
after some hours i now have 465 topics from this expired MAC on mqtt, but they wont get deleted.
when does the creation of this temporary random mac-topics stop? never?

**Describe the solution you'd like**
delete old MAC-Topics after some time?

**Describe alternatives you've considered**
use espresence fingerprinting for topic generation, not the MAC that expires.
maybe some other configuration to keep real MACs and delete random MACs?

**Additional context**
lots of random topics from random macs. too much for 20 devices...

[image]

DigiH, 2022-12-27:
> after some hours i now have 465 topics from this expired MAC on mqtt, but they wont get deleted.
> when does the creation of this temporary random mac-topics stop? never?

@ozett I am really not sure what your issue is here. You have to differentiate between two observed things here

1. OMG correctly reports any devices which broadcast BLE data, with its manufacturerdata or servicedata only if they are unknown devices and not being decoded.
2. MQTT Explorer records and lists all such MQTT messages it receives from OMG or any other MQTT publishing client

The easiest way to clear your long list would be to disconnect and reconnect MQTT Explorer to your MQTT broker, or select the existing BTtoMQTT topic and hit the CLEAR button. Then you will see that this **collection** of temporary MAC addresses will vanish, and the ever longer list is just an addition to all the changing MACs which OMG reports. OMG itself does not store any of the previously recognised MAC addresses, it is MQTT Explorer which adds them all up during one connected session.

> use espresence fingerprinting for topic generation, not the MAC that expires.

I would very much vote against that, as the espresence fingerprinting is very unreliable when there is more than one iPhone around for example - my neighbours' iPhones are not in my control, but they happily mess with any fingerprinting here. Also manual adding of identity resolving keys is not really an option for OMG currently.

So the list getting longer and longer in MQTT Explorer is nothing OMG can really affect. It reports the chaning MAC addresses correctly, the listing of all previously received MQTT messages with their MACs is done there.

If you are not interested in devices which constantly change their MAC addresses I would suggest you apply a white-list to your OpenMQTTGateway with only the devices which have constant MAC addresses and which you are interested in.

ozett, 2022-12-27:
thanks for all explanations.
maybe something with the mqtt-explorer.
its my first try with that explorer, maybe somekind of archiv function.
will explore this and report back,
thanks again for background informations from your side.
👍

DigiH, 2022-12-29:
I think this can be closed. Right, @ozett ?

closed 2022-12-30 by ozett