1technophile/OpenMQTTGateway #1376: [NEW DEVICE] Xiaomi Soocas X3U | Sonic Electric Toothbrush, => no MQTT. why?

ozett, 2023-01-01:
**Is your feature request related to a problem? Please describe.**
one device does not show up in OMG.
must be a toothbrush 😄 

on espresence one of my devices show up.
cannot figure out a reason for that.
any hints to debug? or simply missing in a device list?

**Describe the solution you'd like**
would be pleased, if this device would come up on OMG.

**Describe alternatives you've considered**
enable raw data sending and finding a hidden filter on OMG

**Additional context**
[image]
[image]

https://files.xiaomi-mi.com/files/instructions/soocas-tooth-brush-en/SOOCAS-wm.PDF

DigiH, 2023-01-02:
Could you have a look at this device with the nRFconnect app and post a screenshot of its advertsiment data from there?

ozett, 2023-01-02:
The toothbrush is not paired, to my knowledge
its present in ESPRSENCE, but does not show up on OMG.
is there a filter for devices with no useful data? maybe hte `ID`in espresence is a filter?

i have `nrf-connect for mobile` on my android11 but can only see 4 devices at a time.
is [that app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=de&gl=US) limited?

[image]

___________

[image]

DigiH, 2023-01-02:
>> The toothbrush is not paired, to my knowledge

> Also is your toothbrush paired with the Mi Home app? It might need to be paired there first to start broadcasting its relevant data.

Other than that, and it being close to your Android phone I would expect it to show up in nRFConnect.

ozett, 2023-01-02:
> Other than that, and it being close to your Android phone I would expect it to show up in nRFConnect.
will try this...

koenvervloesem, 2023-01-02:
You probably need to "turn on" your toothbrush before it starts advertising. At least for my Philips Sonicare it's this way: I can only see its BLE advertisements when I pick up the toothbrush and the LEDs turn on.

ozett, 2023-01-02:
> You probably need to "turn on" your toothbrush before it starts advertising. At least for my Philips Sonicare it's this way: I can only see its BLE advertisements when I pick up the toothbrush and the LEDs turn on.

will try this, too.

@DigiH 
nearer to the brush, thats a hit!

| Scan| Raw| 
| :---:              |     :---:      |
| [image]| [image]    |

DigiH, 2023-01-02:
> nearer to the brush, thats a hit!

That is what I was thinking all along, and likely much the same when it is closer to the OMG gateway as well, then showing up there, tool!?!

ozett, 2023-01-02:
@DigiH 
you nailed it! (had put the brush next to my lolin-board.)

(now i need to mesh more of OMGs. and triangulate the rssi. 😄
 [like others seems to do ](https://www.u-blox.com/en/technologies/bluetooth-indoor-positioning)
but 1st, will build the 2nd one with my wemos-board 😄 )

| Screenshot | 
| :---:  |  
| [image] |

ozett, 2023-01-02:
discoverd just this cool archive-thing in mqtt explorer and found some manu-data.. 
maybe it helps anybody out there to decipher?
```
{
  "id": "F7:F1:6D:F4:2E:33",
  "mac_type": 1,
  "name": "SMI-X3",
  "manufacturerdata": "5555736f6f636172657833f7f16df42e33",
  "rssi": -56,
  "servicedata": "3120ce0000332ef46df1f70d",
  "servicedatauuid": "0xfe95"
}

```
| Screenshot | 
| :---:  |  
| [image] |

DigiH, 2023-01-03:
Yes, this information is needed, copied and pasted like the first json, but then also the corresponding properties from the toothbrush app at the same time, to be able to see if there is any correlation between them.

So far all that is apparent is that the MAC address of the toothbrush is also included in the manufacturerdata and in reverse in the servicedata.

closed 2023-09-29 by github-actions[bot]