1technophile/OpenMQTTGateway #1378: Platform.io settings for  wemos d1 mini esp32 ? Build-flags were ignored.

ozett, 2023-01-02:
**Is your feature request related to a problem? Please describe.**
had a wemos lying around an want to make a build. i am a noob.
used an template for doing this, but the build-setting were not considered, i guess.
is there some mistake in my config?
https://github.com/1technophile/OpenMQTTGateway/issues/1372#issuecomment-1368411326

**Describe the solution you'd like**
hint what board to chose in platform.io for the wemos

**Describe alternatives you've considered**
stay with boards that are know wo work

**Additional context**
whats wrong with my production file?

[image]

DigiH, 2023-01-02:
Not having any myself, but isn't the wemos d1 mini line all non-ESP32s, i.e. without Bluetooth? Solely based on ESP8266/ESP8285.

So adding and mixing and matching ESP32 libraries still won't give it any Blietooth ability and will generally nor work with ESP8266/ESP8285 based devices for any kind of build.

ozett, 2023-01-02:
>  wemos d1 mini line all non-ESP32s,

as [shown in this thread ](https://github.com/1technophile/OpenMQTTGateway/issues/1372#issuecomment-1368411326)there exist now a esp32 variant

for only bluetooth function it may not matter what exact board-template,
but after changing my config-template from lolin to this wemos another LED lit up.
maybe also important to choose the right board-template or `default_envs` for the pin-numbering?
or continous scanning with esp32dev-cont ?

any suggestions? try and error?

[image]

DigiH, 2023-01-02:
The `default_env` at the top just seems to add to the confusion, as it'll only really be used when doing task under the default section. 
To use your custom environment it should be your environment name, which is `wemos_d1_mini32`, whereas currently you have the default (mind you, different use of default here ;) `esp32dev-ble` defined.

To be honest I would just completely ignore and comment out/delete all default_envs under the platformio section, much the same as in platformio.ini.

**How do you then build and upload in general?** I think if we establish the functionality and strategy of environments and where and how to Clean, Build and Upload each one individually, a lot of your confusion about platformio building will become a lot clearer and easier.

> continous scanning with esp32dev-cont ?

Have a look in platformio.ini how the two different environments `esp32dev-ble` and `esp32dev-ble-cont` are actually defined further down. Then you will see that the main difference is the **TimeBetweenRead** and **Scan_Duration** for these preset environments, something you have manually added to your own **wemos_d1_mini32** environment already.

ozett, 2023-01-02:
have it now like this. looks good to me (noob, even with comments) and compiled 😄 
now testing...

``` 
[platformio]
;default_envs = nodemcuv2-pilight
;default_envs = esp32-lolin32lite-ble
;default_envs = esp32dev-ble
default_envs = wemos_d1_mini32
; this is the fall-back, it build cmd does not specify the environment

; run build with this command an cmd-line . only on 1st-run
; pio run --target upload --environment wemos_d1_mini32

[env:wemos_d1_mini32]
;extends = env:esp32dev-ble
;platform = espressif32
platform = ${com.esp32_platform}
board = wemos_d1_mini32
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ;${env.lib_deps}           ; Inherit all the library dependencies from [env]
  ${libraries.wifimanager32}
  ${libraries.ble}
  ${libraries.decoder}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=22'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DGateway_Name="omg_wemos_d1_mini32"'
  '-DOMG_VERSION="v1.1.1"'
  '-DBase_Topic="omg2/"'
  ;'-DZsensorDHT="DHT"'
  ;'-DZsensorDS1820="DS1820"'
  ;'-DTimeBtwRead=100'
  '-DTimeBtwRead=0'
  '-DScan_duration=1000'
  '-DMQTT_USER=""'
  '-DMQTT_PASS=""'
  '-DMQTT_SERVER="192.168.14.204"'
  '-DESPWifiManualSetup=true'
```

DigiH, 2023-01-02:
Since you already have created you own portable production_env,ini, now with `default_envs = wemos_d1_mini32` could you still tell me how you then build and upload your custom environment?

Once you create more custom environments in the future it will quickly get cumbersome to comment/uncomment all the different default_envs at the top.

Have a look at the platformio icon on the left side of Visual Studio Code, click on it and then expand the PROJECT TASKS. There you will see all defined environments, from platformio.ini and your custom ones towards the end. There you can then easily Clean, Build and Upload any environment, without having to keep defining and changing any default_envs.

ozett, 2023-01-02:
> There you can then easily Clean, Build and Upload any environment, without having to keep defining and changing any default_envs.

This is all great help, thank a lot.
(hope that i dont have to play with that coding-environment too much.) 🍀

1technophile, 2023-02-04:
Closing, feel free if there are other questions

closed 2023-02-04 by 1technophile