1technophile/OpenMQTTGateway #1380: CC1101 -> Support for ELECHOUSE_cc1101.setPA();

Dattel, 2023-01-07:
**Is your feature request related to a problem? Please describe.**
I'm wondering, if you can implement the ability to define the TX-Power for sending RF433 with the ELECHOUSE and the CC1101 with OpenMQTTGateway. 

Currently i have the problem, that some of my RF-Signals are too powerful and disturbing my neighbours power-sockets.. I'm switching them too with my commands :-D

The ELECHouse-Gateway already seems to support changing the TX-Power using the setPA(); function

`ELECHOUSE_cc1101.setPA(10);  // Set TxPower. The following settings are possible depending on the frequency band.  (-30  -20  -15  -10  -6    0    5    7    10   11   12) Default is max!`

**Describe the solution you'd like**
I think of a compiler switch to define the default TX power (if not set, always use the maximum)
'-DZradioCC1101Power=12'

The second change would be a new optional parameter in the payload sended to the MQTTTo433 topic to change the power on a single 433 transmission

`{"value": 9084946,	  "protocol": 4,	  "length": 24,	  "delay": 350,	  "repeat":5, "CC1101_PA":10 }`

That would be the maximum flexible solution.
I'm interessted in your thoughts..
Thanks

DigiH, 2023-01-07:
Hi @Dattel, while I understand your request, and someone else will answer about this possibility, I had exactly the same situation here with my neighbour. Depending on which power-sockets the neighbour and you are using it might also be possible to change their receiving values - in our case with dip switches - to avoid having the conflicting commands. If they are however self-learning and/or multi-receiving value versions this will not work.

Dattel, 2023-01-07:
Hi @DigiH
Sadly not.... 
I'm using sockets from Lidl, and he is using intertec sockets.
Both remotes are looking the same. Buttons for A to D and 2 global Power on/off. There is no way to change frequency.

My remote sends rolling codes -> each key sends 4 different codes alternately. 
For openmqtt i just picked ONE of these Code to send with and it works for me.

But the strange thing is that my D-Off-Key switches all of my neighbours sockets to off... regardless, if they are programmed to channel A,B,C or D... I already tried another D-Off code (from my list of four), but it doesn't help.

So i already implemented my idea from above in an quick and dirty way. 
Will test the threshold for me and hopefully can give a feedback on that...

Dattel, 2023-01-07:
publishing 
`{"value": 8602176,	  "protocol": 4,	  "length": 24,	  "delay": 350,	  "repeat":5, "cc1101_pa":5}`
on MQTT currently send with a power of 5..

`{"value": 8602176,	  "protocol": 4,	  "length": 24,	  "delay": 350,	  "repeat":5}`
on MQTT sends with the maximum of 12 (default) from my global compiler switch
`'-DRF_CC1101_POWER=12'  ;`

Dattel, 2023-01-07:
Update: seems to work... 
if i throttle down my CC1101 with cc1101_pa:0 so only my socket switches on/off...
But i'm not sure, if i'm happy with that situation overall. Turns out, that my neighbors device also switches, when i use my regular remote instead of the cc1101 :-( there are ~50 meters between his house and mine.. I'm not really willing to only switch these damn sockets with the CC1101 and banish the remote-control
Astonishing that we didn't realized that during the past 5 years.

But nevertheless, maybe it's a good idea to support the Power-Adjustment through OpenMqttGateway.

DigiH, 2023-01-07:
Yes, the remote will have the same issue :( for that very reason I only ever got RF sockets without self learning from the beginning, but with manual dip-switch settings on the sockets as well as the remote. Pollin used to have a great 3 socket/1 remote set for only €10, but unfortunately they stopped selling it - hopefully only temporarily.

The model name is
**Funksteckdosen-Set 0175926**
if you ever find it being sold somewhere else.

When I got a new treadmill, its RF remote also messed with one of my sockets, but that was easily remedied by switching the socket to a new code - something not possible with auto-learn versions, so be aware of any new RF remotes/signals you, or your neighbour, might want to introduce to your homes.

> But nevertheless, maybe it's a good idea to support the Power-Adjustment through OpenMqttGateway.

I would assume that most people would usually want a signal as strong as possible, but having the choice would be a bonus.

Dattel, 2023-01-08:
Here is my sample implementation:

https://github.com/1technophile/OpenMQTTGateway/compare/development...Dattel:OpenMQTTGateway:development

Currently I ignored the "#  if simpleReceiving" MQTTtoRF-function because there seems to be a general bug since i always got an exception and a reboot on my ESP32 if i try to send through a simple topic (even without any changes on that function from me)

Dattel, 2023-01-10:
Thanks for supporting :-D

closed 2023-01-10 by Dattel