1technophile/OpenMQTTGateway #1385: Certificate Verification failed on Lets encrypt cert

drz2k, 2023-01-10:
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -9984
W: failure_number_mqtt: 11
W: failed, rc=-2
W: failed, ssl error code=-9984

We are trying to connect to a stackhero mosquitto instance via TLS. Cert is from Let's Encrypt. (https://letsencrypt.org/certs/trustid-x3-root.pem.txt). (Checking against FQDN d4i07v.stackhero-network.com)

Connection to stackhero is working fine with MQTT Explorer. But the OMG is stuck on the verification of the cert (copied completly with start and end tag). Stackhero suggest : "_espClient.setInsecure();"

1technophile, 2023-01-10:
I'm working on a branch that enables to not use of the certificate validation below:
https://github.com/1technophile/OpenMQTTGateway/tree/invalid-cert

You can try it and let me know

drz2k, 2023-01-10:
> You can try it and let me know

Works like a charm so far. Would love to see the Webbased installer to add the "don't check cert" option. So maybe this turns into a feature request :)

1technophile, 2023-01-10:
Thanks for the feedback. Yes, indeed, we can add something like this.

For my curiosity, do you mind sharing your use case?

drz2k, 2023-01-10:
We will be tracking small BLE devices for the elderly to raise an alert if they wander off. (Dementia)

closed 2024-01-03 by github-actions[bot]