1technophile/OpenMQTTGateway #1393: 1.2.0 does not read the data from MHO-C401

nonflammable, 2023-01-14:
1.2.0
{"id":"A4:C1:38:52:8F:62","mac_type":0,"name":"MHO_SAL","rssi":-92,"servicedata":"628f5238c1a48307e4128f0b64b40f","servicedatauuid":"0x181a"}

1.1.1
{"id":"A4:C1:38:52:8F:62","mac_type":0,"name":"MHO_SAL","rssi":-89,"brand":"Xiaomi","model":"LYWSD03MMC","model_id":"LYWSD03MMC_PVVX","tempc":19.58,"tempf":67.244,"hum":47.84,"batt":100,"volt":2.961}

Firmware: PVVX - MHO_C401_v39.bin

https://github.com/1technophile/OpenMQTTGateway/issues/1160

DigiH, 2023-01-14:
Hi @nonflammable,

with current versions of the decoder we had to include the broadcast name for clear identification and distinction of devices running ATC/PVVX custom firmwares.

Did you set the name `"name":"MHO_SAL"` during flashing of your device yourself?

If so, please change the name to anything starting with **ATC_** (**ATC_MHO_SAL** could be an option in your case), or let us know if the name was included as default with the firmware flashing.

nonflammable, 2023-01-14:
https://github.com/1technophile/OpenMQTTGateway/releases/download/v1.2.0/esp32dev-ble-firmware.bin

https://github.com/pvvx/ATC_MiThermometer/blob/master/MHO_C401_v39.bin

DigiH, 2023-01-14:
Hi @nonflammable 

can you also use the **test** web install at
https://docs.openmqttgateway.com/dev/upload/web-install.html

If not let us know so we can supply the individual binary.

nonflammable, 2023-01-14:
I can't upgrade via mqtt
`mosquitto_pub -t "home/OpenMQTTGateway_ESP32_BLE/commands/firmware_update" -m '{
  "version": "test",
  "password": "OTAPASSWORD",
  "url": "https://github.com/1technophile/OpenMQTTGateway/blob/gh-pages/dev/firmware_build/esp32dev-ble-firmware.bin"}'`

I will try flash with FDT tomorrow

DigiH, 2023-01-14:
Hi @nonflammable, appologies, the link should have been

https://github.com/1technophile/OpenMQTTGateway/raw/gh-pages/dev/firmware_build/esp32dev-ble-firmware.bin

which should work with the MQTT update as well.

nonflammable, 2023-01-15:
test version works fine

DigiH, 2023-01-15:
Thanks for the feedback

Noki, 2023-01-25:
I have the same issue on an M5Stack Atom Lite. The service data is not decoded anymore. I did a fresh install via the web uploader with an erase, the result remains the same.

DigiH, 2023-01-25:
Hi @Noki , the test build mentioned above would have been overwritten by newer and nightly builds, so what you installed through the web uploader didn't have the fix in it.

Please hold on while a new test build run is running …

DigiH, 2023-01-25:
@Noki 

Please try
https://docs.openmqttgateway.com/dev/upload/web-install.html
and let us know if it's fine for you.

Noki, 2023-01-25:
@DigiH Thanks a lot! My problem is solved and all sensors show correct values again.

DigiH, 2023-01-30:
Closing as the fix is included with the release of OpenMQTTGateway 1.3.0

closed 2023-01-30 by DigiH