1technophile/OpenMQTTGateway #1394: Right image / Disable HomeAssistant topics?

ilgrank, 2023-01-14:
Hi
Not really an issue, sorry, just a request for clarification:
I have several LYWSD03MMC / ATC with pvvx firmware, and they do work perfectly with OpenMQTTGateway, thanks! (running on **1.2.0** btw)

Since I'm **passively** reading beacons with these devices, I figured out the best firmware for my ESP32 would be: `esp32dev-ble-cont` 
besides the expected topics tho, i have a homeassistant one with basic (ie: non-json) readings.
I've picked the wrong fw-image? is there an image with continuos scanning and no additional topics? OR, where in `User_config.h` do I disable these additional publications?
Thanks!

1technophile, 2023-01-15:
Hello,

You can disable the discovery with this mqtt command:
https://docs.openmqttgateway.com/use/gateway.html#deactivate

I would advise to publish it with a retain flag so as to keep it upon restarts.

ilgrank, 2023-01-15:
Many thanks for the quick reply!
Should I want to have it by default by building my own image, what config option does that?

1technophile, 2023-01-15:
This macro will disable the MQTT discovery:
https://github.com/1technophile/OpenMQTTGateway/blob/cc81b0ba44d8361cbc5ab4b7efa9d559106b4386/platformio.ini#L470

ilgrank, 2023-01-15:
Many thanks!

closed 2023-01-15 by ilgrank