1technophile/OpenMQTTGateway #1395: Reduce power consumption for BLE devices by defaulting Active Scanning to off

ilgrank, 2023-01-15:
According to @pvvx, active scanning does increase BLE devices power consumption (thus reducing their battery life) (original link [here](https://github.com/pvvx/ATC_MiThermometer/issues/252)):

> Active scanning increases battery consumption by many percent. The BLE device responds to active requests, which consumes additional power.

1technophile, 2023-01-15:
I agree on this fact, nevertheless having passive per default could generate more support with users not seeing their devices.
I have another idea in mind that should make everybody happy, give me some time to implement it.

ilgrank, 2023-01-15:
Thanks! You mean something like an MQTT command to enable/disable it?

Also, while I'm at it, and as I don't want to flood you with issues, since this is just another suggestion.. IMHO it would be great to allow to retain a (password-protected) configuration webpage after the initial wifi config one.

I spent most of the time configuring OpenMQTT with an issue (my fault) I had with my MQTT broker, and so every time I had to re-set, re-configure and hope for the best, until I found what my problem was, dozens of resets later.

Having a webpage (tasmota-like) for configuration would have made changing config way easier (and since I had issue connecting with my broker, using MQTT to reconfigure was not an option)

Again, thanks!

1technophile, 2023-01-15:
> Thanks! You mean something like an MQTT command to enable/disable it?

We have already this for info 
https://docs.openmqttgateway.com/use/ble.html#setting-if-the-gateway-use-active-or-passive-scanning

For the web page, this is something that I'm considering

ilgrank, 2023-01-15:
Whoops, sorry, completely missed that page :(

Double thumbs up for the Web page, it would make things a lot more practical 👍

ilgrank, 2023-01-15:
While I do receive passive beacons turning active scanning off, they don't get decoded. , eg:
```
{
  "id": "A4:C1:38:36:20:D6",
  "mac_type": 0,
  "rssi": -55,
  "servicedata": "d6203638c1a44708a013180c641705",
  "servicedatauuid": "0x181a"
}
```
..i'm missing something else?

closed 2023-01-15 by ilgrank

1technophile, 2023-01-15:
And they are decoded when you turn active scan ON ?

If it is an LYWSD03MMC with PVVX, you may have to try to change the name to have ATC_ in it.
Or alternatively, you can flash this patch which is less restrictive than the production one:
https://docs.openmqttgateway.com/dev/upload/web-install.html

ilgrank, 2023-01-15:
made all the tests with retained messages as you suggested

had the sensor data -> turned active scan off -> no more data
turned active scan back on -> still no sensor data
power cycled the ESP32 (still with active scan retained message) : now I have data

1technophile, 2023-01-15:
did you try this different binary:
https://docs.openmqttgateway.com/dev/upload/web-install.html

ilgrank, 2023-01-15:
sorry, edited while you were replying, I'm already flashing it, will take some minute to reconfigure

1technophile, 2023-01-15:
If you don't erase you don't have to reconfigure for info

ilgrank, 2023-01-15:
**I can confirm that with this version it works perfectly!**
I also can confirm that the active scan is still off as I see way less devices (actually, just the Xiaomi sensors)

1technophile, 2023-01-15:
Great, we will integrate it into the development branch for the next release

ilgrank, 2023-01-15:
again, my thanks! 👍 🥇

ilgrank, 2023-01-15:
btw: I've checked the "Contribute" page, and I would want to make a donation, but apart from contributing code (alas, I'm not able to) is there a way to tip the project (or, as the page says, its contributors?)

1technophile, 2023-01-15:
Thanks for asking, to tip the project you can purchase the Theengs app https://app.theengs.io/
Or a Theengs Plug if you are in the US or Canada
https://shop.theengs.io/

ilgrank, 2023-01-15:
Alas, not in USA/CA, and will surely try Theengs in the future, hope that that PP address is still fine :)

1technophile, 2023-01-15:
Send me an email to florian@theengs.io I can share you a link

1technophile, 2023-01-15:
Received on the previous address thanks

DigiH, 2024-03-13:
Hi @jacol-code 

There are two different scanning intervals for passive and active scanning

**TimeBtwRead** - interval between passive scans
https://github.com/1technophile/OpenMQTTGateway/blob/development/main/config_BT.h#L103

and
**TimeBtwActive** - interval between active scans
https://github.com/1technophile/OpenMQTTGateway/blob/development/main/config_BT.h#L84

hence your 
> … come every minute or so

as the default for both is 55 seconds.

So you can adjust these separately to your requirements.

Hope this clarifies things.