1technophile/OpenMQTTGateway #1396: [FR] What to do with Heltec HTIT-WB32LA ? (since fw1.2.0)

ozett, 2023-01-15:
**Is your feature request related to a problem? Please describe.**
Just a Question: Had the Heltec HTIT-WB32LA lying around for some years, never used.
now read from support from [fw 1.2.0.](https://github.com/1technophile/OpenMQTTGateway/releases/tag/v1.2.0) But:
What use can i make of it? Can i receive all worldwide lora broadcasts?  or do i have to make my own pair of sender/receiver?
[the docs are ](https://docs.openmqttgateway.com/use/lora.html#receiving-data-from-lora-signal)a bit short for me

| module |
|:---:|
[image]

**Describe the solution you'd like**
some hints how to use this. Or how it is used by other with the fw 1.2.0 ?

**Describe alternatives you've considered**
questioning gthe web for years..

**Additional context**
scrrenshot above

NorthernMan54, 2023-01-15:
As Lora is a short/medium range protocol you would need a sender to leverage it.

As the RF portion of your board is 868–915mhz used is limited for straight RF ( unless you have something that transmits at that frequency )

The other use case is Bluetooth, if you have Bluetooth sensors, it could potentially receive them.

ozett, 2023-01-16:
thanks for comment, now i know i need a sender, too.
i have to dive more into it on youtube, i guess.

last question about the frequency-range: can i use the new V3 version of the board as a sender?
the freq-range seems different than what is stated to my old board. did that change for LoRa? 
or is that some misunderstanding of LoRA details of my side?

| freq-range ? |
|:---:|
[image]

1technophile, 2023-02-04:
> last question about the frequency-range: can i use the new V3 version of the board as a sender?

As long as the receiver use the same frequency, modulation and parameters yes.

closed 2023-02-04 by 1technophile