1technophile/OpenMQTTGateway #1400: V1.2.0 doesnt work on M5Stack atom lite

aschr, 2023-01-17:
II tried to install version 1.2.0 via "Upload from the web" on a M5Stack atom lite. After entering the WiFi credentials and MQTT configuration, the board restarts. The credentials are ignored and the AP starts instead. None of the settings seem to be saved.

The same procedure with version 1.1.1 ran without problems. What am I doing wrong? Was anyone already successful with V 1.2.0 on an atom lite?

Thanks in advance for helpful tips!

Translated with www.DeepL.com/Translator (free version)

1technophile, 2023-01-17:
Just flashed my atom-lite with `esp32-m5atom-lite` from the web upload v1.2.0 and got it connected to the HA provided broker:
[image]

Any specific with your network and broker ?

aschr, 2023-01-18:
Hello @1technophile thanks for the quick reply. 

I tried it first with another atom lite and had the same problems. 
A reboot of my ap/router now seems to have fixed that.
A strange behavior, but what the heck, now it works.

closed 2023-01-18 by aschr