1technophile/OpenMQTTGateway #1404: [FR] Timestamp, Retain messages, Filter by name/prefix

ilgrank, 2023-01-19:
Hi
Again, thanks for OpenMQTTGateway!
I'm using it since some day on an ESP32 and.. it's great :)

I have some ideas to propose, that IMHO would be of use:
- Timestamp messages. Since we configure an NTP anyways, it would be useful to have the option to timestamp messages. Since I have several types of sensors, some which report every few seconds and others that report every few hours, it would be useful to know *when* a message was received.. 
- ..and of course this requires the ability to *retain* messages. I've read previous discussions about this, but I have a new argument to support my request: in the scenario above, not having the option to retain messages messages means that any new client subscribing to my broker needs to wait for a sensor to advertise new data.. and it could be second or dozens of minutes. Having the last reading as a retained message would mean knowing immediately the latest reading.

I've already work-arounds in place for the points above (another client that is subscribed to OMQTT topics and re-publishes them with timestamps and retain flag), but I think it would be useful to have these features built-in.

..and last (sry!):
- There are (alas) dozens of BLE devices around me it seems. While I've filtered by whitelist (thanks!) it seems to me that adding a filtering by name/prefix would prove to be more flexible.

As always, thanks for your great work!

1technophile, 2023-01-20:
Hello,

> Timestamp messages: Since we configure an NTP anyways, it would nice to have the option to timestamp messages. E.g.: I have several types of sensors, some which report every few seconds and others that report every few hours: it would be useful to know when a message was received..

This can be managed on the controller side, what do you think would be the added value of adding a timestamp at the firmware level versus adding it when the controller receives it?

ilgrank, 2023-01-20:
pardon me.. when you say "_on the controller side_", you mean on the devices connecting to the broker that subscribe to the sensors topics?

1technophile, 2023-01-20:
By Controller, I meant HA, OpenHAB, NodeRed...

ilgrank, 2023-01-20:
oh, well, that's because I don't use any, as I wrote in my initial post:

> I'm not using HA or any other sofware, just JS clients connecting to the broker)

:)

1technophile, 2023-01-21:
Ok I understand now your need, thanks for sharing

ilgrank, 2023-02-14:
Hi
researched around, and found out that It's quite common to have retain set on sensors topics, so I've gathered some 'evidence' to show that's quite common requirement (hence, it could be useful to more people than just me):

Tasmota ecplicitly supports it with a dedicated funcion: **[SensorRetain](https://tasmota.github.io/docs/MQTT/#retained-mqtt-messages)**

Shelly devices also support retaining, as documented [here](https://shelly-api-docs.shelly.cloud/gen1/#mqtt-support) 

and finally, Steves' internet Guide had an excellent explanation (way better than mine) on why having the Retained flag set is desiderable on sensors, [here](http://www.steves-internet-guide.com/mqtt-retained-messages-example/):

> [...]this can be very useful, as for example, if you have sensor publishing its status only when changed e.g. Door sensor. What happens if a new subscriber subscribes to this status?
> Without retained messages the subscriber would have to wait for the status to change before it received a message.
> However with the retained message the subscriber would see the current state of the sensor
> 

Since we're talking about devices that are not MQTT enabled by themselves here, OMQTTG is the only entry point for such devices to have their info published in real time, and without requiring complex and resource-hungry controller systems.

Sorry coming again at this, and again, thanks :)

1technophile, 2023-02-14:
Having the capability to retain messages would be an easy add.

ilgrank, 2023-02-14:
thanks @1technophile , high hopes for that.. and _timestamping_, if possible.
All of the above sensors (Shelly, Tasmota) do timestamp their message, if possible (ie: not running on battery).
eg:, a sample tasmota payload:
```
{
  "Time": "2023-01-28T22:55:50",
  "ENERGY": {
    "TotalStartTime": "2020-04-10T18:33:07",
    "Total": 0.408,
    "Yesterday": 0,
    "Today": 0,
    "Period": 0,
    "Power": 0,
    "ApparentPower": 0,
    "ReactivePower": 0,
    "Factor": 0,
    "Voltage": 0,
    "Current": 0
  }
}
```
payload unaltered, just used bold on relevant values
Thanks :)

ilgrank, 2023-04-01:
Apart from MAC filtering by prefix, everything has been added in 1.5.0, thanks!

closed 2023-04-01 by ilgrank