1technophile/OpenMQTTGateway #1410: HA MQTT password length

perseids29, 2023-01-21:
The password for my Mosguitto server in HA is 64 characters long by default and the WiFiManager only accepts 60 characters

1technophile, 2023-01-22:
Why not creating a shorter password, 60 gives already room?

perseids29, 2023-01-22:
I would have to change it in all my other devices, which would be a pain

Sent via the Samsung Galaxy S22 Ultra 5G, an AT&T 5G smartphone

1technophile, 2023-01-22:
You could create a new one instead. The other solution could be to change the size of yhe password length below:
https://github.com/1technophile/OpenMQTTGateway/blob/a3603c6a788609281743eeb72aaa44ff58a2db4f/main/User_config.h#L147

And build the firmware:
https://docs.openmqttgateway.com/upload/builds.html#configure-upload-with-platformio

1technophile, 2023-02-04:
Closing, let me know if any other question

closed 2023-02-04 by 1technophile