1technophile/OpenMQTTGateway #1436: Support for new Heltec WiFi LoRa 32 v3 (HTIT-WB32LAF)

RynoM, 2023-02-04:
**Is your feature request related to a problem? Please describe.**
Ordered a v2 since I knew it was supported out-of-the-box, but received a v3..
I've tried flashing the board using the web tool, but there I get "Failed to initialize". The flash download tool seems to work and shows "finished", but nothing happens. LED on the board doesn't flash, the display doesn't turn on and no AP gets created.
Since from v2 to v3 the MCU and the LoRa chip have changed, I feel like I've tried what I can but it just doesn't work because the hardware is different.

**Describe the solution you'd like**
Support (in the web tool?) for the new heltec wifi lora 32 v3 board.

**Describe alternatives you've considered**
I've tried the different heltec firmware versions.

**Additional context**
Pretty new to all this, but I'm very willing to help test.

NorthernMan54, 2023-02-04:
I looked at that board's data sheet pretty closely, and found it used a different radio transceiver chipset, and that chipset did not support the OOK/ASK direct mode used by the receiver.  So unfortunately it will not work as a 433Mhz receiver.  But you would be able to leverage it as a bluetooth receiver.

RynoM, 2023-02-04:
I saw 433 was listed under the operating frequencies so thought there would be something possible, but that is unfortunate.

closed 2023-02-14 by NorthernMan54

ryanm101, 2023-03-03:
Should this be reopened as we'd want support for 868Mhz listening too as devices such as ecowitt weather stations use 868Mhz.