1technophile/OpenMQTTGateway #1437: No service found when directly writing on a characteristic

ov1d1u, 2023-02-05:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I'm trying to write to a device characteristic by following instructions at https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt. The payload I'm sending over MQTT is:

```
{
  "ble_write_address":"a4:c1:38:50:9e:4d",
  "ble_write_service":"6e400000-b5a3-f393-e0a9-e50e24dcca9e",
  "ble_write_char":"6e400003-b5a3-f393-e0a9-e50e24dcca9e",
  "ble_write_value":"bf626d541868626d4e189a62724900ff",
  "value_type":"HEX",
  "ttl":4,
  "immediate":true
}
```

but all it says when I'm publishing the message is:

```
N: BLE Connect begin
N: Failed to find service UUID: 6e400000-b5a3-f393-e0a9-e50e24dcca9e
N: Failed to find service UUID: 6e400000-b5a3-f393-e0a9-e50e24dcca9e
N: Failed to find service UUID: 6e400000-b5a3-f393-e0a9-e50e24dcca9e
N: Failed to find service UUID: 6e400000-b5a3-f393-e0a9-e50e24dcca9e
``` 

Looks like either I'm forgetting a step or a service refresh must be done before calling

`BLERemoteService* pRemoteService = m_pClient->getService(service);`

because if I add `m_pClient->getServices(true);` above that line in ZgatewayBLEConnection.ino the service is found and the caracteristic is being written.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the esp32-lolin32lite-ble to a Lolin32 board
2. Using `/commands/MQTTtoBT/config`, publish a ble_write payload to a device's characteristic

**Expected behavior**
The characteristic should be found.

**Environment (please complete the following information):**
* OpenMQTTGateway v1.3.0
* NimBLE-Arduino 1.4.1

**Additional context**
The device I'm trying to write to is a Godrej Aer Smart Matic.

h2zero, 2023-02-06:
What is the device you are connecting to? I have noticed this issue coming up for a few folks with certain devices that do no support attribute discovery by uuid.

ov1d1u, 2023-02-07:
It's some air freshener device I got from India: https://www.godrejaer.com/collections/godrej-aer-smart-matic

How can I validate if the device support attribute discovery by uuid?

closed 2023-09-27 by github-actions[bot]