1technophile/OpenMQTTGateway #1438: "GPIO output gpio_num error" after upgrading from v0.9.16 to v1.3.0

tomcat-xx, 2023-02-06:
Hello,

since I upgraded one of my ESP32 to 1.3.0, serial monitor is flooded by following errors (while BT Gateway is working):

> 15:20:23.746 -> E (58842476) gpio: gpio_set_level(226): GPIO output gpio_num error
15:20:23.746 -> E (58842481) gpio: gpio_set_level(226): GPIO output gpio_num error
15:20:23.746 -> E (58842487) gpio: gpio_set_level(226): GPIO output gpio_num error
15:20:23.746 -> E (58842493) gpio: gpio_set_level(226): GPIO output gpio_num error
15:20:23.748 -> E (58842499) gpio: gpio_set_level(226): GPIO output gpio_num error

Could just assume that it has to do with LED control, as I saw some related changes when adjusting settings in user_config.h and my board (AZ-Delivery ESP-32 Dev Kit C V4) only has power LED. I found similar when searching the web.

For flashing I used Arduino 2.0.3 (before 1.8.19). First I used "ESP32 Dev" as board definition as before. Then I also tried "ESP32-WROOM" with same result.

Not sure if it would work better with PlatformIO. When I started with OpenMQTTGateway as my first ESP32 project, I could not get it working and therefore used IDE. Anyway I will give PlatformIO a new try (e.g. to get rid of manually adjusting settings in user_config.h for every upgrade), but now sure if it will solve current issue.

Thanks for helping. Maybe someone has same issue. Ore same board without issues ...

1technophile, 2023-02-06:
Hello,

I had this error with the latest Arduino ESP32 environments. You can downgrade your Arduino ESP32 framework or investigate the error. It may be due to the pin's initialization/definition indeed.

I will investigate it also

tomcat-xx, 2023-02-06:
Thanks for quick response! Could you also give hint which version to use?

1technophile, 2023-02-06:
Try this one https://github.com/espressif/arduino-esp32/releases/tag/1.0.6

tomcat-xx, 2023-02-06:
So the highest 1.x version. Thanks. Worked.

closed 2023-02-06 by tomcat-xx