1technophile/OpenMQTTGateway #1439: Support: OLIMEX ESP32 EVB gateway board w. LORA (SX1776)

g6094199, 2023-02-07:
Since the OLIMEX ESP32 gateway is already supported this one, the [OLIMEX ESP32 EVB ](https://www.olimex.com/Products/IoT/ESP32/ESP32-EVB/open-source-hardware) would be a very nice addition.
Many of the OLIMEX boards (also the EVB) come with a connector called UEXT and olimex offes a bunch of UEXT compatible hats for it. One very interesting is the [MOD-LoRa868](https://www.olimex.com/Products/IoT/LoRa/LoRa868/open-source-hardware) which is just plug n play to upgrade the ethernet gateway (EVB) to a lora ethernet gateway! No soldering needed. also a fitting case is offered. 

this would be a hassle free option for a PoE powered LORA gateway!

is there an interest in supporting it? then i would order it.

1technophile, 2023-02-07:
I may have one let me check

g6094199, 2023-02-08:
ordered one anyway.... this is the best ready to go option on the marked i can find.

1technophile, 2023-02-08:
I confirm that I have one, if you want to share an environment proposal I can test it

g6094199, 2023-02-14:
>if you want to share an environment proposal I can test it

and with "environment proposal" you mean....?

1technophile, 2023-02-14:
Something like this:
https://github.com/1technophile/OpenMQTTGateway/blob/d09f13bf86524cc055b26aa34bd6097eedab12e2/environments.ini#L353

closed 2023-09-21 by github-actions[bot]