1technophile/OpenMQTTGateway #1461: 868Mhz rolling code

geromey, 2023-02-15:
Hello, I am trying to controls 868Mhz shutters with rolling code, do you think that OpenMQTTGateway could help me ?

PS: I'm using a ESP32 and I'll soon receive an antenna CC1101

Thank you !

1technophile, 2023-02-15:
Hello,

rolling codes are not supported and I don't think you will find a way to support it.
Your best chances are wires in this case.

Dattel, 2023-04-16:
i'm using Homeassistant with Node-Red and filtering all 4-rolling codes to one dedicated action... it looks a bit ugly, but it works 

have a look

[image]

same way arround... if i trigger an action in homeassistant, i send a single code through openmqtt and it works fine - even if you always send the same code.
But with omg1.5.1 it's broken due too some changes :-(

NorthernMan54, 2023-04-17:
What build/RF Module are you using for this ?

NorthernMan54, 2023-04-17:
RF/RCSwitch or PiLight or RTL_433_ESP ?

Dattel, 2023-04-17:
i have two different setups, 
but don't wanna spam these topic with another issue -> so go on here https://github.com/1technophile/OpenMQTTGateway/issues/1609

closed 2023-09-21 by github-actions[bot]