1technophile/OpenMQTTGateway #1467: Support request for atomax Skale II (bluetooth)

Cadsters, 2023-02-18:
**Is your feature request related to a problem? Please describe.**
I would like my ble scale (atomax Skale II) to be added to the suported bluetooth devices

**Describe the solution you'd like**
Read the weight and batlery level, detect button press, and be able to  send commands to set the device.

**Additional context**
The scale https://skale.cc/en/

Ble API Specification for Skale I and I
https://de.decentespresso.com/scale_api

DigiH, 2023-02-19:
With the currently include [READ and WRITE commands](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt) in OpenMQTTGateway you can already read any data as specified in your link, as long as it doesn't require a Notify, and write any setting as specified in the documentation.

If you would also like the scale to be included in [Theengs Decoder](https://github.com/theengs/decoder) so that the properties are automatically decoded, we would need to see if and how the scale might also have this information in its advertising data - which is not mentioned in the documentation.

You can help by providing some sample message output from your scale, as seen in [MQTT Explorer](https://github.com/thomasnordquist/MQTT-Explorer/releases/tag/v0.3.5), and with [publishing advertisement and advanced data](https://docs.openmqttgateway.com/use/ble.html#advanced-publishing-advertisement-and-advanced-data-default-false) set to **true**.

Cadsters, 2023-02-19:
Yes i would like to have the scale included in Theengs Decoder please.
I have flashed the latest omgw (esp32dev-ble-datatest) This is the data i can retreive from mqtt explorer:
`{"id":"D3:72:9D:F3:1C:E5","mac_type":1,"adv_type":0,"manufacturerdata":"4c00121910441292050996d091d1645f3d12c74e70124d1cd31ad90033","rssi":-96}`
If i can do more please let me know.

DigiH, 2023-02-19:
> {"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81000000ff","rssi":-77}

Thanks, this is the kind of information needed, but we'd also need some additional data as to what the scale was set to and displayed - what weight, in g, kg or oz, lb - to be able to see if and what information might be included in the manufacturerdata.

Cadsters, 2023-02-19:
Oké here some new data:
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81280100ff","rssi":-79}`
The scale is set to g (grams)  and the number showing is 29.6

DigiH, 2023-02-19:
Great, the weight is already decoded now :) 

2801 - little endian - 0128 = 296 / 10 = 29.6 g

If you change the scale to different weight settings, oz, lb or whatever else it has, so that we can see how the different unit might be registered.

Cadsters, 2023-02-19:
may sound stupid but right now i am unable to change from g to oz.... will search the internet for a howto.

DigiH, 2023-02-19:
> may sound stupid but right now i am unable to change from g to oz.... will search the internet for a howto.

From your above linked online manual

https://skale.gitbooks.io/skale-ii-user-manual/content/operation.html

> Touch and hold and button to power on and to switch measurement units between gram and oz. (Part of the model)

DigiH, 2023-02-19:
> I've tried that the past 10 minutes but nothings happening.

I know, the manual description is also a bit unclear ;) any chance with the app maybe?

Also not that much clearer in the FAQ section

https://skale.gitbooks.io/skale-ii-user-manual/content/common-problems.html

One additional question though with the already recorded manufacturerdata above. Was the scale connected to the power adapter in both cases?

If so, could you also do some reading with only 4 AAA batteries in the scale?

Also some different reading with the LED display light being ON and OFF. But this again would need the to be able to set it in the app I suppose.

Cadsters, 2023-02-19:
tried other batteries, usb-powersuply nothing. In the app i can switch to oz but the display on the scal still shows grams. So the app calculates to oz but nothing changes in the scale..... i'll get back to this.

DigiH, 2023-02-19:
> tried other batteries, usb-powersuply nothing. In the app i can switch to oz but the display on the scal still shows grams. So the app calculates to oz but nothing changes in the scale..... i'll get back to this.

Sorry if I wasn't clear enough above. I meant could you do a reading with the message data when the scale is only powered by the 4 AAA batteries and no USB is connected, totally separate from the g/oz issue?

Cadsters, 2023-02-19:
The manufacturerdata only changes when i weigh somthing. If i hold the button to switch it off it displays SLEEP but this does not change the manufacturerdata.

DigiH, 2023-02-19:
> The manufacturerdata only changes when i weigh somthing. If i hold the button to switch it off it displays SLEEP but this does not change the manufacturerdata.

Yes, could you post something when you weigh an item, but only with batteries in the Skale and not connected to the USB, which I assume that was the case with your two readings above.

Cadsters, 2023-02-19:
oké, display shows 93.1
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81a30300ff","rssi":-64}`

Cadsters, 2023-02-19:
and a other object weighs 123.9 (all on batteries btw)
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81d70400ff","rssi":-69}`

DigiH, 2023-02-19:
> 93.1
> `{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81a30300ff","rssi":-64}`

Thanks a lot, and this was with batteries inside, but no USB conencted, correct?

How were the previous messages done above, also the same as now, only with batteries, but no USB?

Cadsters, 2023-02-19:
all on batteries.

Cadsters, 2023-02-19:
i will try the 123.9 object without batteries on usb.

Cadsters, 2023-02-19:
123.9 object on usb without batteries:
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81d70400ff","rssi":-56}`

DigiH, 2023-02-19:
Thanks, so there doesn't seem to be any battery/USB infromation in the manufacturerdata.

I'll create a test decoder for you to try out with your g setting, but for a final merge into Decoder at some stage we should find out the oz changed setting and message information, hopefully being included in the manufacturerdata then, so that the decoder will be useful for all users.

The test build takes about an hour …

Cadsters, 2023-02-19:
oke, i'v asked on discourd howto set to oz, hopefully someone knows. Thanks for your time!!!

Cadsters, 2023-02-19:
found [this](https://github.com/decentespresso/de1app/commit/b7f4a7d29f721226d6fe18190a31c0b6b4486a2f) regarden usb or batteries

DigiH, 2023-02-19:
> found [this](https://github.com/decentespresso/de1app/commit/b7f4a7d29f721226d6fe18190a31c0b6b4486a2f) regarding usb or batteries

Thanks, I think that was only related to the Skale I bug, as also mentioned in the API information about battery level
https://de.decentespresso.com/scale_api

Cadsters, 2023-02-19:
right i have got a Skale 2

DigiH, 2023-02-19:
Hi @Cadsters 

the test build has finished. Please install `esp32dev-ble-datatest` again and post some messages, with confirmation of the correct weight.

The web install is available at

https://docs.openmqttgateway.com/dev/upload/web-install.html

or you can download the firmware directly from

https://github.com/1technophile/OpenMQTTGateway/raw/gh-pages/dev/firmware_build/esp32dev-ble-datatest-firmware.bin

Cadsters, 2023-02-19:
Great! `{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81d60400ff","rssi":-44,"brand":"Atomax","model":"Skale I/II","model_id":"SKALE","type":"SCALE","cidc":false,"weight":123.8}`

DigiH, 2023-02-19:
Thanks for the confirmation!

Hopefully there will be some feedback about the setting to **oz** from Discord in the next few days

Also, when you get the chance, can you see if setting `intervalacts` to very high, with `{"intervalacts":600000}` (10 minutes)

https://docs.openmqttgateway.com/use/ble.html#setting-the-time-between-active-scanning

the manufacturerdata is still reported regularly with then regular passive scanning only.

Once these two issues are addressed we can officially merge the Skale decoder.

Thanks for your corporation.

Cadsters, 2023-02-19:
I will report back when I have the data, for now thank you very much for your help!

DigiH, 2023-02-20:
@Cadsters - While the Skale app seems to be in Chinese only, this video shows how to switch between ounce **盎司**, which it was set to in the beginning of the video, and gram **克**.

Info button - then the the two segmented buttons.

Might help us finding the oz manufacturerdata :) 

https://www.youtube.com/watch?v=X5-fUhk_B9I

Cadsters, 2023-02-20:
Sure i can the the same only the app calculates the g to oz, not the scale the numbers on the display stay the same. I've mailed atomax to here how its do, lets hope the respond.

Cadsters, 2023-02-20:
I found some info on the [FCC ID Database](https://fccid.io/2AD5WSK02/Users-Manual/Users-Manual-3101303). The manual there says nothing about changing units from g to oz. So perhaps its only grams....?

DigiH, 2023-02-20:
Yes, I had a look at that as well, but I just couldn't imagine that the scale would not allow to display in oz as well, as it would mean people preferring oz could only use the app.

Also with the link you provided above to the API there is a command to send to the scale to set to oz, as well as LED display ON/OFF, so maybe trying to WRITE that command to the scale will do the trick.

You can use the nRFConnect app to send that command (0x02) easily and quickly.

Otherwise the scale really only displaying in gram would be a very bad design from their side.

Cadsters, 2023-02-20:
I have the nRFConnect app and can see and connect to the scale but have to study it a bit to figure out how to send the command. Or can you easely explain it to me

DigiH, 2023-02-20:
If you could post a screenshot of the scale when oyu have connected to it in nRFConnect, so that I can see which services are avaialbe there and which allow to write, Then I will probably be able to tell you which one to go in to write the (0x)02 command.

Cadsters, 2023-02-20:
[
[image]
[image]
](url)

Cadsters, 2023-02-20:
Found [code](https://github.com/Nailik/DIY-Scale/blob/master/src/Bluetooth.cpp) from someone who made a clone of the Skale II, perhaps some usefull info....

DigiH, 2023-02-20:
Yes, if you go into any of the above services, until you see the 

0000**ef80**-0000-1000-8000-00805f9b34fb

this is then were you would be able to enter and WRITE the 0x02 command

DigiH, 2023-02-20:
Or by wrtiwritingng with the OpenMQTTGateway [WRITE command](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt)

{
  "ble_write_address":"DB:C5:13:ED:92:8E",
  "ble_write_service":"0000ff08-0000-1000-8000-00805f9b34fb",
  "ble_write_char":"0000ef80-0000-1000-8000-00805f9b34fb",
  "ble_write_value":"02",
  "value_type":"HEX",
  "ttl":4,
  "immediate":true }

Cadsters, 2023-02-20:
so have switched to oz by nrfconnect displays 3.2
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81a10300ff","rssi":-54,"brand":"Atomax","model":"Skale I/II","model_id":"SKALE","type":"SCALE","cidc":false,"weight":92.9}`
without object its 0.0
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81000000ff","rssi":-51,"brand":"Atomax","model":"Skale I/II","model_id":"SKALE","type":"SCALE","cidc":false,"weight":0}`
other object display sees 1.0
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81280100ff","rssi":-51,"brand":"Atomax","model":"Skale I/II","model_id":"SKALE","type":"SCALE","cidc":false,"weight":29.6}`

DigiH, 2023-02-20:
So the display switches to oz, but the advertisement data always reports the weight in gram. 

Thanks for that confirmation! I will merge the decoder like this.

Cadsters, 2023-02-20:
found something strange after a tare with object then removed object (92.9 g) the display shows -92.8
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef8160fcffff","rssi":-56,"brand":"Atomax","model":"Skale I/II","model_id":"SKALE","type":"SCALE","cidc":false,"weight":6460.8}`
the weight shows 6460.8

DigiH, 2023-02-20:
Good find, let me quickly fix that. 

Fixed with https://github.com/theengs/decoder/pull/291

DigiH, 2023-02-20:
If you go to the above mentioned development test build page tomorrow you can install any build with the nightly Decoder including the new Skale decoder.

Cadsters, 2023-02-21:
I tried the dev build of `esp32dev-ble-datatest` (7ffdb5), [webflashed](https://docs.openmqttgateway.com/dev/upload/web-install.html)
It looks like the commit didn't make it.
`{"id":"DB:C5:13:ED:92:8E","mac_type":1,"adv_type":0,"name":"Skale","manufacturerdata":"ef81000000ff","rssi":-54}`

DigiH, 2023-02-21:
@Cadsters

Correct, sorry, I hadn't realised that the nightly OpenMQTTGateway dev builds don't currently link to the Decoder dev branch.

I created a new manual run and the builds can now be accessed on the dev page, indicated by the 83bece hash at the top of the page.

Cadsters, 2023-02-23:
Got reaction from atomax the say: "Please use the iOS Skale app to change the oz/g unit. " so i gues it's not working throug the button on the scale.
While playing a bit with nRFConnect i found a way to detect on wich unit it is set.
When it is set to gram:
`Unknown Service
0000ef83-0000-1000-8000-00805f9b34fb
Properties: READ
Parsed value: (0x) 00-29-00-02-37-51-A6-10-00-01-FF-FF-FF-00-00
Raw value: (0x) 00-29-00-02-37-51-A6-10-00-01-FF-FF-FF-00-00`
and when set to oz:
`Unknown Service
0000ef83-0000-1000-8000-00805f9b34fb
Properties: READ
Parsed value: (0x) 00-29-00-02-37-51-A6-10-00-01-FF-FF-FF-00-00
Raw value: (0x) 00-29-00-02-37-51-A6-10-00-01-FF-FF-FF-00-00`
Using this you could calculate the value in oz and still display it.
Finaly a question, dus omgw connect to the device? or if not, could this be done? Because when not connected the scale auto-switches off even when it's powerd via usb. Also it would decrease the lag a bit.

DigiH, 2023-02-23:
Thanks for finding this out @Cadsters.

OMG with Theengs Decoder only currently interprets and decodes the freely broadcast advertising data sent out by devices, without any connection to get any property values. If and how this might be implemented in the future is open, but your finding about the auto switch off is valuable input in this regard.

One thing you could do is use the [OMG READ command](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt) through your controller to regularly read your above finding, which should then prevent the Skale from switching off whenever it is in use.

I take it the two identical lines for gram an oz above are just a copy and paste issue!?! ;)

Also you not mentioning the previously mis-decoded negative weight I assume it is verified as fixed with your latest test install?

Cadsters, 2023-02-23:
I have corrected the reading of the setting of g/ozin the previous msg, and yes the negative value works now.
I will try you suggestion to do a regularly read command and see if the scale stays powered. Wil report back later this week.

Cadsters, 2023-03-10:
I have tried to read the weight values in a fixed interval but he scale switches of after 60 seconds after the latest weight event dispite this.  However, it stays on when there is an active connection (app-mode).  I also noticed that there is a delay (2-4 seconds) between the moment of weighing and the indication of the value in mqtt.
This behaviour and the fact that omgw does not make an active connection makes it less usable for me but is perhaps usefull for someone else. The issue can be closed as far as I'm concerned unless you need more data or think there is room for improvement.

DigiH, 2023-03-10:
Just to clarify, or maybe I wasn't clear enough above. You used the [OMG READ command](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt) to read a value from the scale, possibly the battery level or weight, although that would be in the broadcast, not just making sure the broadcasts are 'read' at a fixed interval, correct?

As the [OMG READ commands](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt) actually make a connection to the scale, read the value, and then disconnect again I assumed that this would prevent the scale fro switching off after 60 second, if the READ would happen every 30 seconds or so. Implementing as a switch in your controller, this could then be a 'keep scale on' switch - at least that was my idea.

DigiH, 2023-03-10:
Closing, as the weight broadcast decoder is merged and will be included in the next OMG release , but still open for continued discussion here about the possibility of keeping the scale from switching off after 60 seconds.

closed 2023-03-10 by DigiH