1technophile/OpenMQTTGateway #1469: After Connecting as Station, Webinterface is not available any more

4920441, 2023-02-19:
Hi,

I got a ttgo Lora32 (433 MHz)  from lilygo flashed by

`esptool.py --port /dev/ttyUSB0 --chip esp32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader_dio_80m.bin 0x8000 lilygo-rtl_433-partitions.bin 0x10000 lilygo-rtl_433-firmware.bin`

After a reboot the Lilygo goes into "AP Mode" and webinterface is fully functional.

BUT when I created the Client Setup  the ttgo reconnects to the correct accesspoint,  though the webinterface is not accessible any more. The Port is closed.

The Device itself is correctly connected with the SSID, but I can only ping it. No Webinterface port seems to be open:

```
nmap 192.168.64.172
Starting Nmap 7.80 ( https://nmap.org ) at 2023-02-19 17:20 CET
Nmap scan report for esp32-arduino.whatever.com (192.168.64.172)
Host is up (0.0032s latency).
All 1000 scanned ports on esp32-arduino.kh.ug-1.de (192.168.64.172) are closed
MAC Address: 3C:61:05:AA:BB:DD (Unknown)

Nmap done: 1 IP address (1 host up) scanned in 7.22 seconds

```

Anyway, in my configured mqtt Server  are incoming mqtt messages:

```16:17:02.964958 eth0  In  IP 192.168.64.172.53745 > 192.168.64.23.1883: Flags [.], ack 19, win 5678, length 0
16:17:07.649708 eth0  In  IP 192.168.64.172.53745 > 192.168.64.23.1883: Flags [P.], seq 1855:2136, ack 19, win 5678, length 281
16:17:07.692451 eth0  Out IP 192.168.64.23.1883 > 192.168.64.172.53745: Flags [.], ack 2136, win 63184, length 0
16:17:17.862823 eth0  In  IP 192.168.64.172.53745 > 192.168.64.23.1883: Flags [P.], seq 2136:2138, ack 19, win 5678, length 2
16:17:17.862848 eth0  Out IP 192.168.64.23.1883 > 192.168.64.172.53745: Flags [.], ack 2138, win 63184, length 0
16:17:17.862924 eth0  Out IP 192.168.64.23.1883 > 192.168.64.172.53745: Flags [P.], seq 19:21, ack 2138, win 63184, length 2
16:17:17.970371 eth0  In  IP 192.168.64.172.53745 > 192.168.64.23.1883: Flags [.], ack 21, win 5676, length 0
16:17:02.964958 eth0  In  IP 192.168.64.172.53745 > 192.168.64.23.1883: Flags [.], ack 19, win 5678, length 0
16:17:07.649708 eth0  In  IP 192.168.64.172.53745 > 192.168.64.23.1883: Flags [P.], seq 1855:2136, ack 19, win 5678, length 281
16:17:07.692451 eth0  Out IP 192.168.64.23.1883 > 192.168.64.172.53745: Flags [.], ack 2136, win 63184, length 0
16:17:17.862823 eth0  In  IP 192.168.64.172.53745 > 192.168.64.23.1883: Flags [P.], seq 2136:2138, ack 19, win 5678, length 2
16:17:17.862848 eth0  Out IP 192.168.64.23.1883 > 192.168.64.172.53745: Flags [.], ack 2138, win 63184, length 0
16:17:17.862924 eth0  Out IP 192.168.64.23.1883 > 192.168.64.172.53745: Flags [P.], seq 19:21, ack 2138, win 63184, length 2
16:17:17.970371 eth0  In  IP 192.168.64.172.53745 > 192.168.64.23.1883: Flags [.], ack 21, win 5676, length 0
```

The Serial Output looks also good so far, no error messagtes or something....

Is the Webinterface gone until I erase the nvram?

Thanks  a lot for clearifying.

Cheers

4920441

NorthernMan54, 2023-02-22:
Correct, the web interface is only available prior to configuring wifi

foorschtbar, 2023-03-11:
> Correct, the web interface is only available prior to configuring wifi

And how i change settings later?

DigiH, 2023-03-11:
After the set-up the credentials can be changed by MQTT commands

https://docs.openmqttgateway.com/use/gateway.html#change-the-mqtt-broker-credentials

1technophile, 2023-04-09:
Integrated into the development branch for testing:
https://community.openmqttgateway.com/t/webui-proof-on-concept-for-esp32-modules/2322/13

closed 2023-04-09 by 1technophile