1technophile/OpenMQTTGateway #1470: Add Heltec WiFi LoRa 32 v1 do the supported boards

Aggebitter, 2023-02-19:
Everything works except the SX1276 :-) Different pin out
DIO1 (GPIO33 v1) GPIO35 v2
DIO2 (GPIO32 v1) GPIO34 v2

**ref:**
https://docs.heltec.org/en/node/esp32/dev-board/hardware_update_log.html

NorthernMan54, 2023-02-20:
Am thinking if you change the board in the environment to `heltec_wifi_lora_32`

From

`board = heltec_wifi_lora_32_V2`

To

`board = heltec_wifi_lora_32`

It should pick up the correct settings for your board.

1technophile, 2023-02-25:
@Aggebitter could you confirm that the solution from NorthernMan works, and we will propose a new environment build

Aggebitter, 2023-03-16:
Got Build error. Looks like the pinouts are not defined

`S:/OpenMQTTGateway-1.4.0/main/ZgatewayRTL_433.ino: In function 'void MQTTtoRTL_433(char*, ArduinoJson::JsonObject&)':
S:/OpenMQTTGateway-1.4.0/main/ZgatewayRTL_433.ino:301:28: error: 'RF_MODULE_RECEIVER_GPIO' was not declared in this scope
       rtl_433.initReceiver(RF_MODULE_RECEIVER_GPIO, receiveMhz);
                            ^
S:/OpenMQTTGateway-1.4.0/main/ZgatewayRTL_433.ino: In function 'void enableRTLreceive()':
S:/OpenMQTTGateway-1.4.0/main/ZgatewayRTL_433.ino:331:24: error: 'RF_MODULE_RECEIVER_GPIO' was not declared in this scope
   rtl_433.initReceiver(RF_MODULE_RECEIVER_GPIO, receiveMhz);`

Something going on in rtl_433_ESP.h ?
// Agge

NorthernMan54, 2023-03-16:
As support for this board was a recent addition, what version are you running ?  The current development version does include this board

When the board is defined as heltec_wifi_lora_32, this file is included in the build

~/.platformio/packages/framework-arduinoespressif32/variants/heltec_wifi_lora_32/pins_arduino.h

Which includes the DIO definitions

Then rtl_433_ESP.h includes these compiler definitions

https://github.com/NorthernMan54/rtl_433_ESP/blob/04aaf31bb573fa7277f69a3dcc5d8f9e5faa0f75/src/rtl_433_ESP.h#L78

And then RF_MODULE_RECEIVER_GPIO is defined

https://github.com/NorthernMan54/rtl_433_ESP/blob/04aaf31bb573fa7277f69a3dcc5d8f9e5faa0f75/src/rtl_433_ESP.h#L134

Aggebitter, 2023-05-01:
Sorry for the very late respond. 

platform.ini: 
defines the wrong board
`default_envs = heltec-rtl_433` <-- this is defined as "heltec-rtl_433_v2" later in  enviroment.ini

enviroment.ini:
[env:heltec-rtl_433]
platform = ${com.esp32_platform}
board = heltec_wifi_lora_32_V2 <---  WRONG

I added new entries for "heltec_wifi_lora_32_V2" and changed the entries for "heltec-rtl_433" to V1 in both files then it worked

**platform.ini: line 69**

;default_envs = heltec-rtl_433_v2 <-- Added
default_envs = heltec-rtl_433

**enviroment.ini: line 722**

[env:heltec-rtl_433_v2]
platform = ${com.esp32_platform}
board = heltec_wifi_lora_32_V2
; ~/.platformio/packages/framework-arduinoespressif32/variants/.../pins_arduino.h
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager32}
  ${libraries.ssd1306}
  ${libraries.rtl_433_ESP}
build_flags =
  ${com-esp.build_flags}
; *** OpenMQTTGateway Config ***
  ;'-UZmqttDiscovery'          ; disables MQTT Discovery
  '-DvalueAsATopic=true'       ; MQTT topic includes model and device
  '-DGateway_Name="OpenMQTTGateway_heltec_rtl_433_ESP"'
; *** OpenMQTTGateway Modules ***
  '-DZgatewayRTL_433="rtl_433"'
  '-DZradioSX127x="SX127x"'
; *** ssd1306 Display Options ***
  '-DZdisplaySSD1306="HELTEC_SSD1306"'
;  '-DLOG_TO_LCD=false'         ; Enable log to LCD
;  '-DJSON_TO_LCD=true'
;  '-DLOG_LEVEL_LCD=LOG_LEVEL_NOTICE'
;  '-DDISPLAY_IDLE_LOGO=false'
;  '-DDISPLAY_METRIC=false'
custom_description = Gateway using RTL_433_ESP and RadioLib

[env:heltec-rtl_433]
platform = ${com.esp32_platform}
board = heltec_wifi_lora_32
; ~/.platformio/packages/framework-arduinoespressif32/variants/.../pins_arduino.h
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager32}
  ${libraries.ssd1306}
  ${libraries.rtl_433_ESP}
build_flags =
  ${com-esp.build_flags}
; *** OpenMQTTGateway Config ***
  ;'-UZmqttDiscovery'          ; disables MQTT Discovery
  '-DvalueAsATopic=true'       ; MQTT topic includes model and device
  '-DGateway_Name="OpenMQTTGateway_heltec_rtl_433_ESP"'
; *** OpenMQTTGateway Modules ***
  '-DZgatewayRTL_433="rtl_433"'
  '-DZradioSX127x="SX127x"'
; *** ssd1306 Display Options ***
  '-DZdisplaySSD1306="HELTEC_SSD1306"'
;  '-DLOG_TO_LCD=false'         ; Enable log to LCD
;  '-DJSON_TO_LCD=true'
;  '-DLOG_LEVEL_LCD=LOG_LEVEL_NOTICE'
;  '-DDISPLAY_IDLE_LOGO=false'
;  '-DDISPLAY_METRIC=false'
custom_description = Gateway using RTL_433_ESP and RadioLib

closed 2024-01-10 by github-actions[bot]