1technophile/OpenMQTTGateway #1475: Wifi setting when compiling?

ilgrank, 2023-02-21:
Hi
I've just discovered that when compiling using PlatformIO, the wifi_password setting override has to be typed without quotes, ie:
`-Dwifi_password="password!"` <-Works
while
`'-Dwifi_password="password!"'` <-Does **not** work

Now, to be honest, in the main docs it is correctly listed _without_ single quotes, but elsewhere else in the code, it shows with quotes, e.g. in:
`environment.ini: Line  349: ;  '-Dwifi_password="WIFI_PASSWORD"'`
`prod_env.ini.example Line  76:   '-Dwifi_password="MyWiFiAPPassword"'`

is it a typo or I'm missing something? (eg: Why this is the only setting override that does not require to single-quoted?)
Thanks!

ilgrank, 2023-02-21:
closing as this is just a curiosity, and I've found there's 
https://community.openmqttgateway.com/
for discussion, sorry

closed 2023-02-21 by ilgrank

DigiH, 2023-02-21:
Hi @ilgrank, I've been using the quoted
`'-Dwifi_password="password!"'`
for ages - at least since version 0.9.3, so I am more perplexed to now see your stated doc examples without the single quotes, and hearing that you are having build issues with the fully quoted one.

Interested to hear what @1technophile has to say about this :)

ilgrank, 2023-02-21:
To clarify: I'm not having build issues, it's just that if I do build with single quotes around the wifi_password parameter, it won't connect to my AP and default to the captive portal, while if I do remove the single quotes, it does connect.
When saying 

> in the [main docs ](https://docs.openmqttgateway.com/upload/builds.html#configure-upload-with-platformio)it is correctly listed without single quotes

I'm referring to [https://docs.openmqttgateway.com/upload/builds.html#configure-upload-with-platformio](https://docs.openmqttgateway.com/upload/builds.html#configure-upload-with-platformio)  (also updated the original post with the docs link)

DigiH, 2023-02-21:
> I'm referring to https://docs.openmqttgateway.com/upload/builds.html#configure-upload-with-platformio (also updated the original post with the docs link)

I know, that's what I meant that seeing this in the docs really puzzled me, as I've been using the fully quoted version for so long, without build issues or connection issues to my AP. Especially since I've only been using 

```
'-DESPWifiManualSetup=true'
'-Dwifi_ssid="mySSID"'
'-Dwifi_password="myPASSWORD"'
```
for so long on all my various OMG gateways. 

Just makes me wonder why the docs have this unquoted version, and how and why it is the only working version for you … and what the original peculiarity was for this.

ilgrank, 2023-02-21:
I had not tried
'-DESPWifiManualSetup=true' 
before, so out of curiosity, using it with along with quoted wifi password  *does* work for me

DigiH, 2023-02-21:
> '-DESPWifiManualSetup=true'

is required to correctly use the manual WiFi and MQTT credentials. So this was probably the issue.

The docs possibly need some adjustments though, unless, I am wondering, the unquoted version has something to do with the escaped characters in the example?!?

DigiH, 2023-02-21:
FYI - https://github.com/1technophile/OpenMQTTGateway/pull/1476

Thanks for pointing this out.

ilgrank, 2023-02-22:
Thanks for opening the issue :)