1technophile/OpenMQTTGateway #1480: Add compelling webinterface to BLE Gateway Binaries

rubenkr, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
- Currently BLE Gateway binary esp32dev-ble comes without http webinterface
- MQTT works but results are overwhly crowded, hard to read and presence detection requires additional measures

**Describe the solution you'd like**
- Add webinterface of [BLE Scanner](https://github.com/gromeck/BLE-Scanner) existing reposity as awesome work of [@gromeck](https://github.com/gromeck)

Exemplary view (work of [@gromeck](https://github.com/gromeck))
[image]

1technophile, 2023-02-25:
> and presence detection requires additional measures

Could you detail this please ?

Also, what would be the goal of this web interface as you have access to the information in the controller, or you can filter with MQTT client like MQTT explorer ?

rubenkr, 2023-02-27:
thanks for considering this feature. all data is available by the controller. this feature is solely focussing on the frontend to show the data in a instant way. As MQTT explorer are external and hard to read. This simple feature would work out of the box without a need of a MQTT explorer.

ilgrank, 2023-03-03:
I would also add that not everyone is using a controller :)

1technophile, 2023-04-09:
Web interface (not BLE scanner) added to the current development branch. See the progress here:
https://community.openmqttgateway.com/t/webui-proof-on-concept-for-esp32-modules/2322/13

Feel free to share your feedback there

closed 2023-04-09 by 1technophile