1technophile/OpenMQTTGateway #1482: Upload for lilygo-rtl_433 board results in boot loop

hanssipi, 2023-02-23:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Trying to upload the firmware for the lilygo-rtl_433 board.
Proces works and recieved 'All done' after uploading.
Unfortunatly the board reboots every 3 seconds, i see the openmqttgateway logo on the display blinking.
No acces point...

**To Reproduce**
Upload iliygo-rtl_433 software from the website.

**Expected behavior**
A working device

**Screenshots**
none.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V1.4.0).

**Additional context**
I only have one board (new) so i cannot try another. I did try uploading from another computer, same result.
Com driver on usb is  'USB Enhanced-Serial CH9102 (com3)'.  Windows 10.
Am i doing something wrong?
Is there a way to check if the board is not defective?
Board is 433Mhz CH9102F.

Thanks in advance for your help!

NorthernMan54, 2023-02-23:
Unfortunately you will need to connect to the serial console to see what is being output when it restarts

DigiH, 2023-02-23:
@hanssipi , could you also try the current dev build, just to see if it changes anything for you.

https://docs.openmqttgateway.com/dev/upload/web-install.html

And please leave the **Erase Flash** checkbox ticked.

hanssipi, 2023-02-24:
@NorthernMan54 
This is what the serial monito says:

N: Setup SSD1306 Display end
N: OpenMQTTGateway Version: d124cd
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4014482c  PS      : 0x00060630  A0      : 0x8014222b  A1      : 0x3ffb1b50  
A2      : 0x3ffdbe50  A3      : 0x0000ffff  A4      : 0x00000000  A5      : 0x0000ffff  
A6      : 0x00001000  A7      : 0x0000001e  A8      : 0x0000ffff  A9      : 0x00000100  
A10     : 0x0000fbff  A11     : 0x0000ffff  A12     : 0x00000529  A13     : 0x000000fc  
A14     : 0x00000000  A15     : 0x3ffb1b6a  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00010017  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4014482c:0x3ffb1b50 0x40142228:0x3ffb1b90 0x401411f6:0x3ffb1bc0 0x40141c0e:0x3ffb1c00 0x400ff74c:0x3ffb1cc0 0x400db605:0x3ffb1cf0 0x400dc197:0x3ffb1ed0 0x401058d2:0x3ffb1fb0 0x4008a4fa:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
N: 
************* WELCOME TO OpenMQTTGateway **************
N: Setup SSD1306 Display end
N: OpenMQTTGateway Version: d124cd
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4014482c  PS      : 0x00060630  A0      : 0x8014222b  A1      : 0x3ffb1b50  
A2      : 0x3ffdbef8  A3      : 0x0000ffff  A4      : 0x00000000  A5      : 0x0000ffff  
A6      : 0x00001000  A7      : 0x0000001e  A8      : 0x0000ffff  A9      : 0x00000100  
A10     : 0x0000fbff  A11     : 0x0000ffff  A12     : 0x00000529  A13     : 0x000000fc  
A14     : 0x00000000  A15     : 0x3ffb1b6a  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00010017  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4014482c:0x3ffb1b50 0x40142228:0x3ffb1b90 0x401411f6:0x3ffb1bc0 0x40141c0e:0x3ffb1c00 0x400ff74c:0x3ffb1cc0 0x400db605:0x3ffb1cf0 0x400dc197:0x3ffb1ed0 0x401058d2:0x3ffb1fb0 0x4008a4fa:0x3ffb1fd0

Rebooting...
ect...

"Exception was unhandled" so error?

hanssipi, 2023-02-24:
@DigiH 

dev page upload result is the same.

hanssipi, 2023-02-24:
@NorthernMan54 
Now after upload from the dev site i see another error:

************* WELCOME TO OpenMQTTGateway **************
N: Setup SSD1306 Display end
N: OpenMQTTGateway Version: v1.4.0
E (3086) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
W: failed to mount FS -> formating
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0xdd2593c3  PS      : 0x00060230  A0      : 0x80141625  A1      : 0x3ffb1c10  
A2      : 0x3ffdb390  A3      : 0x00000000  A4      : 0x3ffdb280  A5      : 0x3ffdb008  
A6      : 0x00000030  A7      : 0x3ffdb048  A8      : 0x801425b4  A9      : 0x3ffb1c00  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000030  A13     : 0x3ffdb038  
A14     : 0x00001000  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0xdd2593c0  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x5d2593c3:0x3ffb1c10 0x40141622:0x3ffb1c40 0x40141e35:0x3ffb1c80 0x400ffb0b:0x3ffb1cd0 0x400dba4b:0x3ffb1cf0 0x400dc5c3:0x3ffb1ed0 0x40105cfe:0x3ffb1fb0 0x4008a4fa:0x3ffb1fd0

Rebooting...

NorthernMan54, 2023-02-24:
@hanssipi Can you share a picture of your board, want to ensure it is the same model/type.  And am going to try and recreate your issue

hanssipi, 2023-02-24:
[image]

hanssipi, 2023-02-24:
[image]

DigiH, 2023-02-24:
Looks exactly like my board ;)

@hanssipi, do you have VSCode and Platformio installed and ready at your end?

If so, would you mind building and uploading a custom [env:lilygo-rtl_433] environment build with manual WiFi and MQYY broker credentials, by adding the following build_flags to the [env:lilygo-rtl_433] environment

```
'-DESPWifiManualSetup=true'
'-Dwifi_ssid="YOURSSID"'
'-Dwifi_password="YOURSPASSWORD"'
'-DMQTT_SERVER="192.168.X.XX"'
'-DMQTT_USER=""'
'-DMQTT_PASS=""'
'-DMQTT_PORT="1883"'
```

then building and uploading.

hanssipi, 2023-02-24:
@hanssipi, do you have VSCode and Platformio installed and ready at your end

No sorry.

NorthernMan54, 2023-02-24:
I followed the same steps, and was successful

Monitor output from the setup ( I run my mqtt server without any passwords, so the values are blank )

```
*wm:[2] WiFi Scan ASYNC completed in 5112 ms
*wm:[2] WiFi Scan ASYNC found: 4
*wm:[2] Portal Timeout In 101 seconds
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Scan is cached 14339 ms ago
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Scan is cached 15395 ms ago
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Scan is cached 15471 ms ago
*wm:[2] <- HTTP Wifi 
*wm:[2] Scan is cached 24624 ms ago
*wm:[1] 4 networks found
*wm:[2] DUP AP: The_Beach
*wm:[2] DUP AP: The_Beach
*wm:[2] DUP AP: The_Beach
*wm:[2] AP:  -56 The_Beach
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Scan is cached 24725 ms ago
*wm:[2] Portal Timeout In 102 seconds
*wm:[2] <- HTTP WiFi save  
*wm:[2] Parameters 
*wm:[2] -------------------- 
*wm:[2] server: 192.168.1.11
*wm:[2] port: 1883
*wm:[2] user: 
*wm:[2] pass: 
*wm:[2] secure: 
*wm:[2] cert: 
*wm:[2] name: OpenMQTTGateway_lilygo_rtl_433_ESP
*wm:[2] topic: home/
*wm:[2] ota: 
*wm:[2] -------------------- 
*wm:[2] processing save 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to NEW AP: The_Beach
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS] 
*wm:[1] Got IP Address: 
*wm:[1] 192.168.1.160 
*wm:[2] [CB] _savewificallback calling 
*wm:[2] shutdownConfigPortal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[0] [ERROR] disconnect configportal - softAPdisconnect FAILED 
*wm:[2] restoring usermode STA
dhcps: send_nak>>udp_sendto result 0
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed 
*wm:[1] config portal exiting 
{
  "mqtt_server": "192.168.1.11",
  "mqtt_port": "1883",
  "mqtt_user": "",
  "mqtt_pass": "",
  "mqtt_topic": "home/",
  "gateway_name": "OpenMQTTGateway_lilygo_rtl_433_ESP",
  "mqtt_broker_secure": false,
  "mqtt_broker_cert": "",
  "mqtt_ss_index": 0,
  "ota_server_cert": "",
  "ota_pass": ""
}N: ZgatewayRTL_433 setup done 
N: Switching to RTL_433 Receiver: 433.92Mhz
N: OpenMQTTGateway modules: ["LilyGo_SSD1306","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
N: Connected to broker
N: Send on /SYStoMQTT msg {"uptime":82,"version":"v1.4.0","freemem":149624,"mqttport":"1883","mqttsecure":false,"tempc":47.77778,"freestack":4728,"rssi":-54,"SSID":"The_Beach","BSSID":"90:72:40:18:A7:BE","ip":"192.168.1.160","mac":"4C:75:25:A8:85:04","actRec":3,"mhz":433.92,"RTLRssiThresh":-82,"RTLRssi":-94,"RTLAVGRssi":0,"RTLCnt":1,"RTLOOKThresh":90,"modules":["LilyGo_SSD1306","rtl_433"]}
N: uniqueid: LaCrosse-TX141W-0-168901
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/168901 msg {"model":"LaCrosse-TX141W","id":168901,"channel":0,"battery_ok":1,"temperature_C":-9.6,"humidity":68,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-60,"duration":641996}
N: uniqueid: LaCrosse-TX141W-0-168901
```

NorthernMan54, 2023-02-24:
And this is monitor output from a regular startup

```
N: 
************* WELCOME TO OpenMQTTGateway **************
N: Setup SSD1306 Display end
N: OpenMQTTGateway Version: v1.4.0
{
  "mqtt_server": "192.168.1.11",
  "mqtt_port": "1883",
  "mqtt_user": "",
  "mqtt_pass": "",
  "mqtt_topic": "home/",
  "gateway_name": "OpenMQTTGateway_lilygo_rtl_433_ESP",
  "mqtt_broker_secure": false,
  "mqtt_broker_cert": "",
  "mqtt_ss_index": 0,
  "ota_server_cert": "",
  "ota_pass": ""
}*wm:[2] Added Parameter: server
*wm:[2] Added Parameter: port
*wm:[2] Added Parameter: user
*wm:[2] Added Parameter: pass
*wm:[2] Added Parameter: secure
*wm:[2] Added Parameter: cert
*wm:[2] Added Parameter: name
*wm:[2] Added Parameter: topic
*wm:[2] Added Parameter: ota
N: Attempting Wifi connection with saved AP: 0
N: Attempting Wifi connection with saved AP: 1
E (4143) wifi:sta is connecting, return error
[E][WiFiSTA.cpp:221] begin(): connect failed!
N: Attempting Wifi connection with saved AP: 2
E (5145) wifi:sta is connecting, return error
[E][WiFiSTA.cpp:221] begin(): connect failed!
N: ZgatewayRTL_433 setup done 
N: Switching to RTL_433 Receiver: 433.92Mhz
N: OpenMQTTGateway modules: ["LilyGo_SSD1306","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
N: Connected to broker
N: Send on /SYStoMQTT msg {"uptime":10,"version":"v1.4.0","freemem":158240,"mqttport":"1883","mqttsecure":false,"tempc":47.22222,"freestack":4728,"rssi":-52,"SSID":"The_Beach","BSSID":"90:72:40:18:A7:BE","ip":"192.168.1.160","mac":"4C:75:25:A8:85:04","actRec":3,"mhz":433.92,"RTLRssiThresh":-82,"RTLRssi":-96,"RTLAVGRssi":0,"RTLCnt":1,"RTLOOKThresh":90,"modules":["LilyGo_SSD1306","rtl_433"]}
N: uniqueid: Acurite-Tower-C-9899
N: Send on /RTL_433toMQTT/Acurite-Tower/C/9899 msg {"model":"Acurite-Tower","id":9899,"channel":"C","battery_ok":1,"temperature_C":20.4,"humidity":40,"mic":"CHECKSUM","protocol":"Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas","rssi":-68,"duration":181996}
```

NorthernMan54, 2023-02-24:
PS Our boards are identical

NorthernMan54, 2023-02-24:
Looking at my startup log versus your startup log, am thinking that the exception is occurring around the time when the credentials are read from the local storage.  Either they were not saved correctly during the setup phase, or reading them is causing an exception.  ( am thinking a particular setting is triggering this )

Now going to retry with the dev build

NorthernMan54, 2023-02-24:
And this is the serial log from the dev build 

```
*wm:[2] Portal Timeout In 101 seconds
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Scan is cached 18663 ms ago
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Scan is cached 24549 ms ago
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Scan is cached 24637 ms ago
*wm:[2] <- HTTP Wifi 
*wm:[2] Scan is cached 27305 ms ago
*wm:[1] 9 networks found
*wm:[2] DUP AP: The_Beach
*wm:[2] DUP AP: The_Beach
*wm:[2] DUP AP: The_Beach
*wm:[2] DUP AP: The_Beach
*wm:[2] DUP AP: BELL664
*wm:[2] DUP AP: BELL664
*wm:[2] AP:  -46 The_Beach
*wm:[2] AP:  -90 BELL664
*wm:[2] AP:  -91 SmartRG-8bb4
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] Scan is cached 27415 ms ago
*wm:[2] Portal Timeout In 105 seconds
*wm:[2] <- HTTP WiFi save  
*wm:[2] Parameters 
*wm:[2] -------------------- 
*wm:[2] server: 192.168.1.11
*wm:[2] port: 1883
*wm:[2] user: 
*wm:[2] pass: 
*wm:[2] secure: 
*wm:[2] cert: 
*wm:[2] name: OpenMQTTGateway_lilygo_rtl_433_ESP
*wm:[2] topic: home/
*wm:[2] ota: 
*wm:[2] -------------------- 
*wm:[2] processing save 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to NEW AP: The_Beach
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
dhcps: send_nak>>udp_sendto result 0
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS] 
*wm:[1] Got IP Address: 
*wm:[1] 192.168.1.160 
*wm:[2] [CB] _savewificallback calling 
*wm:[2] shutdownConfigPortal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[0] [ERROR] disconnect configportal - softAPdisconnect FAILED 
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed 
*wm:[1] config portal exiting 
{
  "mqtt_server": "192.168.1.11",
  "mqtt_port": "1883",
  "mqtt_user": "",
[... 55 lines cut ...]
*wm:[2] Added Parameter: port
*wm:[2] Added Parameter: user
*wm:[2] Added Parameter: pass
*wm:[2] Added Parameter: secure
*wm:[2] Added Parameter: cert
*wm:[2] Added Parameter: name
*wm:[2] Added Parameter: topic
*wm:[2] Added Parameter: ota
N: Attempting Wifi connection with saved AP: 0
N: Attempting Wifi connection with saved AP: 1
E (3624) wifi:sta is connecting, return error
[E][WiFiSTA.cpp:221] begin(): connect failed!
N: Attempting Wifi connection with saved AP: 2
E (4626) wifi:sta is connecting, return error
[E][WiFiSTA.cpp:221] begin(): connect failed!
N: ZgatewayRTL_433 setup done 
N: Switching to RTL_433 Receiver: 433.92Mhz
N: OpenMQTTGateway modules: ["LilyGo_SSD1306","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
N: Connected to broker
N: Send on /SYStoMQTT msg {"uptime":10,"version":"d124cd","freemem":157876,"mqttport":"1883","mqttsecure":false,"tempc":46.66667,"freestack":4648,"rssi":-52,"SSID":"The_Beach","BSSID":"90:72:40:18:A7:BE","ip":"192.168.1.160","mac":"4C:75:25:A8:85:04","actRec":3,"mhz":433.92,"RTLRssiThresh":-82,"RTLRssi":-70,"RTLAVGRssi":0,"RTLCnt":1,"RTLOOKThresh":90,"modules":["LilyGo_SSD1306","rtl_433"]}
N: uniqueid: LaCrosse-TX141W-0-168901
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/168901 msg {"model":"LaCrosse-TX141W","id":168901,"channel":0,"battery_ok":1,"temperature_C":-10.2,"humidity":70,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-50,"duration":641000}
```

1technophile, 2023-02-25:
@hanssipi could you confirm that you erased the flash ?

hanssipi, 2023-02-25:
@1technophile, The "Erase Flash (erases all saved data)" checkbox was allways checked.
If it actually erased the flash i don't know.

DigiH, 2023-02-25:
@hanssipi, there is a new development test build 0ca5d5 for which I have also added a new LilyGo binary `lilygo-ble`

While not the rtl_433 gateway you want, it would be good to try and see if a different module gateway still has the same boot loop issue for you or not.

Please install  `lilygo-ble` from

https://docs.openmqttgateway.com/dev/upload/web-install.html

again with **Erase Flash** checked, and let us know how it turns out.

hanssipi, 2023-02-25:
@DigiH , @1technophile, @NorthernMan54 
Thanks so much guys for all your effords. The test build was not sussesful, still in the boot loop. So i ordered a new board today and when it arrives i will let you know how i get on.

Test monitor output did produce another error:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
N: 
************* WELCOME TO OpenMQTTGateway **************
N: Setup SSD1306 Display end
N: OpenMQTTGateway Version: 0ca5d5
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40159950  PS      : 0x00060230  A0      : 0x8014f394  A1      : 0x3ffce0e0  
A2      : 0x3ffdb674  A3      : 0x00000000  A4      : 0x000001e0  A5      : 0x3ffdb944  
A6      : 0x00001000  A7      : 0x0000001e  A8      : 0x8014f2ea  A9      : 0x3ffce0a0  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffdb65c  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40159950:0x3ffce0e0 0x4014f391:0x3ffce100 0x4014d272:0x3ffce130 0x4014dcfa:0x3ffce170 0x4010751c:0x3ffce230 0x400e0f65:0x3ffce260 0x400e1ae7:0x3ffce440 0x4010d542:0x3ffce520 0x400911de:0x3ffce540

NorthernMan54, 2023-02-25:
I was wondering if you could collect a monitor output from when the wifi portal is collecting the wifi information until it is running/erroring out.  It may offer more clues into why

DigiH, 2023-02-25:
Hi @hanssipi,

I've created one last test development build **7eb8fb**, this time with the display being disabled, just in case faulty display on your LilyGo might cause these issues.

You can install it again with the `lilygo-ble` build from

https://docs.openmqttgateway.com/dev/upload/web-install.html

until it is overwritten by the nightly dev builds tonight.

hanssipi, 2023-02-25:
@DigiH ,

Display is now off, rboot every second now.

N: 
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: 7eb8fb
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401c5b2c  PS      : 0x00060630  A0      : 0x801859d0  A1      : 0x3ffcdaf0  
A2      : 0x3ffca144  A3      : 0x3ffcdb44  A4      : 0x3ffcdb48  A5      : 0x3ffcdb4c  
A6      : 0x00000006  A7      : 0x00000000  A8      : 0x80185892  A9      : 0x3ffcda40  
A10     : 0x3ffca144  A11     : 0x3ffcdb48  A12     : 0x3ffcda98  A13     : 0x00000000  
A14     : 0x3ffcda6c  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401c5b2c:0x3ffcdaf0 0x401859cd:0x3ffcdb10 0x40185b6b:0x3ffcdb30 0x401813fc:0x3ffcdb70 0x4017f737:0x3ffcdb90 0x401ae8f1:0x3ffcdbb0 0x400e3ea6:0x3ffcdbf0 0x400e43bd:0x3ffcdc20 0x400dfa59:0x3ffcdd30 0x400e05b2:0x3ffcdf10 0x4010a82e:0x3ffcdff0 0x40090926:0x3ffce010

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4

DigiH, 2023-02-25:
A real shame :( I'm out of ideas now, especially as @NorthernMan54 and myself have exactly the same board as you.

Looking forward to hearing (expectedly positive) news from when your new board arrives.

hanssipi, 2023-02-25:
> I was wondering if you could collect a monitor output from when the wifi portal is collecting the wifi information until it is running/erroring out. It may offer more clues into why

Reboot is very fast, 1-3 seconds. AP is not created in this time. The problem is someware before the AP is created or when it starts.
I have never seen the AP en not ever could add my cerdentials.

hanssipi, 2023-02-25:
> A real shame :( I'm out of ideas now, especially as @NorthernMan54 and myself have exactly the same board as you.
> 
> Looking forward to hearing (expectedly positive) news from when your new board arrives.

I will keep you informed. Thanks!!

NorthernMan54, 2023-02-25:
@hanssipi What you said raised my eyebrows a bit on the issue

These are the normal steps

1- Plug in your ESP to a USB port.
2 - Select the firmware in the box below.
3 - Click the install button and choose the port that the ESP is connected to.
4 - Wait until the process is complete.
5 - Once completed you can configure your [Wifi and MQTT credentials](https://docs.openmqttgateway.com/upload/portal.html)
6 - Board connects to your network and starts works

We had been assuming that the error was after step 5, but your saying it was before step 5

Which points us to suspect that the flash memory contains something that is causing the exception, and the `erase flash` option is not working properly.  And wanted you try a different approach to erase the flash memory.

Can you download the `flash tool utility` from here - https://docs.openmqttgateway.com/upload/binaries.html#esp32

And use these options to erase the flash, then try the web download again.

[image]

hanssipi, 2023-02-25:
@NorthernMan54 
Correct! Will try the tool tomorrow. Thanks

hanssipi, 2023-02-26:
@NorthernMan54 ,

Erase was succesfull, see below, but upload not. Same boot loop.

*************************** START ****************************

START TIME: 202302261217

CONNECT BAUD: 115200
set state: ESP_DL_SYNC
serial port opened
-----------
baud:115200
root baud:115200
-------------
===============BAUD : 115200===============CALL DEVICE SYNC
connecting...
mac l: 0x6d0096d8
mac h: 0xa1e89f
crc_cal: 161
crc_read: 161
crc_test: 0
ESP32 MAC CRC OK
get mac res: 1
get flash id : 0x001640c8

manufacturer_id: 0xc8

device_id: 0x4016

com closed
Use time: 5 s

Download result: PASS

*************************** END ****************************

hanssipi, 2023-02-26:
I would like to try option 2: Upload ready to go binaries but not sure what files i need.
Are the files en adresses correct in the image below?

[image]

DigiH, 2023-02-26:
Hi @hanssipi, I have to admit I've never tried this upload option, but apart from the linked bootloader and boot_app0, the additionally required files (firmware and partitions) are available on the OpenMQTTGateway release page

https://github.com/1technophile/OpenMQTTGateway/releases/tag/v1.4.0

Let us know how you get on.

hanssipi, 2023-02-26:
Just loaded this project with arduino: https://github.com/Xinyuan-LilyGO/TTGO-LoRa-Series
No problems loads fine and no boot loop :) So the board works.... i think.
Don't know what the program does exacly...

DigiH, 2023-02-26:
How about trying `lilygo-rtl_433` with the Arduino IDE, just requiring the additionally available

lilygo-rtl_433-libraries.zip

to the above two mentioned files.

Also if you are familiar with the Arduino IDE it might be worth you downloading Visual Studio Code and just installing the Platformio Extension from within VSC. These would be the sole requirements for building and uploading your own `lilygo-rtl_433`, possibly with manual WiFi and MQTT credentials, to see if the initial WiFiManager might have been the issue for you.

hanssipi, 2023-02-27:
@DigiH ,
Upload with pio has the same result. No errors but boot loop :(
Board check in pio has errors: see below. The weird thing is when i start the build i see:

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v21.html
PLATFORM: Espressif 32 (3.5.0) > TTGO LoRa32-OLED v2.1.6

Is this correct? Board version is V1.6.1 so there is a differance.
Should it not be:
CONFIGURATION:  https://docs.platformio.org/en/latest/boards/espressif32/ttgo-lora32-v1.html

-------------
Executing task in folder OpenMQTTGateway-development: C:\Users\hanslt\.platformio\penv\Scripts\platformio.exe check --environment lilygo-rtl_433 

Checking lilygo-rtl_433 > cppcheck (platform: espressif32@3.5.0; board: ttgo-lora32-v21; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------main\ZmqttDiscovery.ino:103: [low:style] The function 'announceDeviceTrigger' is never used. [unusedFunction]
main\ZmqttDiscovery.ino:380: [low:style] The function 'eraseTopic' is never used. [unusedFunction]
main\ZmqttDiscovery.ino:386: [low:style] The function 'pubMqttDiscovery' is never used. [unusedFunction]     
main\ZmqttDiscovery.ino:103: [low:style] The function 'announceDeviceTrigger' is never used. [unusedFunction]
main\ZmqttDiscovery.ino:380: [low:style] The function 'eraseTopic' is never used. [unusedFunction]
main\ZmqttDiscovery.ino:386: [low:style] The function 'pubMqttDiscovery' is never used. [unusedFunction]     
Error: cppcheck failed to perform check! Please examine tool output in verbose mode.
================================================================== [FAILED] Took 458.22 seconds ==================================================================
Component     HIGH    MEDIUM    LOW 
-----------  ------  --------  -----
main           0        0        6  

Total          0        0        6  

Environment     Tool      Status    Duration
--------------  --------  --------  ------------
lilygo-rtl_433  cppcheck  FAILED    00:07:38.218
============================================================= 1 failed, 0 succeeded in 00:07:38.218 ============================================================= 

 *  The terminal process "C:\Users\hanslt\.platformio\penv\Scripts\platformio.exe 'check', '--environment', 'lilygo-rtl_433'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

DigiH, 2023-02-27:
Hi @hanssipi , I know the board version naming is a bit confusing, but from your pictures, and my board showing exactly the same model description I really think yours is one .1 version higher than the specified

http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1130&FId=t3:50003:3

So LoRa32 V2.1 1.6(.1 in our case) - unless you want to try changing the board definition to `ttgo-lora32-v2` and `ttgo-lora32-v1` in the lilygo-rtl_433 environment for some test builds and uploads.

hanssipi, 2023-02-27:
@DigiH , Thanks. So no luck. Don't know why it won't work. 
I give up, I'll wait for the new board.

Thanks for your help!

hanssipi, 2023-03-01:
SUCCESS!
The new board works fine!
Thanks to all of you for helping me! Keep up the good work.

[image]

DigiH, 2023-03-01:
> I have the same issue here !

Hi @Robbe-B, I assume you have tried all possibilities we tried above with @hanssipi's initial board?

Can you confirm that your board is also a T3_V1.6.1?

Can you also tell us with which PC and browser you are doing the web upload?

As you saw, for hanssipi the issue was solved with a new board, being exactly the same T3_V1.6.1 model. If the previous board had a fault or if there might actually be undocumented hardware differences between same T3_V1.6.1 versions we don't really know at this stage.

Any further information will hopefully help in pinning this down.

hanssipi, 2023-03-01:
> I have the same issue here ! What is the conclusion here, was this a faulty board ?

Yes the board was faulty.

DigiH, 2023-03-01:
While Andreas might have used a v1.3.0 in his video, hanssipi, myself and others in the team all have the same T3_V1.6.1 model as you have. With hanssipi's new board of this model and our models all is working fine, so it's really hard to pin down what the issue might be.

From another tread though, would you mind trying the web upload not with the Chrome browser, but with the Microsoft Edge browser?

I cannot guarantee it will solve you problem, but it's definitely worth a try at this stage.

Robbe-B, 2023-03-01:
I've just tried edge, nothing different.
Well under the hood they are essentially both chromium based browsers, so I guess that makes sense.

DigiH, 2023-03-01:
@hanssipi - as you have installed Platformio would you mind doing an **Erase Flash** in Platformio, with your old board which had the problem?

I'm wondering what it comes back with after
```
Connecting......
Chip is
```

Robbe-B, 2023-03-01:
I'm pretty much new to this project, but can I try to do anything more ? 
Maybe gathering logging of some sort ?

DigiH, 2023-03-01:
> I'm pretty much new to this project, but can I try to do anything more ?

Could you try to upload latest dev test build at

https://docs.openmqttgateway.com/dev/upload/web-install.html

> Maybe gathering logging of some sort ?

You could get the serial output with Platformio installed (just an extension to downlaod and enable in Visual Studio Code) or with the Arduino IDE.

Robbe-B, 2023-03-01:
Well, I installed the dev version and took the serial output, here are the results: 
14:43:17.160 -> N: 
14:43:17.160 -> ************* WELCOME TO OpenMQTTGateway **************
14:43:18.277 -> N: Setup SSD1306 Display end
14:43:18.277 -> N: OpenMQTTGateway Version: 454a06
14:43:18.416 -> *wm:[2] Added Parameter: server
14:43:18.416 -> *wm:[2] Added Parameter: port
14:43:18.463 -> *wm:[2] Added Parameter: user
14:43:18.463 -> *wm:[2] Added Parameter: pass
14:43:18.463 -> *wm:[2] Added Parameter: secure
14:43:18.463 -> *wm:[2] Added Parameter: cert
14:43:18.463 -> *wm:[2] Added Parameter: name
14:43:18.463 -> *wm:[2] Added Parameter: topic
14:43:18.463 -> *wm:[2] Added Parameter: ota
14:43:18.463 -> N: Attempting Wifi connection with saved AP: 0
14:43:18.463 -> [E][WiFiSTA.cpp:221] begin(): connect failed!
14:43:19.440 -> N: Attempting Wifi connection with saved AP: 1
14:43:19.440 -> [E][WiFiSTA.cpp:221] begin(): connect failed!
14:43:20.469 -> N: Attempting Wifi connection with saved AP: 2
14:43:20.469 -> [E][WiFiSTA.cpp:221] begin(): connect failed!
14:43:21.448 -> N: Attempting Wifi connection with saved AP: 3
14:43:21.448 -> [E][WiFiSTA.cpp:221] begin(): connect failed!
14:43:22.467 -> N: Attempting Wifi connection with saved AP: 4
14:43:22.467 -> [E][WiFiSTA.cpp:221] begin(): connect failed!
14:43:25.527 -> N: Connect your phone to WIFI AP: OpenMQTTGateway_lilygo_rtl_433_ESP with PWD: your_password
14:43:25.527 -> *wm:[1] AutoConnect 
14:43:25.527 -> *wm:[2] ESP32 event handler enabled 
14:43:25.527 -> *wm:[2] Connecting as wifi client... 
14:43:25.527 -> *wm:[2] setSTAConfig static ip not set, skipping 
14:43:25.527 -> *wm:[1] No wifi saved, skipping 
14:43:25.527 -> *wm:[2] Connection result: WL_NO_SSID_AVAIL
14:43:25.527 -> *wm:[1] AutoConnect: FAILED 
14:43:25.527 -> *wm:[2] Starting Config Portal 
14:43:25.527 -> *wm:[2] AccessPoint set password is VALID 
14:43:25.527 -> *wm:[2] Disabling STA 
14:43:25.527 -> *wm:[2] Enabling AP 
14:43:25.527 -> *wm:[1] StartAP with SSID:  OpenMQTTGateway_lilygo_rtl_433_ESP
14:43:26.085 -> *wm:[1] AP IP address: 192.168.4.1
14:43:26.085 -> *wm:[1] Starting Web Portal 
14:43:26.085 -> *wm:[2] HTTP server started 
14:43:26.085 -> *wm:[2] WiFi Scan ASYNC started 
14:43:26.085 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
14:43:26.085 -> *wm:[2] Portal Timeout In 120 seconds
14:43:31.190 -> *wm:[2] WiFi Scan ASYNC completed in 5113 ms
14:43:31.190 -> *wm:[2] WiFi Scan ASYNC found: 16
14:43:47.140 -> *wm:[2] Portal Timeout In 98 seconds
14:44:17.145 -> *wm:[2] Portal Timeout In 68 seconds
14:44:47.140 -> *wm:[2] Portal Timeout In 38 seconds
14:45:17.117 -> *wm:[2] Portal Timeout In 8 seconds
14:45:25.531 -> *wm:[1] config portal has timed out 
14:45:25.531 -> *wm:[2] shutdownConfigPortal 
14:45:25.531 -> *wm:[0] [ERROR] disconnect configportal - softAPdisconnect FAILED 
14:45:25.531 -> *wm:[2] restoring usermode STA
14:45:26.554 -> *wm:[2] wifi status: WL_DISCONNECTED
14:45:26.554 -> *wm:[2] wifi mode: STA
14:45:26.554 -> *wm:[2] configportal closed 
14:45:26.554 -> *wm:[1] config portal exiting 
14:45:26.554 -> W: failed to connect and hit timeout
14:45:29.576 -> W: Rebooting for reason code 3
14:45:29.576 -> ets Jun  8 2016 00:22:57
14:45:29.576 -> 
14:45:29.576 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:45:29.576 -> configsip: 188777542, SPIWP:0xee
14:45:29.576 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:45:29.576 -> mode:DIO, clock div:2
14:45:29.576 -> load:0x3fff0030,len:1184
14:45:29.576 -> load:0x40078000,len:13160
14:45:29.576 -> load:0x40080400,len:3036
14:45:29.576 -> entry 0x400805e4

DigiH, 2023-03-01:
> 14:43:25.527 -> N: Connect your phone to WIFI AP: OpenMQTTGateway_lilygo_rtl_433_ESP with PWD: your_password
> 14:43:25.527 -> *wm:[1] StartAP with SSID: OpenMQTTGateway_lilygo_rtl_433_ESP
> 14:43:26.085 -> *wm:[2] Portal Timeout In 120 seconds
> 14:45:17.117 -> *wm:[2] Portal Timeout In 8 seconds
> 14:45:25.531 -> *wm:[1] config portal has timed out

Did you try and connect to the OpenMQTTGateway WiFi AP for initial set-up during that time?

The possibly previously entered credentials might have been wrong, as

> 14:43:18.416 -> *wm:[2] Added Parameter: server
> 14:43:18.416 -> *wm:[2] Added Parameter: port
> 14:43:18.463 -> *wm:[2] Added Parameter: user
> 14:43:18.463 -> *wm:[2] Added Parameter: pass
> 14:43:18.463 [...]

the connection failed and the initial set-up AP was started again.

Robbe-B, 2023-03-01:
Well, the problem is, ...  you need to be very fast to fill in all credentials within 2mins.
However, I did managed to do it in time but still restarting periodically afterwards.

DigiH, 2023-03-01:
> Well, the problem is, ... you need to be very fast to fill in all credentials within 2mins.

We are aware of this issue and it is being addressed ;) maybe copy your longest parts first, then paste it into the relevant portal field.

> However, I did managed to do it in time but still restarting periodically afterwards.

There does seem to be an issue with parts of the entered credentials though so that it cannot connect to your WiFi/MQTT broker, as the last quoted reply above shows. In such a case OMG restarts with the initial set-up AP again to allow for correction of the credential to be entered again.

Could you post what details you enter, obviously with WiFi password etc. changed?

Robbe-B, 2023-03-01:
I believe I got it working 🤣.
First reboot cycle is the issue you are working on.
Second reboot cycle was because of diffrent vlans, the board couldn't reach the MQTT Broker (my bad).
These 2 together made me very confused.
Thanks for the help guys ! 
[image]

closed 2023-09-19 by github-actions[bot]