1technophile/OpenMQTTGateway #1483: Unable to write to unlisted BLE services

ithinkido, 2023-02-23:
**Describe the bug**
I would like to read and write BLE data to a device with services that are not actively advertised. These services show up on scan with Nodrics NRF scanner, but are not show in the serial logs when using OMG. I have been able to send and read data from these services using Nordics NRF app. 

**To Reproduce**
This is an example of a read command that I was sending:
```json
mosquitto_pub -h 192.168.1.200 -t home/OpenMQTTGateway_ESP32_BLE/commands/MQTTtoBT/config -m '{
"ble_read_address":"bbbbbbbbbbbb",
"ble_read_service":"f6d75f91-5a10-4eba-a233-47d3f26a907f",
"ble_read_char":"00000001-0000-1000-8000-00805f9b34fb",
"value_type":"INT",
"ttl": 2 }'
```

**Expected behavior**
One possible issue that may be going on here is that the data format for the device is `uint16` , but either way, I would expect some sort of reply, even a decoding error, which I am not seeing in the serial logs. 
The command is appearing in the serial logs , but there is no sign of any reply. 
I did also try to read the value as `HEX` , but this did not change the behavior. 

**LOGS**
The device is show in the OMG logs as follows : 
```
N: Send on /BTtoMQTT/bbbbbbbbbbb msg {"id":"bbbbbbbbbbb","mac_type":0,"adv_type":0,"name":"Pinecil","rssi":-72}
```
However there are many more services that show up on the NRF scanner
[Log_2023-02-23_13_48_22.txt](https://github.com/1technophile/OpenMQTTGateway/files/10815588/Log_2023-02-23_13_48_22.txt)

**Environment (please complete the following information):**
* OpenMQTTGateway version used 
OpenMQTTGateway Version: v1.4.0
* Library version related to the problem you have (
esp32dev-ble-datatest

DigiH, 2023-02-23:
Hi @ithinkido, your above read command should work, even if the services are not advertised, short from a typo not immediately visible.

Is/.was the device connected to another machine/NRF scanner when you tried to read? Does it require a PIN for connecting?

And increasing the TTL maybe.

```
mosquitto_pub -t home/OpenMQTTGateway_ESP32_BLE/commands/MQTTtoBT/config -m '{
  "ble_read_address":"AA:BB:CC:DD:EE:FF",
  "ble_read_service":"f6d75f91-5a10-4eba-a233-47d3f26a907f",
  "ble_read_char":"00000001-0000-1000-8000-00805f9b34fb",
  "value_type":"INT",
  "ttl": 5 }'
```

DigiH, 2023-02-23:
Or possibly related to

https://github.com/1technophile/OpenMQTTGateway/issues/1437

ithinkido, 2023-02-23:
Thanks for the very fast reply !
Could I ask for some clarification - In the [docs ](https://docs.openmqttgateway.com/use/ble.html#example-write-command) the topic to send to is listed as `.... MQTTtoBT` but in MQTT explorer the origanl topic looks like it is `....BTtoMQTT`. The same goes for `OpenMQTTGateway` vs `OpenMQTTGateway_ESP32_BLE`

[image]

I have used the`MQTTtoBT`  and `OpenMQTTGateway_ESP32_BLE` to succesfully change settings so I think it must be the correct one. 

here is an extract from the serial logs and it looks like there is no reply being shown. 

```
N: Found 6 devices, scan number 52 end
N: [ MQTT->OMG ]: {"ble_read_address":"dddddddddA","ble_read_service":"f6d75f91-5a10-4eba-a233-47d3f26a907f","ble_read_char":"00000001-0000-1000-8000-00805f9b34fb","value_type":"HEX","ttl":5}
N: Send on /BTtoMQTT msg {"bleconnect":false,"interval":10000,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"onlysensors":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":true,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"btqblck":0,"btqsum":157,"btqsnd":139,"btqavg":1.129496}
N: Scan begin
N: Device detected: aaaaaaaaaa
N: Device detected: bbbbbbbbbb
N: Device detected: ccccccccccc
N: Send on /BTtoMQTT/dddddddddddd msg {"id":"dddddddddd","mac_type":0,"adv_type":0,"name":"Pinecil","rssi":-83}
N: Device detected: eeeeeeeee
N: Device detected: fffffffffffffffffff
N: Device detected: gggggggg
N: Found 6 devices, scan number 53 end
```

I have set `pubadvdata":true` in the hope of getting OMG to list all services, but it seems like it is still omly listing the advertised services. Is there any way to enable the display of all the service UUIDs  from OMG ?

ithinkido, 2023-02-23:
I have found my error. The  `"value_type":"INT",`  is not working , presumably because the data is in uint16_t format. (?)
Changing over to `"value_type":"HEX"`, now returns useful data.

closed 2023-02-23 by ithinkido

DigiH, 2023-02-23:
@ithinkido, there wasn't any manufacturer- or service-data in your one line log references above, but just out of curiosity, does the Pinecil sometimes show possible useful data which might be included in a decoder for [Theengs Decoder](https://github.com/theengs/decoder)? ;)

ithinkido, 2023-02-23:
It shows a LOT of data, so much so that it causes heap overflow on ESPhome when all settings are added.

[Pinecil.txt](https://github.com/1technophile/OpenMQTTGateway/files/10819223/Pinecil.txt)

These are some links back to the related source code and other projects working on the BLE side of this project.
https://github.com/Ralim/IronOS/blob/68d18a4fb0c468cb276ff89a37461f169cb9e695/source/Core/BSP/Pinecilv2/ble_characteristics.h
https://github.com/joric/pinecil
https://github.com/builder555/PineSAM
https://github.com/TomW1605/esphome_pinecilv2_ble

DigiH, 2023-02-24:
Thanks for all the info, I'll be looking into that.

What I meant though was freely broadcast advertisement data from the Penecil, which I couldn't see any in your 

`N: Send on /BTtoMQTT/bbbbbbbbbbb msg {"id":"bbbbbbbbbbb","mac_type":0,"adv_type":0,"name":"Pinecil","rssi":-72}`

i.e. no manufacturerdata nor servicedata included there which Decoder could use for status decoding.

ithinkido, 2023-02-24:
> Thanks for all the info, I'll be looking into that.
>
> What I meant though was freely broadcast advertisement data from the Penecil, which I couldn't see any in your
>
> `N: Send on /BTtoMQTT/bbbbbbbbbbb [...]

Nope, as far as I know , there is no other advertisement data broadcast. I believe this was a conscious decision as many of the properties are not part of the BLE attribute standard ( eg calibration offset etc)