1technophile/OpenMQTTGateway #1487: rtl_433 MQTT illegal discovery topic for Skylink devices (spaces?)

Snuffy2, 2023-02-25:
**Describe the bug**
I don't have any Skylink devices, but it appears that my neighbors may have some as they are routinely getting detected by OMG. They are giving the MQTT error in Home Assistant below. I don't know for sure, but it appears it may be from the spaces in the name. Per Home Assistant [documentation](https://www.home-assistant.io/integrations/mqtt#discovery-topic) they do't support spaces in the topic. Could the spaces be replaced by underscores?

Home Assistant error:
`
2023-02-25 17:42:32.004 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/binary_sensor/Skylink HA-434TL motion sensor-08000-motion/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/docs/mqtt/discovery/#discovery-topic
`

**Environment (please complete the following information):**
* OpenMQTTGateway version: v1.2.0
* Hardware: Heltec WiFi LoRa 32 (V2.1) 
* Library: rtl_433

[image]

NorthernMan54, 2023-02-26:
I also encountered similar concerns and this development pull request resolves them #1489 

Once this is merged, you should be able to install a development build that resolves this.

1technophile, 2023-03-06:
You can turn off auto-discovery with the following command:
https://docs.openmqttgateway.com/use/gateway.html#auto-discovery

The board will stop creating these entities

1technophile, 2023-04-06:
Note that with the last version, auto discovery is now turned OFF after 30minutes
Also the switch is now visible in HA

Closing, let me know if you have any question

closed 2023-04-06 by 1technophile