1technophile/OpenMQTTGateway #1488: multi-receiver stops working after several days (memory is bleeding)

fenyvesi, 2023-02-26:
**Describe the bug**
After several days of continuos work it doesn't send any message. 
**To Reproduce**
Started 3 times with the same result.

**Screenshots**
> {"uptime":136597,"version":"v1.4.0","freemem":139608,"mqttport":"1883","mqttsecure":false,"tempc":56.66667,"freestack":4456,"rssi":-87,"SSID":"MADOX3G","BSSID":"14:CF:92:4A:4F:E8","ip":"192.168.0.173","mac":"30:AE:A4:02:8B:3C","actRec":3,"mhz":433.92,"RTLRssiThresh":-81,"RTLRssi":-92,"RTLAVGRssi":-90,"RTLCnt":5543,"modules":["RF","RF2","Pilight","RTL_433"]}

**Environment (please complete the following information):**
* OpenMQTTGateway version used: precomplied V1.4.0 with defaults

**Additional context**
The Heltec version (v1.4.0) seems to be OK.

I suggest that the memory is bleeding. The free memory decreases 4-500 bytes/hour.

NorthernMan54, 2023-02-26:
Can you share a bit more about how you are using the device, so we try to recreate.

fenyvesi, 2023-02-26:
What shall I say?
The v1.4.0 is installed from the web. The RF is a CC1101 module.  After giving the credentials I use it to catch 3 thermometer's data receiving by each minute.
I can see the proper MQTT messages for a while, but after several days it stops sending.  I tried it twice, but didn't try to send MQTT message to the device when it was inactive. If it stops now, I shall try it, also.

The only suspicious fact I realized that the free memory in the SYStoMQTT messages decreases 4-500 bytes/hour, but momentary after 2 days it stagnates around 136kBytes.
The last message is:
`{"uptime":175598,"version":"v1.4.0","freemem":136296,"mqttport":"1883","mqttsecure":false,"tempc":57.22222,"freestack":4456,"rssi":-86,"SSID":"MADOX3G","BSSID":"14:CF:92:4A:4F:E8","ip":"192.168.0.173","mac":"30:AE:A4:02:8B:3C","actRec":3,"mhz":433.92,"RTLRssiThresh":-82,"RTLRssi":-92,"RTLAVGRssi":-91,"RTLCnt":7263,"modules":["RF","RF2","Pilight","RTL_433"]}`

It might be a connection problem, but the router is 3 meters away.

Please advice me, what else should I say.

fenyvesi, 2023-02-27:
16 hours more ~6 kB:

`{"uptime":238358,"version":"v1.4.0","freemem":130992,"mqttport":"1883","mqttsecure":false,"tempc":57.22222,"freestack":4456,"rssi":-86,"SSID":"MADOX3G","BSSID":"14:CF:92:4A:4F:E8","ip":"192.168.0.173","mac":"30:AE:A4:02:8B:3C","actRec":3,"mhz":433.92,"RTLRssiThresh":-81,"RTLRssi":-92,"RTLAVGRssi":-90,"RTLCnt":10026,"modules":["RF","RF2","Pilight","RTL_433"]}`

fenyvesi, 2023-02-27:
Yeah, you are probably right.
Relating to memory leak: it is right that 130k is more than enough, but it has been decreasing continually. On the other hand the LilyGo version has stable160k for 3 days.

Is there anythnig we can do to avoid it or should I use the Heltec version?

I have a LilyGo/Heltec version, also, The only reason I hesitate to use it is, that I am afraid of the screen would burn in because of the constant, not really changing content.

Thank you for your investigation.

DigiH, 2023-02-27:
> I have a LilyGo/Heltec version, also, The only reason I hesitate to use it is, that I am afraid of the screen would burn in because of the constant, not really changing content.

Hi @fenyvesi , there is something in the pipeline to reduce the burn-in issue

https://github.com/1technophile/OpenMQTTGateway/pull/1484

but you could also remove the display lib_deps and build_flags from your environment altogether to only have the MQTT publishing functionality.

closed 2023-02-28 by fenyvesi