1technophile/OpenMQTTGateway #1490: BT and RLT_433 Gateways both use `static NO_DEVICE_FOUND`, prevents both from being enabled. With pull-request.

SmittyHalibut, 2023-02-27:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Enabling both `ZgatewayBT` and `ZgatewayRTL_433` results in a compile error because both gateways use the same static variable name, but of different types: `NO_DEVICE_FOUND`.

**To Reproduce**
Steps to reproduce the behavior:

I'm using PlatformIO

1. Add both `-DZgatewayBT="BT"` and `-DZgatewayRTL_433="rtl_433"` to `prod_env.ini`
2. Build.
3. See error.

**Expected behavior**
Build succeeds.

**Proposed solution**
Rename `NO_DEVICE_FOUND` to `NO_BT_DEVICE_FOUND` in `ZgatewayBT.ino`, and `NO_RTL_433_DEVICE_FOUND` in `ZgatewayRTL_433.ino`.

Pull request incoming. :-)

GreenEyedAndy, 2023-03-13:
Hi @SmittyHalibut,
I tried to compile the Lilygo-rtl_433 with both `-DZgatewayBT="BT"` and `-DZgatewayRTL_433="rtl_433"` and add the   `${libraries.ble}` in the lib_deps section of in the environments.ini of the latest github version.
Then I build but still get the `NO_DEVICE_FOUND` error and something with `TheengsDecoder::BLE_ID_NUM::UNKNOWN_MODEL`.  
Can you help?

SmittyHalibut, 2023-03-13:
Resolving. THe PR was merged.

closed 2023-03-13 by SmittyHalibut

SmittyHalibut, 2023-03-13:
I don't see this post on the ticket anymore.  Did you delete it?

-Mark

SmittyHalibut, 2023-03-13:
Heh.  No problem. Just making sure I was looking in the right place.  Glad I could help!

Narrator: He did nothing to help.